rockchip-linux/mpp #365: rk3588 - broken hevc hardware decoding with android API

zerorating, 2023-02-23:
Not sure if this is duplicated case, we are seeing green artifacts when decoding hevc source (encoded with x265 encoder - main10 profile) with Android API (media player: kodi/vlc)
some discussion:
https://forum.radxa.com/t/x265-decoding-broken-on-android-green-screen/14000

The issue doesnt seems to be persisting on rockchip API(rockchip media player, mx-player), though the number of video player utilizing it is limited

hardware used: radxa rock5, may persist on orange pi 5

gehee, 2023-12-21:
I am having the same issue on linux so it's seems to be MPP related. Although ffmpeg plays the stream fine in my case (and it uses mpp as well) so it must be a problem with how MPP is used.

HermanChen, 2024-03-13:
Is it fixed on latest mpp code?

zerorating, 2024-03-13:
for android, mo123 (https://github.com/mo123) managed to find the workaround to get the stream decoded properly. He applied workaround on his android tv build and its rendered perfectly (for hevc, hi10p and av1 10bit)
khadas did issued a fix on their firmware previously too.

closed 2024-03-13 by HermanChen

gehee, 2024-04-17:
> for android, mo123 (https://github.com/mo123) managed to find the workaround to get the stream decoded properly. He applied workaround on his android tv build and its rendered perfectly (for hevc, hi10p and av1 10bit)

What was the workaround ?

CanadianBaconBoi, 2024-09-07:
This should be reopened, the issue is not resolved unless you use mo123's private build. This needs to be fixed in upstream.

zerorating, 2024-09-09:
@HermanChen there is requests to had this case re-open.

HermanChen, 2024-09-11:
Which patch fix the issue?

zerorating, 2024-09-11:
no idea, it havent been disclosed by mr mo123