rockchip-linux/mpp #367: 关于多次开启关闭解码器内存泄漏问题

xiaoquelinyuan, 2023-02-28:
你好，目前我们测试多次开启解码器->解码->关闭解码器再开启解码关闭这个流程会造成内存消耗一直增加，最终定位到这个函数只要使用，就会造成内存一直增加：decode_put_packet，请问这个怎么解决呢？

HermanChen, 2023-02-28:
put_packet 之个，MppPacket 有没有判断之后一销毁

xiaoquelinyuan, 2023-02-28:
目前退出关闭解码器时候这些销毁都有调用，您看一下
if (packet) {
        mpp_packet_deinit(packet);
        packet = NULL;
    }

    if (frame) {
        mpp_frame_deinit(frame);
        frame = NULL;
    }

    if (ctx) {
        mpp_destroy(ctx);
        ctx = NULL;
    }

    if (data.pkt_grp) {
        mpp_buffer_group_put(data.pkt_grp);
        data.pkt_grp = NULL;
    }

    if (data.frm_grp) {
        mpp_buffer_group_put(data.frm_grp);
        data.frm_grp = NULL;
    }

xiaoquelinyuan, 2023-02-28:
> put_packet 之个，MppPacket 有没有判断之后一销毁

目前退出关闭解码器时候这些销毁都有调用，您看一下，不知道还有其他地方要销毁的吗？
`if (packet) {
mpp_packet_deinit(packet);
packet = NULL;
}

if (frame) {
    mpp_frame_deinit(frame);
    frame = NULL;
}

if (ctx) {
    mpp_destroy(ctx);
    ctx = NULL;
}

if (data.pkt_grp) {
    mpp_buffer_group_put(data.pkt_grp);
    data.pkt_grp = NULL;
}

if (data.frm_grp) {
    mpp_buffer_group_put(data.frm_grp);
    data.frm_grp = NULL;
}`

HermanChen, 2023-03-01:
参考下 mpi_dec_test，看下 mpp_packet 的处理，每包一个 packet，put 之后判断长度为 0 之后 deinit

xiaoquelinyuan, 2023-03-01:
> 参考下 mpi_dec_test，看下 mpp_packet 的处理，每包一个 packet，put 之后判断长度为 0 之后 deinit

您说的是mpi_dec_test里面put 之后判断长度为 0 之后的mpp_frame_deinit(&frame);吗？这个现在每包put之后判断长度为0后都有调用

zhanglm666, 2024-01-07:
请问，有解决吗？我也遇到了同样的问题，物理内存不断增长

ruiyoua, 2024-11-23:
同样的问题，应该是库的问题，有作者出来解答下吗

HermanChen, 2024-12-03:
> > 参考下 mpi_dec_test，看下 mpp_packet 的处理，每包一个 packet，put 之后判断长度为 0 之后 deinit
> 
> 您说的是mpi_dec_test里面put 之后判断长度为 0 之后的mpp_frame_deinit(&frame);吗？这个现在每包put之后判断长度为0后都有调用

是输入码流 packet 的

HermanChen, 2024-12-03:
先观察下进程的 fd 有没有不断增长，一个  dmabuf 会对应一个 fd，如果 fd 不段增长，说明 MppBuffer 里的 dmabuf 有泄漏。
再是把 mpp_mem 里的 debug 打开，或者在代码埯一段时间打印一下 mpp_mem_total_now，看有没有不断增长，如果有的话说明 malloc 的 mpp_mem 有泄漏

mpp 本身有内部的 buffer 管理机制，可以看到内存的使用情况