rockchip-linux/mpp #368: [RK3566] mpp_rkvdec2 reset repeatedly with mp4 4K60 video on 4K HDMI monitor

chaiwat-screencloud, 2023-02-28:
Kernel log via dmesg (Kernel version 4.19)
```
[  295.905081] mpp_rkvdec2 fdf80200.rkvdec: frame not complete
[  295.905148] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[  295.905421] mpp_rkvdec2 fdf80200.rkvdec: reset done
[  298.188488] mpp_rkvdec2 fdf80200.rkvdec: process task timeout
[  298.188535] rkvdec_link_reg_dump:185: dump link config status from timeout
[  298.188538] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[  298.188542] rkvdec_link_reg_dump:187: reg 1 ffff8000 - cfg addr
[  298.188545] rkvdec_link_reg_dump:188: reg 2 80000002 - link mode
[  298.188549] rkvdec_link_reg_dump:189: reg 4 0000004f - decoded num
[  298.188551] rkvdec_link_reg_dump:190: reg 5 00000054 - total num
[  298.188554] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
[  298.188556] rkvdec_link_reg_dump:192: reg 6 ffff8000 - next ltb addr
[  298.188560] mpp_rkvdec2 fdf80200.rkvdec: frame not complete
[  298.188621] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[  298.188922] mpp_rkvdec2 fdf80200.rkvdec: reset done
[  302.868318] mpp_rkvdec2 fdf80200.rkvdec: process task timeout
[  302.868365] rkvdec_link_reg_dump:185: dump link config status from timeout
[  302.868370] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[  302.868374] rkvdec_link_reg_dump:187: reg 1 ffff8000 - cfg addr
[  302.868377] rkvdec_link_reg_dump:188: reg 2 80000002 - link mode
[  302.868379] rkvdec_link_reg_dump:189: reg 4 0000014b - decoded num
[  302.868381] rkvdec_link_reg_dump:190: reg 5 00000150 - total num
[  302.868383] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
[  302.868385] rkvdec_link_reg_dump:192: reg 6 ffffb000 - next ltb addr
[  302.868389] mpp_rkvdec2 fdf80200.rkvdec: frame not complete
[  302.868442] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[  302.868792] mpp_rkvdec2 fdf80200.rkvdec: reset done
```
RKMPP
```
[431.897] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.897] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.916] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
```

mediainfo
```
$ mediainfo CostaRica1080p60.mp4
General
Complete name                            : CostaRica1080p60.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 196 MiB
Duration                                 : 5 min 13 s
Overall bit rate                         : 5 235 kb/s
Writing application                      : Lavf57.83.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L4.2
Format settings                          : CABAC / 4 Ref Frames
Format settings, CABAC                   : Yes
Format settings, ReFrames                : 4 frames
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 5 min 13 s
Bit rate                                 : 5 094 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 59.940 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.041
Stream size                              : 191 MiB (97%)
Writing library                          : x264 core 152
Encoding settings                        : cabac=1 / ref=3 / deblock=1:0:0 / analyse=0x3:0x113 / me=hex / subme=7 / psy=1 / psy_rd=1.00:0.00 / mixed_ref=1 / me_range=16 / chroma_me=1 / trellis=1 / 8x8dct=1 / cqm=0 / deadzone=21,11 / fast_pskip=1 / chroma_qp_offset=-2 / threads=24 / lookahead_threads=4 / sliced_threads=0 / nr=0 / decimate=1 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / bframes=3 / b_pyramid=2 / b_adapt=1 / b_bias=0 / direct=1 / weightb=1 / open_gop=0 / weightp=2 / keyint=250 / keyint_min=25 / scenecut=40 / intra_refresh=0 / rc_lookahead=40 / rc=crf / mbtree=1 / crf=23.0 / qcomp=0.60 / qpmin=0 / qpmax=69 / qpstep=4 / ip_ratio=1.40 / aq=1:1.00
Codec configuration box                  : avcC

Audio
ID                                       : 2
Format                                   : AAC LC
Format/Info                              : Advanced Audio Codec Low Complexity
Codec ID                                 : mp4a-40-2
Duration                                 : 5 min 13 s
Bit rate mode                            : Constant
Bit rate                                 : 129 kb/s
Channel(s)                               : 2 channels
Channel layout                           : L R
Sampling rate                            : 48.0 kHz
Frame rate                               : 46.875 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 4.81 MiB (2%)
Language                                 : English
Default                                  : Yes
Alternate group                          : 1
```

Ref:
https://github.com/JeffyCN/meta-rockchip/issues/51

JeffyCN, 2023-03-01:
please upload your test video and maybe try to test it with gst-play-1.0 <video> too

HermanChen, 2023-03-01:
It seems that the decoder has error and can not recover from the error.
RK3566 rkvdec2 has some error which can not recover by cru reset. We need new kernel with sip reset to resolve this problem.
The kernel and trust.img should be updated to new version.

chaiwat-screencloud, 2023-03-01:
@HermanChen  After update `kernel-source/drivers/{dma-buf, gpu/drm/rockchip/, video/rockchip/}/` from https://github.com/rockchip-linux/kernel/tree/develop-4.19, then mpp_rkvdec2 do not reset.
But do not notice the perfermance improvement. I notice about `CONFIG_DMABUF_PAGE_POOL`.
But `linux-rockchip` do not use this kernel config, the https://github.com/rockchip-linux/kernel/tree/develop-5.10 do not have this config too. Should I enable this config or move to kernel 5.10? 

@JeffyCN I will check with my about video for this test and check with gst-play-1.0.
BTW, rkmpp complain about `rkmpp_enum_fmt(189): compressed format(3) not found`.
Is this cause by AFBC? Should I use latest version of MPP?
```
Failed to query video capabilities: Inappropriate ioctl for device
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.[449.251] [RKMPP] [729] rkmpp_global_init(54): libv4l-rkmpp version: 1.3.3~202001124 log_level: 1, log_fps: 1
[449.252] [RKMPP] [729] rkmpp_parse_options(672): parsing options: dec

[449.252] [RKMPP] [729] plugin_init(766): ctx(0x32001bbc00): plugin inited
[449.252] [RKMPP] [754] decoder_thread_fn(232): ctx(0x32001bbc00): starting decoder thread
[449.252] [RKMPP] [729] rkmpp_enum_fmt(189): compressed format(3) not found
[449.252] [RKMPP] [729] plugin_close(791): ctx(0x32001bbc00): closing plugin
Failed to query video capabilities: Inappropriate ioctl for device
[449.257] [RKMPP] [729] rkmpp_global_init(54): libv4l-rkmpp version: 1.3.3~202001124 log_level: 1, log_fps: 1
[449.257] [RKMPP] [729] rkmpp_parse_options(672): parsing options: dec

[449.257] [RKMPP] [729] plugin_init(766): ctx(0x32001bbc00): plugin inited
[449.257] [RKMPP] [755] decoder_thread_fn(232): ctx(0x32001bbc00): starting decoder thread
[449.257] [RKMPP] [729] rkmpp_enum_fmt(189): compressed format(3) not found
[449.257] [RKMPP] [729] plugin_close(791): ctx(0x32001bbc00): closing plugin
Failed to query video capabilities: Inappropriate ioctl for device
[449.264] [RKMPP] [729] rkmpp_global_init(54): libv4l-rkmpp version: 1.3.3~202001124 log_level: 1, log_fps: 1
[449.264] [RKMPP] [729] rkmpp_parse_options(672): parsing options: enc

[449.265] [RKMPP] [729] plugin_init(766): ctx(0x32001bbc00): plugin inited
[449.265] [RKMPP] [729] rkmpp_enum_fmt(189): compressed format(2) not found
[449.265] [RKMPP] [757] encoder_thread_fn(232): ctx(0x32001bbc00): starting encoder thread
[449.265] [RKMPP] [729] plugin_close(791): ctx(0x32001bbc00): closing plugin
Failed to query video capabilities: Inappropriate ioctl for device
[449.269] [RKMPP] [729] rkmpp_global_init(54): libv4l-rkmpp version: 1.3.3~202001124 log_level: 1, log_fps: 1
[449.269] [RKMPP] [729] rkmpp_parse_options(672): parsing options: enc
```

JeffyCN, 2023-03-02:
that log is fine, it will try to enum all formats, and reach the latest one

chaiwat-screencloud, 2023-03-02:
@JeffyCN  I test gst-play-1.0 with https://mega.nz/file/hJdFEIBI#uSOjZtkkjIVYSfqD9aSfONf1yq__uQvlsf47pCtFvdQ . 
On the 1st run, I got error the kernel and need to reboot the device.
```
[  278.287206] [drm:vop2_plane_atomic_update] *ERROR* vp0 Cluster0-win0 actual_w[1905] should align as 4 pixel when enable afbc
[  278.312415] [drm:vop2_plane_atomic_update] *ERROR* vp0 Cluster0-win0 actual_w[1905] should align as 4 pixel when enable afbc
[  582.693732] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
[  807.291492] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 3840x2160p60, type: 11 for vp0
```
After reboot, gst-play-1.0 show only green color on the screen without any error message from the kernel.
I think the hardware decoder do not  work.
The output from gst-play-1.0 is below.
```
# gst-play-1.0 The\ World\ in\ HDR.mkv
Press 'k' to see a list of keyboard shortcuts.
Now playing /home/root/video_for_test_performance/The World in HDR.mkv
0:00:01.2 / 0:02:34.5       ali is 'g2p0-01eac0', rk_so_ver is '3'.0:00:00.0 / 0:02:34.5
```

chaiwat-screencloud, 2023-03-02:
with http://distribution.bbb3d.renderfarming.net/video/mp4/bbb_sunflower_2160p_60fps_stereo_abl.mp4
gst-play-1.0  display only top left part of first picture of video. The video do not play.
I do not sure this video do not suport by hardware codec. 
```
# gst-play-1.0 bbb_sunflower_2160p_60fps_stereo_abl.mp4
Press 'k' to see a list of keyboard shortcuts.
Now playing /home/root/video_for_test_performance/bbb_sunflower_2160p_60fps_stereo_abl.mp4
WARNING No decoder available for type 'audio/x-ac3, framed=(boolean)true, rate=(int)48000, channels=(int)6, alignment=(string)frame'.
WARNING debug information: ../gst-plugins-base-1.18.5/gst/playback/gsturidecodebin.c(959): unknown_type_cb (): /GstPlayBin:playbin/GstURIDecodeBin:uridecodebin0
WARNING A lot of buffers are being dropped.eac0', rk_so_ver is '3'.0:00:00.0 / 0:10:34.5
WARNING debug information: ../gstreamer-1.18.5/libs/gst/base/gstbasesink.c(3136): gst_base_sink_is_too_late (): /GstPlayBin:playbin/GstPlaySink:playsink/GstBin:vbin/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink:
There may be a timestamping problem, or this computer is too slow.
WARNING A lot of buffers are being dropped.
WARNING debug information: ../gstreamer-1.18.5/libs/gst/base/gstbasesink.c(3136): gst_base_sink_is_too_late (): /GstPlayBin:playbin/GstPlaySink:playsink/GstBin:vbin/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink:
There may be a timestamping problem, or this computer is too slow.
0:00:31.5 / 0:10:34.5
```

mo123, 2023-03-03:
> with http://distribution.bbb3d.renderfarming.net/video/mp4/bbb_sunflower_2160p_60fps_stereo_abl.mp4 gst-play-1.0 display only top left part of first picture of video. The video do not play. I do not [...]

Hardware MP4 decoder only supports 1920x1080.

JeffyCN, 2023-03-04:
right, 3566's gpu+vpu could not support 4k60fps, unless using afbc compressed format plus zero-copy drm direct display(not supported in current chromium-ozone-wayland)

chaiwat-screencloud, 2023-03-07:
Thank you for an explanation. Previously, my unstanding is v4l-rkmpp use MPP with pure external mode then it use AFBC and dmabuf. So only MPP with DRM  can use AFBC and dmabuf. Is this correct?

JeffyCN, 2023-03-07:
the v4l-rkmpp doesn't use afbc, only gst-rkmpp enabled it with gpu sink(xv/glimage/wayland) and drm sink(kms)

chaiwat-screencloud, 2023-03-08:
OK, thank you

closed 2023-03-08 by chaiwat-screencloud