rockchip-linux/mpp #369: Not finding rockchip_mpp doing ffmpeg config on Orange Pi 5

Fredrum, 2023-02-28:
Hello!

I'm trying to build official master ffmpeg from source on my Armbian Orange Pi 5 but am getting this error:

```
./configure --enable-shared --disable-static --enable-libdrm --enable-opengl --enable-rkmpp --enable-version3
ERROR: rockchip_mpp not found using pkg-config
```

I have tried pointing 'configure' to the .so using

```
./configure --enable-shared --disable-static --enable-libdrm --enable-opengl --enable-rkmpp --enable-version3 --extra-ldflags='-L/usr/lib/aarch64-linux-gnu/ -lrockchip_mpp'
and
 ./configure --enable-shared --disable-static --enable-libdrm --enable-opengl --enable-rkmpp --enable-version3 --extra-libs=-lrockchip_mpp
```

But both of those fail.

I have confirmed that thelib exists,
`ls /usr/lib/aarch64-linux-gnu/librockchip_mpp.so`

My board is
Linux orangepi5 5.10.110-rockchip-rk3588 #23.02.2 SMP Fri Feb 17 23:59:20 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

using latest Armbian + Ubuntu desktop with the ppa multimedia additions for gpu and video.

How can I get the ffmpeg to build?

Cheers!

Fredrum, 2023-03-01:
Solved by building mpp from source and installing.
Some file was missing from the distro install which meant librockchip_mpp.so wasn't getting found.

Cheers!

closed 2023-03-01 by Fredrum

gaop921766, 2024-12-23:
> Solved by building mpp from source and installing. Some file was missing from the distro install which meant librockchip_mpp.so wasn't getting found.
> 
> Cheers!
build ffmpeg, I have some problems；ERROR: rockchip_mpp not found using pkg-config;
also need which libs? which files missed from install?  thanks

HermanChen, 2024-12-23:
https://github.com/nyanmisaka/ffmpeg-rockchip
try this

gaop921766, 2024-12-23:
> https://github.com/nyanmisaka/ffmpeg-rockchip try this

 currently, I use jefflyffmpeg for rkmpp; 
 I try ,thanks