rockchip-linux/mpp #370: 奇怪的问题：使用mpp解码，调用decode_get_frame成功，但frame数据一直为0

callanzhao, 2023-03-03:
环境是rk3399 ，linux4.4系统 （firefly的3399的板子，自带的linux ubuntu系统）.  自己曾经自己在上面编译ffmpeg，使用h264_rkmpp解码，但解码失败，ffmpeg解码接口返回-11.
遂自己写demo测试，直接按mpi_dec_test流程写代码调用mpp解码，但decode_get_frame接口返回0，但frame一直为0.
然后修改mpi_dec_test, 仅仅在里边添加printf日志。然后编译整个mpp，然后执行mpi_dec_test， ./mpi_dec_test  -i /usr/local/test.mp4 -t 7 -w 1920 -h 1080 -o ./aaa.yuv,  结果和自己写的demo一致，decode_get_frame接口返回0，frame是0. 
我感觉代码应该是没有问题的，是不是需要配置哪里呢？

callanzhao, 2023-03-03:
但我用ffmpeg 解出视频的 AVPACKET，然后转成 MppPacket，放到 decode_put_packet，然后decode_get_frame 返回值是0， frame也是0.

自己编译ffmpeg，（4.0以上，5.0以上的版本都试过） 然后通过ffplay 调用 mpp解码器 h264_rkmpp，仍然解不出数据来。但板子内自带的ffplay却可以利用h264_rkmpp解码播放。
下图是自己编译的ffmpeg，使用ffplay调用 h264_rkmpp 播放test.mp4,解码报错的日志。
[image]

也自己写过通过ffmpeg解析出AVPACKET，使用mpp_packet_init，或mpp_packet_set_data&mpp_packet_set_size，转成MppPacket, 再decode_put_packet，然后decode_get_frame 返回值是0， frame也是0.

 所以我觉的有什么定义或配置项没有配置导致的，自己编译的程序调用mpp一直失败。

1qingfeng-1, 2023-04-19:
同样问题

pipiwoaini, 2023-08-18:
相同的问题 返回-11 大佬有解决吗

stz-source, 2024-07-16:
解决了吗？

sherlockchou86, 2024-07-18:
没人解决吗？ 文件用ffmpeg解复用，传给mpp解不了

HermanChen, 2024-07-18:
https://github.com/nyanmisaka/ffmpeg-rockchip
ffmpeg 对接 mpp 参考这个大佬的