rockchip-linux/mpp #375: RGA处理解码器中数据问题

lawaarch, 2023-03-28:
怎么将MPPfram中的数据映射到RGA中，上个issue为我遇见的具体的问题

HermanChen, 2023-03-28:
MppFrame 里带的 MppBuffer 可以 get fd 出来，这个 fd 是 dmabuf，可以导入到 rga 里去使用

lawaarch, 2023-03-28:
> MppFrame 里带的 MppBuffer 可以 get fd 出来，这个 fd 是 dmabuf，可以导入到 rga 里去使用

我确实是这么整的 
  RK_U32 buf_size = mpp_frame_get_buf_size(decoder->frame);

  int fd=mpp_buffer_get_fd(buffer);

    /*--------------RGA----------------*/
    src_handle = importbuffer_fd(fd, buf_size);

    dst_handle = importbuffer_virtualaddr(yuv_buff, buf_size);

    if (src_handle == 0 || dst_handle == 0) {
        printf("importbuffer failed!\n");
        return -1;
    }
    
    src_img = wrapbuffer_handle(src_handle, width, height, fmt);
    dst_img = wrapbuffer_handle(dst_handle, width, height, fmt);

    ret = imcheck(src_img, dst_img, {}, {});
    if (IM_STATUS_NOERROR != ret) 
    {
        printf("%d, check error! %s", __LINE__, imStrError((IM_STATUS)ret));
        return -1;
    }

    ret = imcopy(src_img, dst_img);

报错： src has not fd and address for render

howieOfive, 2023-03-28:
我也遇到这个问题 ，所以follow一下。RK3568上， mpp解码后。拷贝解码后的的YUV数据到内存，帧率只有20fps。

lawaarch, 2023-03-28:
> 我也遇到这个问题 ，所以follow一下。RK3568上， mpp解码后。拷贝解码后的的YUV数据到内存，帧率只有20fps。

所以RGA是首选

howieOfive, 2023-03-28:
> 

> 

是的，之前3399路数不多，勉强还可以。3568就不行了

lawaarch, 2023-03-28:
> > 
> 
> > 
> 
> 是的，之前3399路数不多，勉强还可以。3568就不行了
你的也没法映射到RGA? 解决了没，我停滞好几周了

howieOfive, 2023-03-28:
> > > 
> > 
> > 
> > > 
> > 
> > 
> > 是的，之前3399路数不多，勉强还可以。3568就不行了
> > 你的也没法映射到RGA? 解决了没，我停滞好几周了

还没有。 这几天才开始看。这个issue，感觉也是比较古老了：
https://github.com/rockchip-linux/mpp/issues/80

HermanChen, 2023-03-28:
https://github.com/airockchip/librga
可以把 rga 相关的问题提到这里去，我们的rga开发者会协助回复

leitao080709, 2023-04-06:
希望mpp和rga整合在一起，至少定义都宏能够统一起来，比如YUV420用同一个类型的枚举变量

leitao080709, 2023-04-06:
> https://github.com/airockchip/librga 可以把 rga 相关的问题提到这里去，我们的rga开发者会协助回复

希望mpp和rga整合在一起，至少定义的宏能够统一起来，比如YUV420用同一个类型的枚举变量

hbiyik, 2023-05-22:
Have a look at here.
https://github.com/hbiyik/FFmpeg/blob/7612f91257a22339bfde5e3258b4b8c2671bb876/libavcodec/rkmppdec.c#L192

lawaarch, 2023-05-23:
> Have a look at here. https://github.com/hbiyik/FFmpeg/blob/7612f91257a22339bfde5e3258b4b8c2671bb876/libavcodec/rkmppdec.c#L192

First of all, thank you for your help. I haven't been working on this function recently. I will have a good look when I have time

lazylazypig, 2024-08-22:
请问有什么进展吗？我也遇到类似的问题~_~

shi156, 2024-11-13:
mppbuffer得到的指针进行缩放后无法映射到原有的mppbuffer中，类似的问题已经纠缠了几周了，请问还有人关注这个issue吗