rockchip-linux/mpp #377: mpp fills syslog with fd-vir-phy-hnd-format,  f-blend-size-rotation-col-log-mmu,  RgaBlit(1387) RGA_BLIT fail: Invalid argument on every frame

RogerHardiman, 2023-04-01:
I am using this pipeline to re-encode a video file

gst-launch-1.0  filesrc location=source.mp4 ! decodebin  ! textoverlay text="2023-04-01" ! mpph264enc ! h264parse ! filesink location=encode_mpp.264

On every frame of video, the MPP encoder writes the text below to the syslog journal causing the log file to fill up

```
fd-vir-phy-hnd-format[47, (nil), (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[44, (nil), (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
 RgaBlit(1387) RGA_BLIT fail: Invalid argument
 RgaBlit(1388) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[52, (nil), (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[44, (nil), (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
 RgaBlit(1387) RGA_BLIT fail: Invalid argument
 RgaBlit(1388) RGA_BLIT fail: Invalid argument
```

I've tried --gst-debug-level=0 but that makes do difference. I think there error is generated in the MPP level, not the GStreamer level.

How do I make MPP less verbose?

RogerHardiman, 2023-04-01:
This is on a RK3568 board from Firefly

Linux firefly 4.19.206 #369 SMP Thu Dec 16 10:17:57 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

 gst-launch-1.0 --version
GStreamer 1.16.2

gst-inspect-1.0 mpp264enc

Factory Details:
  Rank                     primary + 1 (257)
  Long-name                Rockchip Mpp H264 Encoder
  Klass                    Codec/Encoder/Video
  Description              Encode video streams via Rockchip Mpp
  Author                   Randy Li <randy.li@rock-chips.com>, Jeffy Chen <jeffy.chen@rock-chips.com>

Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.14.4
  License                  LGPL
  Source module            gst-rockchip
  Binary package           GStreamer Rockchip Plug-ins
  Origin URL               Unknown package origin

cat /root-ro/usr/lib/aarch64-linux-gnu/pkgconfig/rockchip_mpp.pc
prefix=/usr
exec_prefix=${prefix}
libdir=${prefix}/lib/aarch64-linux-gnu
includedir=${prefix}/include

Name: rockchip_mpp
Description: Rockchip Media Process Platform
Requires.private:
Version: 1.3.8
Libs: -L${libdir} -lrockchip_mpp
Libs.private:
Cflags: -I${includedir}

mpp_info_test
mpp[3792273]: mpp_info_test: normal version log:
mpp[3792273]: mpp_info: mpp version: 8a85dc5d author: Herman Chen   2022-03-14 [mpp_enc]: Fix stuck on reset async mode encoder
mpp[3792273]: mpp_info_test: history version log:
mpp[3792273]: mpp_info: mpp version history 10:
mpp[3792273]: mpp_info: 8a85dc5d author: Herman Chen   2022-03-14 [mpp_enc]: Fix stuck on reset async mode encoder  (HEAD -> master, origin/develop)
mpp[3792273]: mpp_info: c07169d5 author: Jeffy Chen    2022-03-11 [misc]: Fix a few compile warnings and errors
mpp[3792273]: mpp_info: da6654c1 author: Yandong Lin   2022-03-10 [h264d_init]: fix h264 video with 10bit parse err
mpp[3792273]: mpp_info: f02fd231 author: Herman Chen   2022-03-09 [mpp_enc]: Use meta with default value to get info
mpp[3792273]: mpp_info: 72a6118b author: Herman Chen   2022-03-09 [mpp_meta]: Add mpp_meta_get with default value
mpp[3792273]: mpp_info: 98ac9b25 author: Hongjin Li    2022-03-09 [h264d_dpb]: Fix h264 video flicker issue
mpp[3792273]: mpp_info: 22caf3f1 author: toby.zhang    2022-03-08 [hal_vepu580]: Add scene_mode to distinguish ipc and none ipc scene.
mpp[3792273]: mpp_info: 2eb15d59 author: Herman Chen   2022-03-08 [mpp_enc_cfg]: Add tuning scene mode check
mpp[3792273]: mpp_info: ff40ff43 author: Herman Chen   2022-03-07 [mpp_enc_cfg]: Add fine tuning paramter define
mpp[3792273]: mpp_info: 3dea9cc4 author: zj            2022-03-07 [mpp_soc]: rk3368 has h264 decoder with vdpu1_2160p

HermanChen, 2023-04-03:
The rga_test in mpp is quit old. Try librga to use RGA https://github.com/airockchip/librga

closed 2023-11-29 by HermanChen