rockchip-linux/mpp #378: RK3399 编码1920*1080分辨率报错

linlin4426, 2023-04-01:
@HermanChen    你好，我这边测试RK3399，通过v4l2-ctrl抓取1080P的帧是正常的，但编码一帧数据，无论是编码为H264还是编码为JPEG，1920*1080P的分辨率就报错，而1280*720、1920*1072、1920*1056都正常，请是这是哪里的原因造成的呢？

编码日志如下：

```
root@rock4b:/tmp# v4l2-ctl --verbose -d /dev/video0 --set-fmt-video=width=1920,height=1080,pixelformat='NV12' --stream-mmap=3
VIDIOC_QUERYCAP: ok
VIDIOC_G_FMT: ok
VIDIOC_S_FMT: ok
Format Video Capture Multiplanar:
        Width/Height      : 1920/1080
        Pixel Format      : 'NV12' (Y/CbCr 4:2:0)
        Field             : None
        Number of planes  : 1
        Flags             : 
        Colorspace        : Default
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Full Range
        Plane 0           :
           Bytes per Line : 1920
           Size Image     : 3110400
                VIDIOC_REQBUFS returned 0 (Success)
                VIDIOC_QUERYBUF returned 0 (Success)
                VIDIOC_QUERYBUF returned 0 (Success)
                VIDIOC_QUERYBUF returned 0 (Success)
                VIDIOC_QBUF returned 0 (Success)
                VIDIOC_QBUF returned 0 (Success)
                VIDIOC_QBUF returned 0 (Success)
[13629.426667] rkisp1 ff910000.rkisp1: can not get first iq setting in stream on
[13629.427992] rockchip-mipi-dphy-rx ff770000.syscon:mipi-dphy-rx0: stream on:1
[13629.428007] rockchip-mipi-dphy-rx: data_rate_mbps 972
                VIDIOC_STREAMON returned 0 (Success)
cap dqbuf: 0 seq:      0 bytesused: 3110400 ts: 13629.449989 (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:      1 bytesused: 3110400 ts: 13629.469981 delta: 19.992 ms (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:      2 bytesused: 3110400 ts: 13629.489933 delta: 19.952 ms (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:      3 bytesused: 3110400 ts: 13629.509968 delta: 20.035 ms (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:      4 bytesused: 3110400 ts: 13629.529986 delta: 20.018 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:      5 bytesused: 3110400 ts: 13629.549998 delta: 20.012 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:      6 bytesused: 3110400 ts: 13629.569868 delta: 19.870 ms fps: 50.05 (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:      7 bytesused: 3110400 ts: 13629.589997 delta: 20.129 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:      8 bytesused: 3110400 ts: 13629.609983 delta: 19.986 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:      9 bytesused: 3110400 ts: 13629.629996 delta: 20.013 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:     10 bytesused: 3110400 ts: 13629.649995 delta: 19.999 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:     11 bytesused: 3110400 ts: 13629.669995 delta: 20.000 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:     12 bytesused: 3110400 ts: 13629.689979 delta: 19.984 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:     13 bytesused: 3110400 ts: 13629.709979 delta: 20.000 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:     14 bytesused: 3110400 ts: 13629.729979 delta: 20.000 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:     15 bytesused: 3110400 ts: 13629.749971 delta: 19.992 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:     16 bytesused: 3110400 ts: 13629.769990 delta: 20.019 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:     17 bytesused: 3110400 ts: 13629.789989 delta: 19.999 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:     18 bytesused: 3110400 ts: 13629.809988 delta: 19.999 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 1 seq:     19 bytesused: 3110400 ts: 13629.829988 delta: 20.000 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 2 seq:     20 bytesused: 3110400 ts: 13629.849987 delta: 19.999 ms fps: 50.00 (ts-monotonic, ts-src-eof)
cap dqbuf: 0 seq:     21 bytesused: 3110400 ts: 13629.869973 delta: 19.986 ms fps: 50.00 (ts-monotonic, ts-src-eof)
root@rock4b:/tmp# 
root@rock4b:/tmp# 
root@rock4b:/tmp# 
root@rock4b:/tmp# 
root@rock4b:/tmp# 
root@rock4b:/tmp# mpi_enc_test -i /dev/video0 -o /tmp/1.jpeg -w 1920 -h 1072 -f 0 -t 8 -n 1
mpp[16699]: mpi_enc_utils: cmd parse result:
mpp[16699]: mpi_enc_utils: input  file name: /dev/video0
mpp[16699]: mpi_enc_utils: output file name: /tmp/1.jpeg
mpp[16699]: mpi_enc_utils: width      : 1920
mpp[16699]: mpi_enc_utils: height     : 1072
mpp[16699]: mpi_enc_utils: format     : 0
mpp[16699]: mpi_enc_utils: type       : 8
mpp[16699]: mpi_enc_test: mpi_enc_test start
mpp[16699]: mpi_enc_test: open camera device
[... 51 lines cut ...]
[12338.107270] mpp_vepu2 ff650000.vepu: fault addr 0xffef8500 status 4b
[12338.107313] rk_vcodec: mpp_task_dump_mem_region:1718: --- dump mem region ---
[12338.107360] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 48]: 0x00000000ffc00000, size 2f8000
[12338.107403] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 49]: 0x00000000ffc00000, size 2f8000
[12338.107446] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 50]: 0x00000000ffc00000, size 2f8000
[12338.107472] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 77]: 0x00000000ff800000, size 2f8000
[12338.107562] rk_iommu ff650800.iommu: Page fault at 0x00000000ffef8500 of type read
[12338.107616] rk_iommu ff650800.iommu: iova = 0x00000000ffef8500: dte_index: 0x3ff pte_index: 0x2f8 page_offset: 0x500
[12338.107674] rk_iommu ff650800.iommu: mmu_dte_addr: 0x0000000031486000 dte@0x0000000031486ffc: 0x30f0f001 valid: 1 pte@0x0000000030f0fbe0: 0x28835006 valid: 0 page@0x0000000000000000 flags: 0x0
[12338.107722] mpp_vepu2 ff650000.vepu: fault addr 0xffef8500 status 4b
[12338.107746] rk_vcodec: mpp_task_dump_mem_region:1718: --- dump mem region ---
[12338.107790] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 48]: 0x00000000ffc00000, size 2f8000
[12338.107833] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 49]: 0x00000000ffc00000, size 2f8000
[12338.107876] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 50]: 0x00000000ffc00000, size 2f8000
[12338.107919] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 77]: 0x00000000ff800000, size 2f8000
[12338.108018] rk_iommu ff650800.iommu: Page fault at 0x00000000ffef8500 of type read
[12338.108072] rk_iommu ff650800.iommu: iova = 0x00000000ffef8500: dte_index: 0x3ff pte_index: 0x2f8 page_offset: 0x500
[12338.108130] rk_iommu ff650800.iommu: mmu_dte_addr: 0x0000000031486000 dte@0x0000000031486ffc: 0x30f0f001 valid: 1 pte@0x0000000030f0fbe0: 0x28835006 valid: 0 page@0x0000000000000000 flags: 0x0
[12338.108178] mpp_vepu2 ff650000.vepu: fault addr 0xffef8500 status 4b
[12338.108218] rk_vcodec: mpp_task_dump_mem_region:1718: --- dump mem region ---
[12338.108244] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 48]: 0x00000000ffc00000, size 2f8000
[12338.108288] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 49]: 0x00000000ffc00000, size 2f8000
[12338.108331] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 50]: 0x00000000ffc00000, size 2f8000
[12338.108375] rk_vcodec: mpp_task_dump_mem_region:1724: reg[ 77]: 0x00000000ff800000, size 2f8000
```

linlin4426, 2023-04-01:
补充说明：内核已升级为 4.19.219~

HermanChen, 2023-04-03:
ffef8500 这个地址出现 pagefault，从 buffer 大小打印来看 reg[ 48]: 0x00000000ffc00000, size 2f8000 这个是输入原始图像的 buffer 大小，应该是输入图像的 buffer 开得不够大导致的，1920x1080 的图像 buffer 需要对齐到 1920x1088

closed 2023-04-03 by linlin4426