rockchip-linux/mpp #380: 运行mpi_dec_test解码h264裸流文件花屏

wangbinfei, 2023-04-02:
我是在toybrick板子rk3399prod上debian10上编译的mp.当我运行mpi_dect_test时，解码得到的yuv是花屏甚至绿屏。以下下是mpi_dec_test的结果
[image]

Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: input file /home/toybrick/CProjects/mympp/_video/out1.h264 size 5929586
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: cmd parse result:
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: input  file name: /home/toybrick/CProjects/mympp/_video/out1.h264
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: output file name: /home/toybrick/CProjects/mympp/_video/out1.yuv
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: width      :    0
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: height     :    0
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: type       :    7
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_utils: max frames :   10
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: mpi_dec_test start
Apr  2 14:45:21 debian10 mpp[31374]: mpp_info: mpp version: unknown mpp version for missing VCS info
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 mpi_dec_test decoder test start w 0 h 0 type 7
Apr  2 14:45:21 debian10 kernel: [18349.105688] rk_vcodec: vpu_service_ioctl:2140: error: unknown vpu service ioctl cmd 40086c01
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode_get_frame get info changed found
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decoder require buffer w:h [1280:720] stride [1280:720] buf_size 1843200
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 0
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 1
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 2
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 3
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 4
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 5
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 6
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 7
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 8
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: 0x55802a2380 decode get frame 9
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: decode 10 frames time 118 ms delay  18 ms fps 84.43
Apr  2 14:45:21 debian10 mpp[31374]: mpi_dec_test: test success max memory 7.03 MB
我运行mpp_paltform_test，得到的结果
[image]
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: kernel version: 4.4
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: ioctl  version: vcodec_service
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: 
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: chip name: rockchip,rk3399pro-toybrick-prod-linux rockchip,rk3399pro
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: 
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: chip vcodec type 0004020a
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: found vpu2 codec
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: found rkvdec decoder
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: found vpu2 encoder
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: 
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: start probing decoder device name:
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: H.264 decoder: /dev/rkvdec
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: H.265 decoder: /dev/rkvdec
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: VP9   decoder: /dev/rkvdec
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: avs+  decoder: /dev/vpu_service
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: 
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: start probing encoder device name:
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: H.264 encoder: /dev/vpu_service
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: H.265 encoder: /dev/vpu_service
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: MJPEG encoder: /dev/vpu_service
Apr  2 14:46:34 debian10 mpp[32371]: mpp_plat_test: mpp platform test done

请问大佬们，这个是哪里出错了呢，为什么运行mpi_dec_test会不成功，debian10 kernel: [18349.105688] rk_vcodec: vpu_service_ioctl:2140: error: unknown vpu service ioctl cmd 40086c01是什么严重错误码

HermanChen, 2023-04-03:
unknown vpu service ioctl cmd 40086c01 这个可以忽略，这时内核版本检测时候的一个打印，旧内核+新mpp的时候会出现
这个内核是 4.4 版本的，比较旧了。
从 log 上看，并硬件并没有报错出现，是否是数据保存方面出现了问题？

wangbinfei, 2023-04-03:
> unknown vpu service ioctl cmd 40086c01 这个可以忽略，这时内核版本检测时候的一个打印，旧内核+新mpp的时候会出现 这个内核是 4.4 版本的，比较旧了。 从 log 上看，并硬件并没有报错出现，是否是数据保存方面出现了问题？

多谢了大佬，问题解决了。之前不了解yuv，然后看了，最后意识到原来是yuvplayer没有设置正确分辨率。。。

closed 2023-04-03 by wangbinfei