rockchip-linux/mpp #381: rk3588解码线程阻塞

NCU-xiaoxiaobai, 2023-04-03:
我调用了4路解码, 此时解码线程正常运行, CPU接近满载, 一切都是正常的. 但是当我再开一路解码时, 会出现某个或多个解码线程阻塞, 解码器内部线程mpp_dec_hal_thread阻塞在mpp_dec_normal.cpp 784行, mpp_dec_parser_thread线程阻塞在mpp_dec_normal.cpp 701行
[image]
[image]

我想请教一下这种情况会是什么原因导致解码线程阻塞

HermanChen, 2023-04-03:
这应该是输出没有取走的问题

NCU-xiaoxiaobai, 2023-04-03:
每次我都是放入完整的一帧数据, 当调用get_frame第一次获取到mpp_frame后就立刻返回了.
按照你的思路, 我放入一帧数据, 解码后可能会有多帧数据, 而我只取了一帧数据(或者说输出有一帧,而我只取出了一部分),那我想请教一下有什么办法判断放入一帧压缩数据后, 输出是否已全部取出的方法

HermanChen, 2023-04-03:
可以一直取，直到取失败为止

NCU-xiaoxiaobai, 2023-04-03:
请问一下解码输出的格式一定是NV12格式

HermanChen, 2023-04-04:
走 rkvdec 的视频解码都是 nv12 输出，只有 jpeg 解码可以修改一些的输出格式

closed 2023-04-27 by NCU-xiaoxiaobai