rockchip-linux/mpp #382: 拷贝速度太慢

cuilingling, 2023-04-04:
RK瑞芯微中使用mpp进行硬件编解码时，将解码后的yuv数据通过内存拷贝函数memcpy，拷贝时发现耗时很长，大约80-100ms ，这样就会导致解码数据速度跟不上，对播放效果会有影响。

lion117, 2023-08-07:
> 图像拷贝建议用 RGA 等硬件来进行，解码器输出buffer默认为uncachable的，对cpu访问不友好

[image]
# 这是实测的CPU性能数据. 

# 拷贝数据性能开销大, RGA 实现NV12转I420P 性能开销更大.

HermanChen, 2023-08-07:
从 dma_buf 拷贝到 cpu malloc 的虚存空间，需要补全未分配的页表，重新为虚存建 iommu 表，开销是比较大的
如果一定要 cpu 拷贝的话，可以把把 buffer 开成 cacheable 的，现在 5.10 内核上 dma-heap 有支持这样的做法

lion117, 2023-08-07:
> 从 dma_buf 拷贝到 cpu malloc 的虚存空间，需要补全未分配的页表，重新为虚存建 iommu 表，开销是比较大的 如果一定要 cpu 拷贝的话，可以把把 buffer 开成 cacheable 的，现在 5.10 内核上 dma-heap 有支持这样的做法

不是一定要用CPU拷贝.   是没有看到RK还支持其他什么方式. 

[image]

HermanChen, 2023-08-07:
图像拷贝的 2D 拷贝一般用 RGA，复杂效果用 GPU，看是否熟悉 egl 使用。
从 dmabuf 与 malloc buffer 数据交互始终存在一个 buffer 导入导出时补页和建表的问题，如果有好的解决方案，也可以给 Linus 提补丁，我们很愿意接受更好的方案 ：）

> > 从 dma_buf 拷贝到 cpu malloc 的虚存空间，需要补全未分配的页表，重新为虚存建 iommu 表，开销是比较大的 如果一定要 cpu 拷贝的话，可以把把 buffer 开成 cacheable 的，现在 5.10 内核上 dma-heap 有支持这样的做法
> 
> 不是一定要用CPU拷贝. 是没有看到RK还支持其他什么方式.
> 
> [image]

lion117, 2023-08-09:
1. 使用RGA的方式拷贝数据，cpu开销为3%。整体还是偏高。
2. ffmpeg中的实现模块有使用DRM技术，这项技术能够带来多大的收益？外部具体是怎么消费数据的？
3. 关于egl的技术方案，我曾若干次咨询过你。
我所期望的是将解码器后的显存数据，用通过指针方式直接映射到opengl显存中，避免出现GPU和CPU直接的反复数据传输。
 你给过一个链接文件，但没看明白具体用法，
一点注释都没有

HermanChen, 2023-08-09:
1. RGA 还是要建表等一系列过程，可以 profile 或者用 ftrace 看一下开销在哪里
2. drm 只是 dmabuf 的一个封装器，我们内核里直接使用 dmabuf 接口，只是用户态代码用了 drm 的一点分配器接口
3. egl 的方式是把 dmabuf 导入到 egl 的内存管理体系内，可以实现零拷贝

Ryan3812, 2023-08-10:
> 1. RGA 还是要建表等一系列过程，可以 profile 或者用 ftrace 看一下开销在哪里
> 2. drm 只是 dmabuf 的一个封装器，我们内核里直接使用 dmabuf 接口，只是用户态代码用了 drm 的一点分配器接口
> 3. egl 的方式是把 dmabuf 导入到 egl 的内存管理体系内，可以实现零拷贝

你好，egl方式有相关demo吗

HermanChen, 2023-08-10:
搜索看下 eglCreateImageKHR 的介绍，看看下面的链接有没有帮助
https://registry.khronos.org/EGL/extensions/EXT/EGL_EXT_image_dma_buf_import.txt

HermanChen, 2023-11-29:
最新版本已经支持带 cache 的 buffer group，可以看下 cpu 拷贝的效率有提升多少，是否满足要求

closed 2023-11-29 by HermanChen