rockchip-linux/mpp #383: rk3588 编码packet size为0

zhaoxue726, 2023-04-06:
执行命令
 ./mpi_enc_test -i /dev/video12 -t 7 -w 1280 -h 720  -o test.h264 -n 10

输出：
mpp[150473]: mpi_enc_utils: cmd parse result:
mpp[150473]: mpi_enc_utils: input  file name: /dev/video12
mpp[150473]: mpi_enc_utils: output file name: test.h264
mpp[150473]: mpi_enc_utils: width      : 1280
mpp[150473]: mpi_enc_utils: height     : 720
mpp[150473]: mpi_enc_utils: format     : 0
mpp[150473]: mpi_enc_utils: type       : 7
mpp[150473]: mpi_enc_test: mpi_enc_test start
mpp[150473]: mpi_enc_test: open camera device
mpp[150473]: camera_source: get width 640 height 400
mpp[150473]: camera_source: get dma buf(0)-fd: 5
mpp[150473]: camera_source: get dma buf(1)-fd: 6
mpp[150473]: camera_source: get dma buf(2)-fd: 7
mpp[150473]: camera_source: get dma buf(3)-fd: 8
mpp[150473]: mpi_enc_test: new framecap ok
mpp[150473]: mpp_rt: NOT found ion allocator
mpp[150473]: mpp_rt: NOT found drm allocator
mpp[150473]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[150473]: mpi_enc_test: 0x5586369200 mpi_enc_test encoder test start w 1280 h 720 type 7
mpp[150473]: mpp_log: can not found match soc name: rockchip,rk3588-firefly-itx-3588j-mipi101-M101014-BE45-A1 rockchip,rk3588
mpp[150473]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1280:720] stride [1280:720]
mpp[150473]: h264e_api_v2: MPP_ENC_SET_RC_CFG bps 3456000 [216000 : 3672000] fps [30:30] gop 60
mpp[150473]: mpp_enc_v2: send header for set cfg change input/format
mpp[150473]: mpp_enc_v2: mode vbr bps [216000:3456000:3672000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 0    size 154
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 1    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 2    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 3    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 4    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 5    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 6    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 7    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 8    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encoded frame 9    size 0
mpp[150473]: mpi_enc_test: 0x5586369200 encode max 10 frames
mpp[150473]: mpi_enc_test: 0x5586369200 mpi_enc_test success total frame 10 bps 3696
mpp[150473]: mpp_buffer: ~MppBufferService cleaning misc group

HermanChen, 2023-04-07:
mpp[150473]: mpp_rt: NOT found ion allocator
mpp[150473]: mpp_rt: NOT found drm allocator
环境上没有 ion 或者 drm 来分配硬件内存，环境应该有点问题

HermanChen, 2023-05-25:
NOT found drm allocator
正常情况下平台都会有 drm 内存分配器，编译 mpp 的时候要带上 HAVE_DRM 标志