rockchip-linux/mpp #384: rk3588 安装mpp库并运行mpi_dec_test 的问题

danke1896, 2023-04-12:
当我们使用apt安装的mpp_test 中的mpi_dec_test 进行解码时,可以正常解码.
但使用github//rockchip-linux /mpp-release编译后的test中的mpi_dec_test解码时,汇报如下错误:
[image]
运行的系统为开发板自带的debian系统,内核为5.10.66,
运行mpp_platform_test, 输出如下:
[image]
运行mpi_runtime_test,结果如下:
[image]

danke1896, 2023-04-12:
使用mpp-develop源码编译完,运行mpi_dec_test解码时会报段错误,syslog 中输出的日志:
[image]

HermanChen, 2023-04-12:
这个看起来是使用了 dma-heap 来解码，但权限有问题，导致 dmaheap 打不开

danke1896, 2023-04-12:
已解决,是权限的问题,加完权限就可以了.

closed 2023-04-12 by danke1896