rockchip-linux/mpp #396: 解码265码流1080P时跨距问题

NCU-xiaoxiaobai, 2023-05-04:
在解码第一帧1080P 265压缩数据时,获取解码后mpp_frame的hor_stride和ver_stride, 发现二者分别是2304和1080, 而解码第一帧1080P 264压缩数据时,二者分别是1920和1088, 如下图所示.
[image]
这个问题是什么原因导致的,我如何将解码265后的跨距也变成1920和1088?

NCU-xiaoxiaobai, 2023-05-04:
265解码720P和4k, 解码后的跨距都是正常的, 唯独1080P出现 2304/1080的跨距, 而不是1920/1088

HermanChen, 2023-05-05:
主要是 264 之前的老器件，为了兼容，在 1080p 以下的都使用原尺寸，264 大于 1080p 的，以及 265，都是对齐到 256 奇数倍的 stride
这个是 hal 层里 hor_align 相关的函数里处理的。

NCU-xiaoxiaobai, 2023-05-05:
哦哦, 我还有一个问题是 跨距是 2304/1080时, 解码后mpp_frame的buffer_size是4478976, 这个怎么算出来的, 我根据跨距算不出这个值

HermanChen, 2023-05-05:
看芯片，老芯片在像素数据尾部需要加一些额外数据，所以 buffer 大小要加大，在 generate_info_set 这里可以看到计算过程

NCU-xiaoxiaobai, 2023-05-05:
我这里有一个场景, 就是持续送一帧完整的压缩数据给解码器解码, 送一帧解码出一帧, 当解码线程达到一定数量时, 偶尔会出现某个线程解码后的mpp_frame会出现错误信息, 调用get_error_info函数第一次返回256, 紧接着一直返回16, 这个是送进去的码流有问题还是硬件的问题, 使用的是rk3588板子

HermanChen, 2023-05-10:
先看看内核 log 有没有报错，解码有多少路，多大分辨率，有没有超过芯片能力上限