rockchip-linux/mpp #397: mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK); 耗时过长

itgo067, 2023-05-05:
程序是根据 mpp_dec_test.c 改写的，v4l2 读取UVC相机，mpp进行硬解。
 
解析mjpeg 的过程，查看日志，硬解时间只有2ms左右，如下图所示

[image]

但是在 `ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK); ` 这个步骤却花了40ms

[image]
[image]

这个步骤有没有什么办法能够做优化嘛？或者说能不能提供一个解决这个问题的方向？

HermanChen, 2023-05-05:
可能是 uncache 的 buffer 的数据拷贝需要比较多的时间

itgo067, 2023-05-05:
从硬件的buffer拷贝到内存的buffer 这个过程耗时比较多？2560*800的图片需要拷贝那么久嘛？