rockchip-linux/mpp #400: RV1106不想用rockit重新写一遍编码（不熟悉且rockit接口不好精细控制），用之前写好的调mpp编码有报错，能帮忙解决一下吗？

Yg-Luo, 2023-05-10:
RV1106不想用rockit重新写一遍编码（不熟悉且rockit接口不好精细控制），用之前写好的调mpp编码有报错，能帮忙解决一下吗？
[image]

Yg-Luo, 2023-05-10:
[14:59:57]=====>>[root@Rockchip:/tmp]# ./captest 
[14:59:57]=====>>[12426.473816] stream_cif_mipi_id0: open video, entity use_countt 1
[14:59:57]=====>>Get dma buf(0)-fd: 5
[14:59:57]=====>>Get dma buf(1)-fd: 6
[14:59:57]=====>>Get dma buf(2)-fd: 7
[14:59:57]=====>>Get dma buf(3)-fd: 8
[14:59:57]=====>>[12426.478655] rkcif-mipi-lvds: stream[0] start streaming
[14:59:57]=====>>[12426.479240] rockchip-mipi-csi2 ffa20000.mipi-csi2: stream on, src_sd: e8fa2234, sd_name:rockchip-csi2-dphy0
[14:59:57]=====>>[12426.480137] rockchip-mipi-csi2 ffa20000.mipi-csi2: stream ON
[14:59:57]=====>>[12426.480677] rockchip-csi2-dphy0: dphy0, data_rate_mbps 700
[14:59:57]=====>>[12426.481188] rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:1, dphy0
[14:59:57]=====>>[12426.481839] rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:1, dphy0
[14:59:57]=====>>[12426.485224] rk628-csi-v4l2 0-0050: src 1920x1080 clock:148500000
[14:59:57]=====>>[12426.485766] rk628-csi-v4l2 0-0050: dst 1920x1080 clock:148500000
[14:59:58]=====>>[Yg-Luo] ====> Open 104
[14:59:58]=====>>mpp[236]: mpp_info: mpp version: 0959a4ca author: Zhihua Wang   2023-01-31 [mpp]: fix chan start log
[14:59:58]=====>>
[14:59:58]=====>>Jan  1 03:27:06 mpp[236]: mpp_info: mpp version: 0959a4ca author: Zhihua Wang   2023-01-31 [mpp]: fix chan start log
[14:59:58]=====>>
[14:59:58]=====>>[Yg-Luo] ====> Open 112
[14:59:58]=====>>[12426.606249] mpp_vcodec: 215: MppCtxType error 0
[14:59:58]=====>>[12426.606285] 8<--- cut here ---
[14:59:58]=====>>[12426.608370] Unable to handle kernel NULL pointer dereference at virtual address 00000028
[14:59:58]=====>>[12426.609104] pgd = 5f374396
[14:59:58]=====>>[12426.609357] [00000028] *pgd=00da8835, *pte=00000000, *ppte=00000000
[14:59:58]=====>>[12426.609984] Internal error: Oops - BUG: 17 [#10] THUMB2
[14:59:58]=====>>[12426.610465] Modules linked in: 8821cu cfg80211 rockit(O) rknpu mpp_vcodec(O) rga3 phy_rockchip_csi2_dphy phy_rockchip_csi2_dphy_hw video_rkisp video_rkcif rk_dvbm
[14:59:58]=====>>[12426.611843] CPU: 0 PID: 236 Comm: captest Tainted: G      D    O      5.10.160 #1
[14:59:58]=====>>[12426.612509] Hardware name: Generic DT based system
[14:59:58]=====>>[12426.613722] PC is at mpp_vcodec_chan_control+0x12/0x1b4 [mpp_vcodec]
[14:59:58]=====>>[12426.614815] LR is at mpp_vcodec_get_chan_entry+0x2f/0x3c [mpp_vcodec]
[14:59:58]=====>>[12426.615407] pc : [<af893fd2>]    lr : [<af8931df>]    psr: 60000033
[14:59:58]=====>>[12426.615970] sp : b0719eb0  ip : 00000000  fp : 00000003
[14:59:58]=====>>[12426.616449] r10: 00000000  r9 : b09623d4  r8 : 00200007
[14:59:58]=====>>[12426.616928] r7 : 00000000  r6 : 00000000  r5 : 00000000  r4 : 00000000
[14:59:58]=====>>[12426.617516] r3 : 00000000  r2 : b03fe1fc  r1 : 40000093  r0 : 00000000
[14:59:58]=====>>[12426.618108] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment user
[14:59:58]=====>>[12426.618764] Control: 50c53c7d  Table: 00970059  DAC: 00000055
[14:59:58]=====>>[12426.619297] Process captest (pid: 236, stack limit = 0x3d89ab28)
[14:59:58]=====>>[12426.619841] Stack: (0xb0719eb0 to 0xb071a000)
[14:59:58]=====>>[12426.620247] 9ea0:                                     b0719eb0 b0719eb0 b085d800 ffde7f4f
[14:59:58]=====>>[12426.620986] 9ec0: b0719ef8 b0992180 00000009 00000000 00000000 00000000 00000000 00000000
[14:59:58]=====>>[12426.621725] 9ee0: b09623c0 aede7fdc b09623d4 af895327 b080ccc0 00200007 00000003 00200007
[14:59:58]=====>>[12426.622464] 9f00: 00000000 00000000 aede7fdc 00000000 00000018 b0d2a030 40045601 aede7f38
[14:59:58]=====>>[12426.623203] 9f20: b0992180 00000009 b0718000 b0992180 aede8c34 b00777d5 00002000 b0077dbf
[14:59:58]=====>>[12426.623940] 9f40: b080c180 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[14:59:58]=====>>[12426.624678] 9f60: 00000000 00000000 b080c180 b0719f88 b080c180 00000000 b080c180 00000000
[14:59:58]=====>>[12426.625419] 9f80: 00000000 0002d268 aede7f38 00200007 00000036 b0008424 b0718000 00000036
[14:59:58]=====>>[12426.626159] 9fa0: aede8c34 b0008201 0002d268 aede7f38 00000009 40045601 aede7f38 00000000
[14:59:58]=====>>[12426.626898] 9fc0: 0002d268 aede7f38 00200007 00000036 00000000 00000000 a6f80f70 aede8c34
[14:59:58]=====>>[12426.627638] 9fe0: a6d7a520 aede7f10 a6f52c1c a6d7a558 60000010 00000009 00000000 00000000
[14:59:58]=====>>[12426.629302] [<af893fd2>] (mpp_vcodec_chan_control [mpp_vcodec]) from [<af895327>] (vcodec_ioctl+0x1ef/0x2a0 [mpp_vcodec])
[14:59:58]=====>>[12426.630738] [<af895327>] (vcodec_ioctl [mpp_vcodec]) from [<b00777d5>] (vfs_ioctl+0x11/0x1c)
[14:59:58]=====>>[12426.631514] [<b00777d5>] (vfs_ioctl) from [<b0077dbf>] (sys_ioctl+0x73/0x47c)
[14:59:58]=====>>[12426.632179] [<b0077dbf>] (sys_ioctl) from [<b0008201>] (ret_fast_syscall+0x1/0x58)
[14:59:58]=====>>[12426.632861] Exception stack(0xb0719fa8 to 0xb0719ff0)
[14:59:58]=====>>[12426.633327] 9fa0:                   0002d268 aede7f38 00000009 40045601 aede7f38 00000000
[14:59:58]=====>>[12426.634067] 9fc0: 0002d268 aede7f38 00200007 00000036 00000000 00000000 a6f80f70 aede8c34
[14:59:58]=====>>[12426.634798] 9fe0: a6d7a520 aede7f10 a6f52c1c a6d7a558
[14:59:58]=====>>[12426.635268] Code: 460c 4690 f7ff f8ef (6a83) 4605 
[14:59:58]=====>>[12426.635846] ---[ end trace 6f28cee6df4f1b9e ]---
[14:59:58]=====>>[12426.640042] rkcif-mipi-lvds: stream[0] start stopping, total mode 0x1, cur 0x1
[... cut ...]
[14:59:58]=====>>[12426.659629] rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream_stop stream stop, dphy0
[14:59:58]=====>>[12426.660380] rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:0, dphy0
[14:59:58]=====>>[12426.661078] rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:0, dphy0
[14:59:58]=====>>[12426.664075] rkcif-mipi-lvds: stream[0] stopping finished, dma_en 0x0
[14:59:58]=====>>[12426.665786] stream_cif_mipi_id0: close video, entity use_count 0
[14:59:58]=====>>Segmentation fault

FumasterLin, 2023-05-11:
请问之前写好的mpp编码程序是基于哪个版本写的？1106需要基于mpp_interface分支开发才可以。
另外看下lsmod看下

Yg-Luo, 2023-05-11:
之前用的是RV1126_RV1109_LINUX_SDK_V2.1.0_20210512这个SDK里面自带的MPP，换成1106把代码迁移过来，MPP编码直接调用有问题。mpp_interface和之前的库区别大吗，要怎么玩

[root@Rockchip:/]# lsmod
Module                  Size  Used by    Tainted: G  
8821cu               1743107  0 
cfg80211              168970  1 8821cu
rockit                167991 10 
rknpu                  25000  0 
mpp_vcodec            387343  1 rockit
rga3                   87081  1 rockit
phy_rockchip_csi2_dphy     7930  0 
phy_rockchip_csi2_dphy_hw     9578  0 
video_rkisp           156881  1 rockit
video_rkcif           136522  1 rockit
rk_dvbm                 5813  2 mpp_vcodec,video_rkisp
[root@Rockchip:/]#

Yg-Luo, 2023-05-11:
> 请问之前写好的mpp编码程序是基于哪个版本写的？1106需要基于mpp_interface分支开发才可以。 另外看下lsmod看下
 之前调用的MPP是RV1126_RV1109_LINUX_SDK_V2.1.0_20210512这个SDK里面自带的,现在换成1106将代码迁移过来就会报这样的错误，mpp_interface与之前的mpp差距大吗？要怎么玩，谢谢！

[09:49:22]=====>>[root@Rockchip:/]# lsmod
[09:49:22]=====>>Module                  Size  Used by    Tainted: G  
[09:49:22]=====>>8821cu               1743107  0 
[09:49:22]=====>>cfg80211              168970  1 8821cu
[09:49:22]=====>>rockit                167991 10 
[09:49:22]=====>>rknpu                  25000  0 
[09:49:22]=====>>mpp_vcodec            387343  1 rockit
[09:49:22]=====>>rga3                   87081  1 rockit
[09:49:22]=====>>phy_rockchip_csi2_dphy     7930  0 
[09:49:22]=====>>phy_rockchip_csi2_dphy_hw     9578  0 
[09:49:22]=====>>video_rkisp           156881  1 rockit
[09:49:22]=====>>video_rkcif           136522  1 rockit
[09:49:22]=====>>rk_dvbm                 5813  2 mpp_vcodec,video_rkisp

FumasterLin, 2023-05-11:
mpp在rv1106 sdk里的路径在media/mpp/mpp下。
可以参考这里面的

Yg-Luo, 2023-05-11:
是SDK问题么？里面只有头文件库文件以及一些bin文件，是RV1106_RV1103_IPC_LINUX_SDK这版SDK,没看到直接调mpp_interface的相关例子可以参考，谢谢！
[image]
[image]

FumasterLin, 2023-05-11:
这块代码在sdk里没有开放，建议直接对接rockit的接口开发。rockit的文档应该也有详细说明

Yg-Luo, 2023-05-11:
> 这块代码在sdk里没有开放，建议直接对接rockit的接口开发。rockit的文档应该也有详细说明

好的，明白，谢谢！

closed 2023-05-11 by Yg-Luo