rockchip-linux/mpp #404: Bug report: color primaries, trc, colorspace are not initialized correctly when not detected.

hbiyik, 2023-05-22:
Mppframe has attributes of color primaries, trc and colorspace.

When they are not detected they are set the 0. But the 0 values in your enums are representing RESERVED/RGB values, UNSPECIFIED values are represented by 2 or non zero values.

For primaries, trc and its somehow innocent but 0 for colorspace it is evil because 0 value represents RGB. So when a frame does not know its colorspace, it is always reporting RGB.

https://github.com/hbiyik/FFmpeg/commit/7612f91257a22339bfde5e3258b4b8c2671bb876

Even though the variables above are based on ffmpeg, internal mpp enums are the same and based on ffmpeg, you should see it when you look at it, i am currently on mobile cant give link to internal mpp enums or definitions.

hbiyik, 2023-05-23:
here are the defines of the mppframe properties mentioned above:
https://github.com/rockchip-linux/mpp/blob/e025b079a2b01aadc09b0da118b9509c2f02dc6c/inc/mpp_frame.h#L122
https://github.com/rockchip-linux/mpp/blob/e025b079a2b01aadc09b0da118b9509c2f02dc6c/inc/mpp_frame.h#L96
https://github.com/rockchip-linux/mpp/blob/e025b079a2b01aadc09b0da118b9509c2f02dc6c/inc/mpp_frame.h#L74

hbiyik, 2023-09-15:
fixed in https://github.com/rockchip-linux/mpp/commit/6617e4feedb7f56ddedc731689b95dd182dfb581

closed 2023-09-15 by hbiyik

great9, 2023-11-23:
not fixed. it just sends back "unspecified"