rockchip-linux/mpp #406: 请问一下mpp支不支持BGR888数据编码成JPEG

kunknu, 2023-06-05:
我用最新的mpi_enc_test分别将一张yuv420p和一张bgr888的数据编码成jpeg图片，yuv420p的数据正常，但是bgr888编码出来的图片打不开，这个是正常的吗？但是我测试过用bgr888编码成h264，但是换到mjpeg就不行了，是否这部分模块还没补齐，需要我们先转成yuv再进行jpeg编码？

kunknu, 2023-06-06:
我根据mpp的打印，看到了mpp调用提示报错的源码，里面并没有brg888的选择分支，直接就报错了，但是还是执行下去了，我是否能在这个if的分支上增加bgr888的分支使这个流程正常，还是说在硬件设计上就不支持bgr888的数据直接编码成jpeg.
[image]
[image]
[image]

HermanChen, 2023-06-14:
这个是在什么芯片上跑的？

kunknu, 2023-06-14:
> 这个是在什么芯片上跑的？

RK3588 系统是debain

HermanChen, 2023-06-19:
RK3588 上的 jpeg 编码器是 vepu2，支持 YUV420，YUV422，RGB565 以及 32bit 的 RGBX 输入，24bit 的 RGB 是不支持的

kunknu, 2023-06-19:
> RK3588 上的 jpeg 编码器是 vepu2，支持 YUV420，YUV422，RGB565 以及 32bit 的 RGBX 输入，24bit 的 RGB 是不支持的

感谢

closed 2023-06-19 by kunknu