rockchip-linux/mpp #407: hdmi格式4K视频输入用mpph264enc进行编码封装后视频不正常

jiangfeng1992, 2023-06-08:
使用gst-launch-1.0 v4l2src device=/dev/video80 io-mode=4 ! video/x-raw,format=BGR,width=3840,height=2160,framerate=30/1 ! videoconvert ! queue ! mpph264enc ! queue ! h264parse ! matroskamux ! filesink location=./4k.mkv -e 发现封装后的视频不正常
[image]
正常应该就是一个电脑桌面，用videotestsrc测试可以正常保存视频，命令如下gst-launch-1.0 videotestsrc num-buffers=300 ！ video/x-raw,format=NV12,width=3840,height=2160,framerate=30/1 ! queue ! mpph264enc ! queue ! h264parse ! matroskamux ! filesink location=./4k.mkv -e，只预览hdmi输入gst-launch-1.0 v4l2src device=/dev/video80 io-mode=4 ! video/x-raw,format=BGR,width=3840,height=2160,framerate=30/1 !  glimagesink可以正常预览，用vlc查看视频发现视频从4k变成1280x720了，想问下是什么原因？

closed 2023-06-09 by jiangfeng1992

LazyDog-pj, 2024-03-20:
我也遇到这个情况，感觉像是mpp插件的bug，请问你解决了吗？

huochen1, 2025-01-15:
大佬们有解决吗，我发现这个好像bgr3的输入使用这个gstreamer就会很卡很卡的

LazyDog-pj, 2025-01-20:
那可能是格式转换需要消耗更多计算资源，如果cpu或GPU性能不强的话，还真的会卡顿