rockchip-linux/mpp #408: Bug fix function camera_source_init in camera_source.c ,cau  failed to open again after shutting down the devicese

xiangxud, 2023-06-14:
// mmap() the buffers into userspace memory
    for (i = 0 ; i < ctx->bufcnt; i++) {
        buf = (struct v4l2_buffer) {0};
        buf.type    = type;
        buf.memory  = V4L2_MEMORY_MMAP;
        buf.index   = i;
        struct v4l2_plane planes[FMT_NUM_PLANES];
        buf.memory = V4L2_MEMORY_MMAP;
        if (V4L2_BUF_TYPE_VIDEO_CAPTURE_MPLANE == type) {
            buf.m.planes = planes;
            buf.length = FMT_NUM_PLANES;
        }

        if (-1 == camera_source_ioctl(ctx->fd, VIDIOC_QUERYBUF, &buf)) {
            mpp_err_f("ERROR: VIDIOC_QUERYBUF\n");
            goto FAIL;
        }
//此段代码会导致关闭后重新打开失败,需要去掉
/*
        // ctx->fbuf[i].start = mmap(NULL, buf.length,
        //                           PROT_READ | PROT_WRITE, MAP_SHARED,
        //                           ctx->fd, buf.m.offset);
*/        
        if (V4L2_BUF_TYPE_VIDEO_CAPTURE_MPLANE == buf.type) {
            // tmp_buffers[n_buffers].length = buf.m.planes[0].length;
            buf_len = buf.m.planes[0].length;
            ctx->fbuf[i].start =
                mmap(NULL /* start anywhere */,
                     buf.m.planes[0].length,
                     PROT_READ | PROT_WRITE /* required */,
                     MAP_SHARED /* recommended */,
                     ctx->fd, buf.m.planes[0].m.mem_offset);
        } else {
            buf_len = buf.length;
            ctx->fbuf[i].start =
                mmap(NULL /* start anywhere */,
                     buf.length,
                     PROT_READ | PROT_WRITE /* required */,
                     MAP_SHARED /* recommended */,
                     ctx->fd, buf.m.offset);
        }

HermanChen, 2023-06-20:
会导致什么样的失败 log 呢，能具体描述一下么？

xiangxud, 2023-06-22:
多了一个绑定，会导致unmmap没有完全解绑，并且带入解绑时的buffer长度不一定正确mplane模式，也会导致unmmap解绑不完整，在第二次打开并mmap时一定会失败