rockchip-linux/mpp #409: rv1109 mpp出现timeout异常

AlfredLeng, 2023-06-19:
1109上突然抛出这个异常，平时正常用着的
```bash
[361176.181635] mpp_task_timeout_work:328: task 29f3917a processing time out!
[361176.181678] mpp_rkvenc ffbb0000.rkvenc: resetting...
[361176.181835] mpp_rkvenc ffbb0000.rkvenc: reset done
[361176.182013] rga: Rga sync pid 568 wait 1 task done timeout
[361176.182025] rga: interrupt = 704 status = 100
```
请问一下通常是什么原因造成的？会不会卡死系统？

AlfredLeng, 2023-06-19:
> 这个不太对，正常编码器不应该报出 timeout 来的，会卡死系统

能定位到大概的问题点或者有什么排查方向吗，还是mpp的版本问题，他出现的也没啥规律，流的配置上只使用了2路的vi->rga->venc的配置，负载也没满

```bash
[root@RV1126_RV1109:/userdata]# cat /proc/mpp_service/session_summary 
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|2d1375bf|  RKVENC|    1280|     720|    hevc|      15|      15|     vbr|  540000|      15|    8.05|    main|
------------------------------------------------------------------------------------------------------------
| session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
|d2ed3e78|   VEPU2|    1280|     720|   mjpeg|      30|      30|   fixqp|       0|       0|    6.89|    base|

```

```bash
[root@RV1126_RV1109:/userdata]# ./rk-msch-probe -c rv1109

V1.40_20221215

2kijec4hi======================================================================================================

ddr freq: 924Mhz CH0:
probes statistics:
                        master  bw(MB/s)  bw prorated(%)  utilization(%)
                           cpu    394.47          14.00%          10.67%
                           npu    266.95           9.48%           7.22%
     isp,ispp,cif,cif_lite,iep   1846.07          65.53%          49.95%
                 other_masters    309.53          10.99%           8.37%
                         total   2817.01         100.00%          76.22%
                theoretical bw   3696.00

-------------------------------------------------------------------------------------------------
    recorded load: max 2862.93MB/s(77.46%), min 2862.93MB/s(77.46%), avg 2862.93MB/s(77.46%)
         ddr load: 2862.93MB/s(77.46%) [RD:1764.09MB/s(47.73%), WR:1098.84MB/s(29.73%)
ACT(access : active): 12.91, srex:0.00%, pdex:0.01%, clkstp:0.00%, lp:0.01%]

-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=

2kijec4hi======================================================================================================
```

HermanChen, 2023-06-19:
这个最好挺到公司 redmine 上找产品部的人先看下，mpp 只个子系统，系统环境相关的问题不太好判断
试试更新 mpp 库到最新版本看看还会不会出现，如果还出现的话，比较大概率是内核驱动相关的问题

closed 2023-06-21 by AlfredLeng

githubabin, 2023-11-21:
> > 这个不太对，正常编码器不应该报出 timeout 来的，会卡死系统
> 
> 能定位到大概的问题点或者有什么排查方向吗，还是mpp的版本问题，他出现的也没啥规律，流的配置上只使用了2路的vi->rga->venc的配置，负载也没满
> 
> ```shell
> [root@RV1126_RV1109:/userdata]# cat /proc/mpp_service/session_summary 
> ------------------------------------------------------------------------------------------------------------
> | session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
> |2d1375bf|  RKVENC|    1280|     720|    hevc|      15|      15|     vbr|  540000|      15|    8.05|    main|
> ------------------------------------------------------------------------------------------------------------
> | session|  device|   width|  height|  format|  fps_in| fps_out| rc_mode| bitrate|gop_size|fps_calc| profile|
> |d2ed3e78|   VEPU2|    1280|     720|   mjpeg|      30|      30|   fixqp|       0|       0|    6.89|    base|
> ```
> 
> ```shell
> [root@RV1126_RV1109:/userdata]# ./rk-msch-probe -c rv1109
> 
> V1.40_20221215
> 
> 2kijec4hi======================================================================================================
> 
> ddr freq: 924Mhz CH0:
> probes statistics:
>                         master  bw(MB/s)  bw prorated(%)  utilization(%)
>                            cpu    394.47          14.00%          10.67%
>                            npu    266.95           9.48%           7.22%
>      isp,ispp,cif,cif_lite,iep   1846.07          65.53%          49.95%
>                  other_masters    309.53          10.99%           8.37%
>                          total   2817.01         100.00%          76.22%
>                 theoretical bw   3696.00
> 
> -------------------------------------------------------------------------------------------------
>     recorded load: max 2862.93MB/s(77.46%), min 2862.93MB/s(77.46%), avg 2862.93MB/s(77.46%)
>          ddr load: 2862.93MB/s(77.46%) [RD:1764.09MB/s(47.73%), WR:1098.84MB/s(29.73%)
> ACT(access : active): 12.91, srex:0.00%, pdex:0.01%, clkstp:0.00%, lp:0.01%]
> 
> -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-
> =-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
> 
> 2kijec4hi======================================================================================================
> ```
> 
> ```shell
> [root@RV1126_RV1109:/userdata]# cat /proc/mpp_service/version 
> 4cdfd254f099 author: Yandong Lin 2021-02-21 video: rockchip: mpp: Fix irq state err.
> ```

./rk-msch-probe哪里可以获取

AlfredLeng, 2023-11-23:
> ./rk-msch-probe哪里可以获取

这个是rk查看DDR带宽的工具，SDK里面没有可以找一下FAE拿