rockchip-linux/mpp #411: This file crahes the  hevc decoder

hbiyik, 2023-06-25:
[bb_hevc_rkfail.hevc.zip](https://github.com/rockchip-linux/mpp/files/11861127/bb_hevc_rkfail.hevc.zip)

@HermanChen @JeffyCN 
whenever this file is pushed to decoder, it crashes with below messages in the dmesg. I think there is something wrong here, it plays fine with the libx265

```
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: input file out.hevc size 8635738
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: cmd parse result:
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: input  file name: out.hevc
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: output file name:
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: width      : 1920
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: height     : 1080
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: type       : 16777220
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_utils: max frames :    0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: mpi_dec_test start
Jun 26 00:33:04 alarm mpp[187160]: mpp_info: mpp version: e025b079 author: Johnson Ding  2023-05-18 [hal_jpege1]: Fix spliting ECS not work problem
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 mpi_dec_test decoder test start w 1920 h 1080 type 16777220
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode_get_frame get info changed found
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decoder require buffer w:h [1280:720] stride [1280:720] buf_size 1658880
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 0 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 1 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 2 err 1 discard 0
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc48100.rkvdec-core: resetting...
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc48100.rkvdec-core: reset done
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc38100.rkvdec-core: resetting...
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc38100.rkvdec-core: reset done
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc48100.rkvdec-core: resetting...
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc48100.rkvdec-core: reset done
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc38100.rkvdec-core: resetting...
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc38100.rkvdec-core: reset done
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc48100.rkvdec-core: resetting...
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc48100.rkvdec-core: reset done
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc38100.rkvdec-core: resetting...
Jun 26 00:33:04 alarm kernel: mpp_rkvdec2 fdc38100.rkvdec-core: reset done
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 3 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 4 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 5 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 6 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 7 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 8 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 9 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 10 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 11 err 1 discard 0
Jun 26 00:33:04 alarm mpp[187160]: mpi_dec_test: 0x55c319f300 decode get frame 12 err 1 discard 0

```

Sherman86, 2023-06-29:
1. bb_hevc_rkfail.hevc has redundancy VPS, SPS and PPS storing error message. It results in mpi_dec_test decoding failure.
[bb_hevc_rkfail_amend.zip](https://github.com/rockchip-linux/mpp/files/11903314/bb_hevc_rkfail_amend.zip)
2. mpi_dec_test can only decode streams without encapsulation.

Sherman86, 2023-06-30:
Notice that the streams are encoded by RK encoder. Please check the encode process, which may be the source of error.

hbiyik, 2023-06-30:
@Sherman86 yes, i think i have a good idea why this is happening. Thanks for the explanation. Ill verify it next week when i have acces to my environment.

hbiyik, 2023-07-05:
@Sherman86 i am still confused about one thing, the attached file is also including duplicate SPS/PPS/VPS packets as well but it does not crash the decoder. Is it really because SPS/VPS/PPS is redundant or some other reason?

[image]
[test2.hevc.zip](https://github.com/rockchip-linux/mpp/files/11962207/test2.hevc.zip)

Sherman86, 2023-07-06:
@hbiyik It crashes the decoder that VPS, SPS and PPS **storing error message**.  The previously provided stream is indicated to 720p in the second SPS of GOP but it is actually 1080p. Only the latter SPS will take effect in decorder because of in-order parse.

hbiyik, 2023-07-07:
@Sherman86 ok now fully understood the problem. And it is happening when Encoder is HEVC, and MPP_ENC_GET_HDR_SYNC must be sent before MPP_ENC_SET_CFG, and the config must have a weightxheight different than the default 1280x720.

When i send the MPP_ENC_GET_HDR_SYNC after the config is set with MPP_ENC_SET_CFG the problem is solved. This does not happen with h264 though.

hbiyik, 2023-07-14:
issue completely resolved closing myself, thanks for the support

closed 2023-07-14 by hbiyik

hbiyik, 2023-09-15:
issue is also fixed in mpp as at https://github.com/rockchip-linux/mpp/commit/e2df27b29f233d2fe0e7cd8450b7a143f9bd69a3