rockchip-linux/mpp #415: [rga]【RV1126】YUV420SP NV12的数据转成RGB格式，提示不支持。

wddnzjwd, 2023-07-11:
我想要把YUV420SP NV12（数据来源：用mpp将jpg解码，解码输出格式为默认得到）的数据转成RGB格式，但却提示不支持：

[image]

芯片：RV1126
RGA版本：rga_api version 1.3.2_[0] (1b4d29d build: 2022-05-26 11:11:29 base: )
系统：Debian

相关代码设置部分：
 rga_src = wrapbuffer_virtualaddr(yuv_buf, 1920, 1080, RK_FORMAT_YCbCr_420_SP);
 rga_src.format = RK_FORMAT_YCbCr_420_SP;

 rga_dst = wrapbuffer_virtualaddr(rgb_buf, 1920, 1080, RK_FORMAT_RGB_888);
 rga_dst.format = RK_FORMAT_RGB_888;

 ret = imcheck(rga_src, rga_dst, src_rect, dst_rect);
 ret = imcvtcolor(rga_src, rga_dst, rga_src.format, rga_dst.format);

请问这是真的不支持yuv420SP NV12格式吗？但我看截图上下面写着支持yuv420呀，这是什么原因呢？还是芯片或者rga版本不匹配呢？

HermanChen, 2023-07-12:
这个问题需要找 rga 相关的人问下，应该是可以支持的，mpp 的 rga test case 很旧了，找下 librga 仓库，问下维护者

closed 2023-07-12 by HermanChen

wddnzjwd, 2023-07-12:
好的，感谢回答