rockchip-linux/mpp #417: VP8 encoder with yuv422p segfaults

hbiyik, 2023-07-14:
always reproducable with below.

mpi_enc_test -i test.yuv -w 1920 -h 1080 -t 9 -f 6 test_yuv422p.yuv

i think having YUV422P is accepted as config, but since the HAL does not accept yuv422p, there should have been some plane conversion there, or just simply not accept such config in the first place.

here is the stack trace:
Jul 14 17:55:55 alarm systemd-coredump[5246]: [🡕] Process 5242 (mpi_enc_test) of user 1000 dumped core.
                                              
```
  Stack trace of thread 5244:
  #0  0x0000007f8cf22760 hal_vp8e_enc_strm_code (librockchip_mpp.so.1 + 0x162760)
  #1  0x0000007f8cf21818 hal_vp8e_vepu2_gen_regs_v2 (librockchip_mpp.so.1 + 0x161818)
  #2  0x0000007f8cdf93bc mpp_enc_normal (librockchip_mpp.so.1 + 0x393bc)
  #3  0x0000007f8cc80aec start_thread (libc.so.6 + 0x80aec)
  #4  0x0000007f8ccea5dc thread_start (libc.so.6 + 0xea5dc)
  
  Stack trace of thread 5242:
  #0  0x0000007f8cc7d2d8 __futex_abstimed_wait_common64 (libc.so.6 + 0x7d2d8)
  #1  0x0000007f8cc824b8 __pthread_clockjoin_ex (libc.so.6 + 0x824b8)
  #2  0x000000555c3378c4 enc_test_multi (mpi_enc_test + 0x78c4)
  #3  0x000000555c3351a8 main (mpi_enc_test + 0x51a8)
  #4  0x0000007f8cc27b80 __libc_start_call_main (libc.so.6 + 0x27b80)
  #5  0x0000007f8cc27c60 __libc_start_main_impl (libc.so.6 + 0x27c60)
  #6  0x000000555c335230 _start (mpi_enc_test + 0x5230)
  
  Stack trace of thread 5243:
  #0  0x0000007f8cc7d2d8 __futex_abstimed_wait_common64 (libc.so.6 + 0x7d2d8)
  #1  0x0000007f8cc7fe84 __pthread_cond_wait_common (libc.so.6 + 0x7fe84)
  #2  0x0000007f8ce1fda8 _ZN9Condition4waitEP5Mutex (librockchip_mpp.so.1 + 0x5fda8)
  #3  0x0000007f8cdec428 _ZN3Mpp9put_frameEPv (librockchip_mpp.so.1 + 0x2c428)
  #4  0x0000007f8cdef094 mpi_encode_put_frame (librockchip_mpp.so.1 + 0x2f094)
  #5  0x000000555c3366a8 test_mpp_run (mpi_enc_test + 0x66a8)
  #6  0x000000555c3372fc enc_test (mpi_enc_test + 0x72fc)
  #7  0x0000007f8cc80aec start_thread (libc.so.6 + 0x80aec)
  #8  0x0000007f8ccea5dc thread_start (libc.so.6 + 0xea5dc)
  ELF object binary architecture: AARCH64
```

HermanChen, 2023-07-17:
Thank you for report this issue : )

hbiyik, 2023-07-22:
closing the issue since its fixed

closed 2023-07-22 by hbiyik