rockchip-linux/mpp #418: mpp_buffer_get failed

EZreal-zhangxing, 2023-07-17:
`Jul 17 08:43:27 firefly mpp[6417]: mpp_dma_heap: mmap failed: Invalid argument
Jul 17 08:43:27 firefly mpp[6417]: mpp_buffer: mpp_buffer_mmap buffer 0 group 4 fd 21 map failed caller convert_frame_to_drm
Jul 17 08:43:27 firefly mpp[6417]: mpp_buffer: Assertion p->info.ptr != __null failed at mpp_buffer_get_ptr_with_caller:177
Jul 17 08:43:27 firefly mpp[6417]: mpp_buffer: mpp_buffer_get_ptr buffer 0x555c15a208 ret NULL from convert_frame_to_drm
Jul 17 08:43:27 firefly mpp[6417]: mpp_dma_heap: mmap failed: Invalid argument
Jul 17 08:43:27 firefly mpp[6417]: mpp_buffer: mpp_buffer_mmap buffer 0 group 4 fd 21 map failed caller convert_frame_to_drm
Jul 17 08:43:27 firefly mpp[6417]: mpp_dma_heap: Assertion data->fd > 0 failed at os_allocator_dma_heap_import:235
Jul 17 08:43:27 firefly mpp[6417]: mpp_buffer: Assertion fd >= 0 failed at mpp_buffer_get_fd_with_caller:193
Jul 17 08:43:27 firefly mpp[6417]: mpp_buffer: mpp_buffer_get_fd buffer 0x555c15a6a8 fd -1 from setup_vepu580_io_buf
Jul 17 08:43:27 firefly kernel: [ 2284.074419] rk_vcodec: mpp_translate_reg_address:1826: reg[  0]: 0xffffffff fd -1 failed
Jul 17 08:43:27 firefly kernel: [ 2284.074436] rk_vcodec: mpp_task_dump_mem_region:2012: --- dump mem region ---
Jul 17 08:43:27 firefly kernel: [ 2284.074443] mpp_rkvenc2 fdbd0000.rkvenc-core: no memory region mapped
Jul 17 08:43:27 firefly kernel: [ 2284.074449] rk_vcodec: mpp_process_task_default:617: alloc_task failed.
Jul 17 08:43:27 firefly kernel: [ 2284.074457] rkvenc2_wait_result:1831: session 00000000a0f5700f pending list is empty!
Jul 17 08:43:27 firefly kernel: [ 2284.074461] rk_vcodec: mpp_msgs_wait:1622: session 23 wait result ret -5
`

i use opencv recive a  Mat data from camera and create a object of mppbufferinfo 
i set the info object ptr points to mat::data.
then i send it to buffer group at external model
and i invoke mpp_buffer_get() want to get a buffer which contains  info describated above
but it seems like that memory map failed ,
 the buffer have null ptr and buffer's info ptr is also null which returns by mpp_buffer_get() 

so can you help me out

HermanChen, 2023-07-17:
看起来内核的 dma-heap 没有配置好，/dev/dma_heap/ 目录是否存在，权限是否正常

EZreal-zhangxing, 2023-07-17:
> 看起来内核的 dma-heap 没有配置好，/dev/dma_heap/ 目录是否存在，权限是否正常

感谢回复，目录存在，然后我修改了/dev/dma_heap/目录下的权限 添加了读写权限

但是仍然存在问题

`Jul 17 16:22:28 firefly kernel: [19512.462869] fdee0000.hdmirx-controller: rcv frames 
Jul 17 16:22:28 firefly mpp[10717]: mpp_dma_heap: mmap failed: Invalid argument
Jul 17 16:22:28 firefly mpp[10717]: mpp_buffer: mpp_buffer_mmap buffer 0 group 4 fd 21 map failed caller convert_frame_to_drm
Jul 17 16:22:28 firefly mpp[10717]: mpp_buffer: Assertion p->info.ptr != __null failed at mpp_buffer_get_ptr_with_caller:177
Jul 17 16:22:28 firefly kernel: [19512.500918] mpp_dma_import_fd:208: dma_buf_get fd 24 failed(-22)
Jul 17 16:22:28 firefly kernel: [19512.500938] rk_vcodec: mpp_task_attach_fd:1769: can't import dma-buf 24
Jul 17 16:22:28 firefly kernel: [19512.500944] rk_vcodec: mpp_translate_reg_address:1826: reg[  0]: 0x00000018 fd 24 failed
Jul 17 16:22:28 firefly kernel: [19512.500949] rk_vcodec: mpp_task_dump_mem_region:2012: --- dump mem region ---
Jul 17 16:22:28 firefly kernel: [19512.500958] mpp_rkvenc2 fdbd0000.rkvenc-core: no memory region mapped
Jul 17 16:22:28 firefly kernel: [19512.500967] rk_vcodec: mpp_process_task_default:617: alloc_task failed.
Jul 17 16:22:28 firefly kernel: [19512.500978] rkvenc2_wait_result:1831: session 00000000f2b4870c pending list is empty!
Jul 17 16:22:28 firefly kernel: [19512.500983] rk_vcodec: mpp_msgs_wait:1622: session 71 wait result ret -5
Jul 17 16:22:28 firefly mpp[10717]: mpp_buffer: mpp_buffer_get_ptr buffer 0x55689f0168 ret NULL from convert_frame_to_drm
`

如果我不用external 模式下的buffer_group 直接从internal 模式中创建一个buffer
然后修改他的ptr 这样可以吗？还是说必须要把外部缓存数据提交至group进行扭转

HermanChen, 2023-07-18:
external 模式效率高一点，可以控制帧数，internal 也以用
不能直接修改 ptr，创建之后一定要通过 get_ptr 来 mmap 获取地址，这个地址和 buffer 关联，不能修改
比较好的做法是直接使用这个 ptr

HermanChen, 2023-07-18:
用 mpp_buffer_test 测试看下 dma_heap 的使用是不是正常

EZreal-zhangxing, 2023-07-18:
> 
buffer_test报错
mpp_dma_heap: ioctl alloc failed for Bad file descriptor

Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_group_init leave
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_get_unused enter
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_get_unused leave
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_create enter
Jul 18 10:47:04 firefly mpp[6839]: mpp_dma_heap: ioctl alloc failed for Bad file descriptor
Jul 18 10:47:04 firefly mpp[6839]: mpp_dma_heap: os_allocator_dma_heap_alloc dma_heap_alloc failed ret -1
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_create failed to create buffer with size 1024
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_create leave
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer_test: mpp_buffer_test mpp_buffer_get mode normal failed
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_group_deinit enter
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: group   4 mode 0 type 1 ops grp release
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: group   4 mode 0 type 1 ops grp destroy
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: mpp_buffer_group_deinit leave
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer_test: mpp_buffer_test failed
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: ~MppBufferService cleaning misc group
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: group   2 mode 0 type 1 ops grp release
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: group   2 mode 0 type 1 ops grp destroy
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: group   3 mode 1 type 1 ops grp release
Jul 18 10:47:04 firefly mpp[6839]: mpp_buffer: group   3 mode 1 type 1 ops grp destroy

HermanChen, 2023-07-18:
/dev/dma_heap/目录下有哪些文件，权限是什么样的？

EZreal-zhangxing, 2023-07-18:
> 
感谢您的回复，我的文件列表和权限如下
[image]

HermanChen, 2023-07-18:
看起来比较正常，跑测试用例的时候是 root 权限么

EZreal-zhangxing, 2023-07-19:
> 看起来比较正常，跑测试用例的时候是 root 权限么

是的 用的root权限，代码分支是用的develop 分支

EZreal-zhangxing, 2023-07-26:
已解决

closed 2023-07-26 by EZreal-zhangxing

belveder79, 2023-09-25:
how did you solve this?

HermanChen, 2023-09-26:
run test case by root

belveder79, 2023-09-26:
Thx... Actually this did not work for us... root user gives a lot of kernel error messages... what worked, however, is the following
1) make non-root user part of `video` group
2) change ownership and permissions on /dev/dma_heap/* to 
```
crw-rw----  1 root video 251, 4 Dec 22  2022 cma
crw-rw----  1 root video 251, 5 Dec 22  2022 cma-uncached
crw-rw----  1 root video 251, 0 Dec 22  2022 system
crw-rw----  1 root video 251, 1 Dec 22  2022 system-dma32
crw-rw----  1 root video 251, 2 Dec 22  2022 system-uncached
crw-rw----  1 root video 251, 3 Dec 22  2022 system-uncached-dma32
```
3) change ownership and permissions of `mpp_service` to 
```
crw-rw---- 1 root video 241, 0 Dec 22  2022 /dev/mpp_service
```
It is interesting, but `root` still does not work after those changes, but regular user in video group works....

gausswin, 2024-09-18:
可以分析一下代码吗