rockchip-linux/mpp #420: mpp dec 解码后提取数据效率问题

lion117, 2023-07-19:
# 需求背景
1. 低性能, 低时延
2. 要求解码器解码数据后,第一时间获取解码数据送去渲染. 减少解码帧在解码器缓存区的滞留时间

```
ret = mApi->decode_put_packet(mCtx, mPkt);    // 将packet数据输入解码器

ret = mApi->decode_get_frame(mCtx, srcFrm);   // 用户起定时器主动去读解码器中是否有解码好的数据
```

# 问题
1.  目前官方的解码demo, 都是需要用户起定时器去轮询解码器是否有解码好的数据. 这种方式性能开销相对会比较大一点. 是否有解码器主动通知回调的接口. 通知用户数据已经解码完成了?   
2. 在使用ffmpeg的rkmppdec 解码接口时, 当前帧总是依赖下一帧传入, 才能取出当前帧. 导致播放端产生至少40ms的内部延迟.  是否有更好的ffmpeg硬解封装接口. 在解码数据完成后, 第一时间将解码数据送去渲染

HermanChen, 2023-07-20:
有个 IMMEDIATE_OUT 的 control，可以把解码器配置成解码序输出，这样延迟就小了，但是存在 B 帧的码流会由于解码序和显示序不一致而出现帧序异常
1. 一个是 get_frame 配置成阻塞调用，回调接口也是有的，使用方式没有在 demo 里，用的是 cb:frm_rdy_cb 和 cb:frm_rdy_ctx 配置
2. 由于专利和版权问题，ffmpeg 插件不再支持，推荐使用 rockit 接口或者 gstreamer 来调用编解码器

lion117, 2023-07-20:
> 有个 IMMEDIATE_OUT 的 control，可以把解码器配置成解码序输出，这样延迟就小了，但是存在 B 帧的码流会由于解码序和显示序不一致而出现帧序异常
> 
> 1. 一个是 get_frame 配置成阻塞调用，回调接口也是有的，使用方式没有在 demo 里，用的是 cb:frm_rdy_cb 和 cb:frm_rdy_ctx 配置
> 2. 由于专利和版权问题，ffmpeg 插件不再支持，推荐使用 rockit 接口或者 gstreamer 来调用编解码器

1. 能提供一个demo参考学习一下用法吗? 
2. gstreamer 哪个版本集成rkmpp? 具体是在哪个文件里面实现的rkmpp api的调用

HermanChen, 2023-08-03:
参考下 mpi_mt_test，里面就双线程的处理，输出就是阻塞方式来获取的

closed 2023-11-29 by HermanChen