rockchip-linux/mpp #421: mpp 稳定版本

lion117, 2023-07-21:
# 背景
- RK3399
- 麒麟aarch64-linux4
- 程序依赖sdk ffmpeg4.2.2 和mpv

# 问题描述
在系统自带的mpp库相对陈旧.  在实际使用的场景中遇到不少bug. 虽然后续自行修复了bug, 整体成本较高.   
1. 下载官方git最新的版本, 出现mpv与最新版本的mpp不兼容的情况. 直接崩溃. 
2. 麒麟系统方不维护更新mpp版本. 
3. 官方提供可下载的 2017年稳定版本太过陈旧. 截止2023年都没有发布一个稳定的sdk.  客户对瑞芯RK系列产品质量产生很大的质疑

# 需求
1. 针对RK3399芯片, 目前mpp是否有最新的稳定版本可使用? 
2. MPP的最新稳定版本是否有配套稳定的ffmpeg 和mpv sdk 可用?

HermanChen, 2023-07-21:
由于版权和专利问题，ffmpeg 下的 mpp 插件我们也不能支持了
产品化的话，最好使用  mpp 上层封装 rockit 来支持

lion117, 2023-07-21:
rockit具体是是哪个文件或者类?

HermanChen, 2023-07-21:
是什么样的崩溃报错呢？
rockit 也是 rk 提供的一个中间件，用来做播放和录像用

lion117, 2023-07-21:
> 是什么样的崩溃报错呢？ rockit 也是 rk 提供的一个中间件，用来做播放和录像用

1.  https://github.com/HermanChen/mpp/issues/61#issuecomment-1588429787 报错提交信息
2. rockit能提供些资料看看吗?    
3. 官方mpi_dec_test demo 体验太槽了.     采用rkmppdec.c的封装形式来处理mpp接口反而更香一点

HermanChen, 2023-08-03:
1. 后面 develop 分支会用版本号来对外发布 release 分支现在没在维护了。。。。
2. 简单介绍可以看这里 https://zhuanlan.zhihu.com/p/563564705
3. mpi_dec_test 这个是我们底层接口的验证 demo，和产品化实用的接口差距比较大的
层次上看，rockit 这种是高层接口，面向应用的，mpi 接口是被 rockit 调用的低层次接口，面向专业客户定制功能用的

lion117, 2023-08-03:
# RK趣闻
###issuer: 一脚大力射门, 足球直奔RK球门, RKMPP解码怎么破
###RK DEV: 飞身将球扑出. 球飞到了RK技术支持身边 
###RK 技术支持: 将球用手抱起, 放入怀中. 足球消失了. 
###这时候RK裁判给了issuer一张红牌
###裁判: issuer缺乏各种证件,不符合RK踢球和进球流程和规范, 被罚下场

lion117, 2023-08-03:
# RK 技术支持的全流程
1. RK3399上遇到在旧版SDK严重bug,  底层SDK接口一堆 void* 实现. 没有文档, 不看源码到处都可能是坑, 
2. RK提供了非常牛逼ROCKIT, 非常牛逼
3. 怎么获取ROCKIT
4. 联系技术支持 曾雅敏
5. 转联系SARA
6. ROCKIT是我们新开发的SDK, 旧用户没有获取资格.

HermanChen, 2023-08-03:
上网找下 MPP 开发参考 http://opensource.rock-chips.com/wiki_Mpp
在 ffmpeg 使用 rkmpp 的时候有遇到的问题先反馈，提供下 log 相关信息，建议在最新版本基础上开发

HermanChen, 2023-08-03:
rockit 目前没开源，做产品的要求比较高，建议使用官方 sdk 开发，rockit 库本身是随产品 sdk 发布，走官方的客户支持路径应该会得到对应的支持
个人开发者和第三方开发者的话，目前官方支持的媒体中间件平台为 gstreamer，直接使用 mpi 的话也可以

ffmpeg 插件支持有问题的话，因为 ffmpeg 本身版本众多，不同版本间的兼容性问题也很麻烦，麻烦提供下崩溃 log 以及对应的 ffmpeg 版本信息，mpp 版本信息

ffmpeg 的流程是码流输入和图像获取一体式的接口，这种接口在使用过程中存在很多问题，mpp 的接口方式是为了兼容多种媒体中间件包括 omx，gst，codec2，以及定制中间件，请查阅 MPP 开发参考。

lion117, 2023-08-07:
> rockit 目前没开源，做产品的要求比较高，建议使用官方 sdk 开发，rockit 库本身是随产品 sdk 发布，走官方的客户支持路径应该会得到对应的支持 个人开发者和第三方开发者的话，目前官方支持的媒体中间件平台为 gstreamer，直接使用 mpi 的话也可以
>
> ffmpeg 插件支持有问题的话，因为 ffmpeg 本身版本众多，不同版本间的兼容性问题也很麻烦，麻烦提供下崩溃 log [...]

#### 1.  gstreamer是直接使用https://github.com/GStreamer/gstreamer的官方分支? 还是需要授权获取贵司珍藏82年的gstreamer分支?  封装MPI的文件是叫什么名字?
#### 2. 直接使用底层MPI接口自行封装. 遇到解码器内部缓冲帧导致高延迟问题, 解码错误帧问题, 到贵司82年的论坛上反馈问题. LOG, 代码, 截图一应俱全.   大佬帮忙分析解答一下? https://redmine.rock-chips.com/issues/431067?next_issue_id=431054

closed 2023-09-28 by HermanChen