rockchip-linux/mpp #422: mpp provides wrong hstride when decoding this hevc file

hbiyik, 2023-07-22:
always  reproducable. for a 1080 with of NV12 frame it gives 2304 hstride which is completely wrong, surrely no idea where 2304 is coming from?

This smells like a bug unless there is something wrong in the hevc file.

[image]

hevc file is at https://filetransfer.io/data-package/jD2ACyB0#link due to filesize limit

it is jusr decapsulated form of this well known test file: 
https://test-videos.co.uk/vids/bigbuckbunny/mp4/h265/1080/Big_Buck_Bunny_1080_10s_30MB.mp4

HermanChen, 2023-08-03:
This is an optimization for hardware decoder. The stride for hevc decoder is aligned to 256 odd times value.
2304 is 256 * 9 which is 1920 aligned value.
This behavior can be changed in the code.

hbiyik, 2023-08-08:
ok, when i saw the huge hstride i thought something was wrong, thanks for the explanation

hbiyik, 2023-10-02:
I am reopeening this, since this is really creating a big issue in my code.
How can i disable this?
I tried MPP_COMPAT_DEC_FBC_HDR_256_ODD=0 in mpp_compat.h but still decoder spits 256 oddaligned strides.

hbiyik, 2023-10-02:
@HermanChen

HermanChen, 2023-10-07:
The MPP_COMPAT_DEC_FBC_HDR_256_ODD is for fbc mode stride setup.
Can you setup the correct hor_stirde in your code using decoder output?
Or you code must set the hor_stride to be the same with width?

hbiyik, 2023-10-07:
do you mean something like this? if so, should i create a buffer manually of will mpp take care of it?
```c
ret = mpp_frame_init(&mppframe);
mpp_frame_set_hor_stride(mppframe, EXPECTED_STRIDE_ALIGNED_TO16);
//should I set a buffer or will mpp handle this?
codec->mpi->decode_get_frame(codec->ctx, &mppframe);
```

HermanChen, 2023-10-08:
[7d39aba.diff.zip](https://github.com/rockchip-linux/mpp/files/12840370/7d39aba.diff.zip)
Try this patch and set MPP_COMPAT_DEC_NFBC_NO_256_ODD compat value to 1

hbiyik, 2023-10-29:
@HermanChen 

I just had the chance to test this and it works generally but there are 2 problems.

1st, the pointer checks should be fixed as below, otherwise it always disable 256 oddaling (due to address check not value check.)

```patch
diff --git a/mpp/hal/rkdec/h265d/hal_h265d_com.c b/mpp/hal/rkdec/h265d/hal_h265d_com.c
index d7e8db7..40639d4 100644
--- a/mpp/hal/rkdec/h265d/hal_h265d_com.c
+++ b/mpp/hal/rkdec/h265d/hal_h265d_com.c
@@ -88,7 +88,7 @@
 
 RK_U32 hevc_hor_align(RK_U32 val)
 {
-    if (compat_ext_non_fbc_no_256_odd)
+    if (*compat_ext_non_fbc_no_256_odd)
         return MPP_ALIGN(val, 8);
 
     return MPP_ALIGN(val, 256) | 256;
diff --git a/mpp/hal/rkdec/vp9d/hal_vp9d_com.c b/mpp/hal/rkdec/vp9d/hal_vp9d_com.c
index eacd11c..59d217b 100644
--- a/mpp/hal/rkdec/vp9d/hal_vp9d_com.c
+++ b/mpp/hal/rkdec/vp9d/hal_vp9d_com.c
@@ -997,7 +997,7 @@
 
 RK_U32 vp9_hor_align(RK_U32 val)
 {
-    if (compat_ext_non_fbc_no_256_odd)
+    if (*compat_ext_non_fbc_no_256_odd)
         return MPP_ALIGN(val, 64);
 
     return MPP_ALIGN(val, 256) | 256;

```

The VP9 frames are provided with 64 aligned hstrides but, the dispayed picture has another stride becase i can see the displayed picture is broken by a wrong. May be hal also needs adaptaion?

When above fixes are applied, it works as expected for h264 & h265 frames.

closed 2024-11-15 by hbiyik