rockchip-linux/mpp #423: h264解码失败

lyj0309, 2023-07-24:
芯片：rk3328

info
```
./mpp_info_test 
mpp[12599]: mpp_info_test: normal version log:
mpp[12599]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
mpp[12599]: mpp_info_test: history version log:
mpp[12599]: mpp_info: mpp version history 10:
mpp[12599]: mpp_info: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format  (HEAD -> develop, origin/develop, origin/HEAD)
mpp[12599]: mpp_info: b2455c7b author: Johnson Ding  2023-07-14 [h264d][h265d]: Support decoding GDR stream
mpp[12599]: mpp_info: dfcdb607 author: Johnson Ding  2023-07-14 [tools]: Fix astyle not working problem
mpp[12599]: mpp_info: 28b645f2 author: Yandong Lin   2023-07-12 [h265d_parse]: do not fill syntax when parse err
mpp[12599]: mpp_info: 2531b4ed author: Yandong Lin   2023-07-12 [h265d_sei]: fix parse hdr vivid dynamic data issue
mpp[12599]: mpp_info: 897e7fd6 author: Yandong Lin   2023-07-12 [h265d_sei]: fix sei parse cause by prevention bytes
mpp[12599]: mpp_info: d180aebb author: Yandong Lin   2023-07-12 [mpp_bitread]: add new type for sei payload parse
mpp[12599]: mpp_info: cb8fa73d author: Rimon Xu      2023-07-11 [vpu_api_legacy]: Support input timeout control
mpp[12599]: mpp_info: dcedc397 author: Rimon Xu      2023-07-11 [vpu_api_legacy]: Support frame ready callback control
mpp[12599]: mpp_info: 62922f3c author: Rimon Xu      2023-07-10 [m2vd_parser] change frame period to 27MHZ
mpp[12599]: mpp_compat: id| name -- mpp compat info
mpp[12599]: mpp_compat: 0 | increase decoder fbc buffer size
mpp[12599]: mpp_compat: 1 | support encoder async input mode
```

```
ffmpeg -i output.h264 
ffmpeg version git-2023-07-16-c541ecf Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)
  configuration: --prefix=/usr/local/my/ffmpeg --enable-version3 --enable-libdrm --enable-rkmpp --enable-libx264 --enable-nonfree --enable-gpl
  libavutil      58. 14.100 / 58. 14.100
  libavcodec     60. 22.100 / 60. 22.100
  libavformat    60. 10.100 / 60. 10.100
  libavdevice    60.  2.101 / 60.  2.101
  libavfilter     9.  8.102 /  9.  8.102
  libswscale      7.  3.100 /  7.  3.100
  libswresample   4. 11.100 /  4. 11.100
  libpostproc    57.  2.100 / 57.  2.100
Input #0, h264, from 'output.h264':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264 (High 4:2:2), yuv422p(progressive), 1280x720, 25 fps, 60 tbr, 1200k tbn
At least one output file must be specified

./mpi_dec_test -w 1280 -h 720 -t 7 -i output.h264 -n 10
mpp[12706]: mpi_dec_utils: input file output.h264 size 710038
mpp[12706]: mpi_dec_utils: cmd parse result:
mpp[12706]: mpi_dec_utils: input  file name: output.h264
mpp[12706]: mpi_dec_utils: output file name: 
mpp[12706]: mpi_dec_utils: width      : 1280
mpp[12706]: mpi_dec_utils: height     :  720
mpp[12706]: mpi_dec_utils: type       :    7
mpp[12706]: mpi_dec_utils: max frames :   10
mpp[12706]: mpi_dec_test: mpi_dec_test start
mpp[12706]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
mpp[12706]: mpi_dec_test: 0xaaaacb0fb400 mpi_dec_test decoder test start w 1280 h 720 type 7
mpp[12706]: vcodec_service: open vcodec_service (null) failed
mpp[12706]: hal_h264d_api: mpp_dev_init failed ret: -1
mpp[12706]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[12706]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[12706]: mpp_dec: mpp_dec_init could not init hal
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[12706]: mpp: error found on mpp initialization
mpp[12706]: mpi_dec_test: 0xaaaacb0fb400 mpp_init failed
mpp[12706]: mpi_dec_test: test failed ret -1
mpi_dec_test[12706]: mpp_mem_pool: put_pool found 1 used buffer size 304
mpi_dec_test[12706]: mpp_mem_pool: put_pool found 4 used buffer size 224
```
只能看见失败了，不知道是什么原因

lyj0309, 2023-07-24:
./mpp_platform_test 
mpp[22426]: mpp_plat_test: kernel version: 5.10
mpp[22426]: mpp_plat_test: ioctl  version: vcodec_service
mpp[22426]: mpp_plat_test: 
mpp[22426]: mpp_plat_test: chip name: firefly,roc-rk3328-cc rockchip,rk3328
mpp[22426]: mpp_plat_test: 
mpp[22426]: mpp_plat_test: chip vcodec type 0104020a
mpp[22426]: mpp_plat_test: found vpu2 codec
mpp[22426]: mpp_plat_test: found rkvdec decoder
mpp[22426]: mpp_plat_test: found vpu2 encoder
mpp[22426]: mpp_plat_test: found h265 stand-alone encoder
mpp[22426]: mpp_plat_test: 
mpp[22426]: mpp_plat_test: start probing decoder device name:
mpp[22426]: mpp_plat_test: H.264 decoder: (null)
mpp[22426]: mpp_plat_test: H.265 decoder: (null)
mpp[22426]: mpp_plat_test: MJPEG decoder: (null)
mpp[22426]: mpp_plat_test: VP9   decoder: (null)
mpp[22426]: mpp_plat_test: avs+  decoder: (null)
mpp[22426]: mpp_plat_test: 
mpp[22426]: mpp_plat_test: start probing encoder device name:
mpp[22426]: mpp_plat_test: H.264 encoder: (null)
mpp[22426]: mpp_plat_test: H.265 encoder: (null)
mpp[22426]: mpp_plat_test: MJPEG encoder: (null)
mpp[22426]: mpp_plat_test: mpp platform test done

HermanChen, 2023-07-24:
看起来 mpp 相关的 dts 都没有打开啊
平台是 5.10 内核，但是没有 /dev/mpp_service 说明是 dts 没开，对应的内核驱动没有，上层自然就崩溃了
另外，建议使用 rockit 来开发应用，ffmpeg 因为专利与版权问题，后续没法提供支持

lyj0309, 2023-07-24:
> 看起来 mpp 相关的 dts 都没有打开啊 平台是 5.10 内核，但是没有 /dev/mpp_service 说明是 dts 没开，对应的内核驱动没有，上层自然就崩溃了 另外，建议使用 rockit 来开发应用，ffmpeg 因为专利与版权问题，后续没法提供支持

 grep -n "vpu" rk3328-roc-cc.dts
```
1203:           compatible = "rockchip,rk3328-vpu";
2796:           vpu = "/video-codec@ff350000";
2797:           vpu_mmu = "/iommu@ff350800";
```
我看dts里面有相关代码呢，这算开了吗

HermanChen, 2023-07-24:
看下 status 是不是 okay

lyj0309, 2023-07-24:
[image]
没有status，默认是okay的

HermanChen, 2023-07-24:
这个内核不是 rk 的官方内核吧？

lyj0309, 2023-07-25:
```
uname -a
Linux mkspi 5.16.20-rockchip64 #trunk SMP PREEMPT Thu Feb 23 10:12:27 CST 2023 aarch64 GNU/Linux
```
这算是吗

HermanChen, 2023-07-25:
不太清楚，是从 RK 官网下载的么？github 上有个 rockchip-linux 的仓库
https://github.com/rockchip-linux/kernel
这个是官方的

closed 2023-08-03 by HermanChen

daiaji, 2023-10-21:
主线内核似乎就没有/dev/vpu-service,我看了以下主线内核的编译配置和dts,貌似都Ok的。

very-handsome-boy, 2024-06-06:
楼主解决了吗，我用的主线内核，不知道在哪配置vpu，还是说vpu驱动是不开源的