rockchip-linux/mpp #424: 705x480 crashes the encoder

hbiyik, 2023-07-27:
always reproducable. does not happen with inputs of 704x480 or 706x480. only 705x480 crashes, some thing is misaligned somehwere

`mpp_cfg_debug=1 mpi_enc_test -i yuv420p.yuv -w 705 -h 480 -f 4 -t 7 -o yuv420p.h264`
[705-480_yuv420p.zip](https://github.com/rockchip-linux/mpp/files/12188542/705-480_yuv420p.zip)

dmesg:
```
Jul 27 22:10:52 alarm mpp[10812]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Jul 27 22:10:52 alarm mpp[10812]: mpi_enc_test: 0x7f94001940 encoder test start w 705 h 480 type 7
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:width                update 1280 -> 705
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:height               update 720 -> 480
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:hor_stride           update 1280 -> 720
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:ver_stride           update 720 -> 480
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:format               update 0 -> 4
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:mode                   keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:fps_in_flex            keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:fps_in_num             keep   30
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:fps_in_denorm          keep   1
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:fps_out_flex           keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:fps_out_num            keep   30
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:fps_out_denorm         keep   1
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set u32 rc:drop_mode              keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set u32 rc:drop_thd               update 0 -> 20
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set u32 rc:drop_gap               update 0 -> 1
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:bps_target             update 2000000 -> 1269000
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:bps_max                update 2500000 -> 1348312
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:bps_min                update 1500000 -> 79312
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:qp_init                update 26 -> -1
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:qp_max                 update 48 -> 51
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:qp_min                 update 8 -> 10
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:qp_max_i               update 0 -> 51
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:qp_min_i               update 0 -> 10
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:qp_ip                  keep   2
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 codec:type                keep   7
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 h264:profile              update 66 -> 100
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 h264:level                update 31 -> 40
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 h264:cabac_en             update 0 -> 1
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 h264:cabac_idc            keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 h264:trans8x8             update 0 -> 1
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:mirroring            keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:rotation             keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 prep:flip                 keep   0
Jul 27 22:10:52 alarm mpp[10812]: mpp_cfg: set s32 rc:gop                    keep   60
Jul 27 22:10:52 alarm mpp[10812]: mpp_enc: MPP_ENC_SET_RC_CFG bps 1269000 [79312 : 1348312] fps [30:30] gop 60
Jul 27 22:10:52 alarm mpp[10812]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [705:480] stride [720:480]
Jul 27 22:10:52 alarm mpp[10812]: mpp_enc: mode vbr bps [79312:1269000:1348312] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
Jul 27 22:10:53 alarm kernel: rk_vcodec: mpp_task_timeout_work:568: task 10813:397:5468 processing time out!
Jul 27 22:10:53 alarm kernel: rk_vcodec: task 5468 dump timing at 99715667055 us:
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: create         : 0 us
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: create end     : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: pending        : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: run            : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: timeout start  : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: run end        : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: irq            : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: timeout cancel : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: isr            : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: finish         : invalid
Jul 27 22:10:53 alarm kernel: mpp_rkvenc2 fdbd0000.rkvenc-core: resetting...
Jul 27 22:10:53 alarm kernel: mpp_rkvenc2 fdbd0000.rkvenc-core: reset done
Jul 27 22:10:53 alarm mpp[10812]: mpi_enc_test: chn 0 encoded frame 0    size 202
Jul 27 22:10:53 alarm mpp[10812]: mpi_enc_test: chn 0 encoded frame 1    size 0
Jul 27 22:10:53 alarm kernel: rk_vcodec: mpp_task_timeout_work:568: task 10813:397:5469 processing time out!
Jul 27 22:10:53 alarm kernel: rk_vcodec: task 5469 dump timing at 99716173638 us:
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: create         : 0 us
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: create end     : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: pending        : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: run            : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: timeout start  : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: run end        : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: irq            : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: timeout cancel : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: isr            : invalid
Jul 27 22:10:53 alarm kernel: rk_vcodec: timing: finish         : invalid
Jul 27 22:10:53 alarm kernel: mpp_rkvenc2 fdbd0000.rkvenc-core: resetting...
Jul 27 22:10:53 alarm kernel: mpp_rkvenc2 fdbd0000.rkvenc-core: reset done

```

HermanChen, 2023-08-03:
705 can not be encoded. try 704 or 706
YUV420 must be 2 pixel aligned.

hbiyik, 2023-08-03:
hmm not necessarily.

i can make it work for
NV12, NV16, NV24, YUV422p, 
yuv444p, yuyv422, uyvy422, bgr24, bgra, bgr0

but not for
YUV422p and YUV420p

this can also be verified with
`mpp_cfg_debug=1 mpi_enc_test -i nv12.yuv -w 705 -h 480 -f 0 -t 7 -o nv12.h264`
[nv12yuv.zip](https://github.com/rockchip-linux/mpp/files/12248555/nv12yuv.zip)
[nv12h264.zip](https://github.com/rockchip-linux/mpp/files/12248569/nv12h264.zip)

this does not crash, but outputs a file with wrong stride config. Thats because hstride is calculated wrong: 712. it is not aligned to 16 and must be 720.

```
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: cmd parse result:
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: input  file name: nv12.yuv
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: output file name: nv12.h264
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: width      : 705
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: height     : 480
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: format     : 0
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_utils: type       : 7
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: mpi_enc_test start
Aug 03 08:12:18 alarm mpp[1791]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: 0x7f9c001940 encoder test start w 705 h 480 type 7
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:width                update 1280 -> 705
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:height               update 720 -> 480
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:hor_stride           update 1280 -> 712
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:ver_stride           update 720 -> 480
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:format               keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:mode                   keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:fps_in_flex            keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:fps_in_num             keep   30
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:fps_in_denorm          keep   1
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:fps_out_flex           keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:fps_out_num            keep   30
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:fps_out_denorm         keep   1
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set u32 rc:drop_mode              keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set u32 rc:drop_thd               update 0 -> 20
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set u32 rc:drop_gap               update 0 -> 1
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:bps_target             update 2000000 -> 1269000
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:bps_max                update 2500000 -> 1348312
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:bps_min                update 1500000 -> 79312
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:qp_init                update 26 -> -1
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:qp_max                 update 48 -> 51
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:qp_min                 update 8 -> 10
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:qp_max_i               update 0 -> 51
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:qp_min_i               update 0 -> 10
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:qp_ip                  keep   2
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 codec:type                keep   7
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 h264:profile              update 66 -> 100
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 h264:level                update 31 -> 40
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 h264:cabac_en             update 0 -> 1
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 h264:cabac_idc            keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 h264:trans8x8             update 0 -> 1
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:mirroring            keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:rotation             keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 prep:flip                 keep   0
Aug 03 08:12:18 alarm mpp[1791]: mpp_cfg: set s32 rc:gop                    keep   60
Aug 03 08:12:18 alarm mpp[1791]: mpp_enc: MPP_ENC_SET_RC_CFG bps 1269000 [79312 : 1348312] fps [30:30] gop 60
Aug 03 08:12:18 alarm mpp[1791]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [705:480] stride [712:480]
Aug 03 08:12:18 alarm mpp[1791]: mpp_enc: mode vbr bps [79312:1269000:1348312] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: chn 0 encoded frame 0    size 34269   qp 18
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: chn 0 encoded frame 1    size 25052   qp 28
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: chn 0 encoded frame 2    size 17798   qp 34
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: chn 0 encoded frame 3    size 15008   qp 37
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: chn 0 encoded frame 4    size 14824   qp 39
Aug 03 08:12:18 alarm mpp[1791]: mpi_enc_test: chn 0 encoded frame 5    size 13653   qp 41
[... 62 lines cut ...]
Aug 03 08:11:54 alarm mpp[1762]: mpp_cfg: set s32 rc:qp_max                 update 48 -> 30
[... cut ...]
Aug 03 08:11:54 alarm mpp[1762]: mpp_cfg: set s32 prep:ver_stride           keep   480
Aug 03 08:11:54 alarm mpp[1762]: mpp_cfg: set s32 prep:format               update 0 -> 65548
Aug 03 08:11:54 alarm mpp[1762]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [705:480] stride [2832:480]
Aug 03 08:11:54 alarm mpp[1762]: mpp_enc: mode cbr bps [5625000:6000000:6375000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]

```

[odd_align_n12.zip](https://github.com/rockchip-linux/mpp/files/12248605/odd_align_n12.zip)

closed 2023-11-29 by HermanChen