rockchip-linux/mpp #425: Expose MppFrameStatus to the mpp_frame.h

hbiyik, 2023-07-29:
This is a feature request.

After the frame is decoded, there are people who are interested in the frame type. IE: P, I or B frame. I see this information is already a part MppFrameStatus which is a part of MppFrame, but is not exposed to mpp_frame.h

[image]

is it possible to add somehting 
`MppFrameStatus* mpp_frame_get_status(const MppFrame frame);`

to the file:
https://github.com/rockchip-linux/mpp/blob/e34f0dd1520181eda061f91883c670504432fb8f/inc/mpp_frame.h#L422

using `MPP_FRAME_ACCESSORS`?

HermanChen, 2023-08-03:
The MppFrame may not be the decoded frame for it will pass the iep deinterlace flow. Then the frame rate may change and frame I, P, B type will not be accurate.
The MppFrameStatus is used mainly in encoder part.

zoelechat, 2023-08-05:
Whatever it's called internally, might it be possible that decoder reports frame type in any way? Like most (if not all) decoders actually.

HermanChen, 2023-08-07:
It is possible to do so.

closed 2024-11-15 by hbiyik