rockchip-linux/mpp #426: rk3588 编译mpp报错 undefined reference to `__log_finite'

jiangduwang, 2023-08-03:
报错信息：
[image]
请问有解决方法吗

HermanChen, 2023-08-03:
最新的？用的什么版本的 ndk 编译还是用的 linux 环境编译？

jiangduwang, 2023-08-03:
> 最新的？用的什么版本的 ndk 编译还是用的 linux 环境编译？

github上刚拉的代码，linux环境编译：Linux rk3588 5.10.66 #116 SMP Tue Aug 30 22:20:45 CST 2022 aarch64 aarch64 aarch64 GNU/Linux

HermanChen, 2023-08-03:
奇怪，我这边编译正常，我是在  ubuntu 20 上用 aarch64-linux-gnu-gcc-4.8 交叉编译的，看起来正常
我把我手里的版本推上来，你那边更新下试试？

jiangduwang, 2023-08-03:
> 奇怪，我这边编译正常，我是在 ubuntu 20 上用 aarch64-linux-gnu-gcc-4.8 交叉编译的，看起来正常 我把我手里的版本推上来，你那边更新下试试？

我是直接修改了build/linux/aarch64/arm.linux.cross中gcc和g++的路径，执行make-Makefiles，然后make，我这个编译方法有问题吗，还有我们rk3588中ubuntu的系统是18.04

HermanChen, 2023-08-03:
https://bugs.launchpad.net/ubuntu/+source/gcc-9/+bug/1875532
会不会是这个问题？__log_finite 是数学函数，估计是 libm 不太匹配？

HermanChen, 2023-08-03:
https://otland.net/threads/otclient-problems-compiling-on-ubuntu-20.270236/
还有参考这个看看

jiangduwang, 2023-08-03:
> https://otland.net/threads/otclient-problems-compiling-on-ubuntu-20.270236/ 还有参考这个看看

请问您有在18.04上编译通过的版本吗

HermanChen, 2023-08-03:
正常都是可以编译过的，我这边的平台上没有看到有 问题的情况

closed 2023-10-13 by HermanChen