rockchip-linux/mpp #427: MPP_ERR_BUFFER_FULL

saokuazi, 2023-08-04:
您好，在使用MPP硬件解码长时间运行的情况下，出现ret = mpi->decode_put_packet(ctx, packet)执行失败，返回MPP_ERR_BUFFER_FULL，卡在循环中，视频中断，请问这种情况该如何处理。
[image]

HermanChen, 2023-08-04:
这种情况比较常见是图像没有及时取走，输出队列满了，还有可能是图像 buffer 有泄漏，分不出来内存，内部解码线程卡住，输入端也就送不进去了。

HermanChen, 2023-08-04:
把环境变量 mpp_dec_debug 配置为 0x10 然后跑，看最后 log 停在什么状态下
安卓下 setprop mpp_dec_debug 0x10
linux下 export mpp_dec_debug=0x10

saokuazi, 2023-08-04:
> 把环境变量 mpp_dec_debug 配置为 0x10 然后跑，看最后 log 停在什么状态下 安卓下 setprop mpp_dec_debug 0x10 linux下 export mpp_dec_debug=0x10

您好，尝试了该方法。程序卡在mpp[13037]: mpp_log: decode put packet failed. -1012。
[image]

HermanChen, 2023-08-04:
不是，要看 check_task_wait 的内部等待停在哪里了，看 0x200 的等待看起来是等待前一个任务完成，是不是内核驱动有报错出来，建议看下串口的 log

walletiger, 2025-08-21:
遇到 相同 问题， 输入数据有丢包，就容易出现， 楼主解决了吗？