rockchip-linux/mpp #428: mpp_buffer_group_put报错

xlwlp, 2023-08-04:
调用mpp_buffer_group_put释放资源时报错
if(enccfg.frm_buf)
{
  mpp_buffer_put(enccfg.frm_buf);
  enccfg.frm_buf = NULL;
}
if(enccfg.pkt_buf)
{
  mpp_buffer_put(enccfg.pkt_buf);
  enccfg.pkt_buf = NULL;
}
if(enccfg.buf_grp)
{
  mpp_buffer_group_put(enccfg.buf_grp);
  enccfg.buf_grp = NULL;
}
[image]
这个错误时指内存释放失败吗？流程应该没有问题吧

HermanChen, 2023-08-04:
说明有一个 buffer 没释放，这个 buffer 是在 InitEncRes 里分配的

xlwlp, 2023-08-04:
mpp_buffer_put和mpp_buffer_group_put是用来释放buffer的吧？所以说没有释放成功？

HermanChen, 2023-08-08:
mpp_buffer_put 是用来释放间个 buffer，把 buffer 释放回 group
mpp_buffer_group_put 是用来把整个 group 销毁，但是在销毁之前，要把从 group 里已经分配的 buffer 都用 mpp_buffer_put 释放掉

closed 2023-10-13 by HermanChen