rockchip-linux/mpp #429: [RK3588] rc-mode cbr not working correctly (bitrate is not fixed)

raphaelscholle, 2023-08-07:
Hello, I'm trying to get hw-encoding to work in gstreamer, when running I get a bitrate of 17Mbit, and any changes to that just doesn't actually change the output-bitrate.

```
Aug 07 00:32:30 openhd openhd[2090]: [2023-08-07 00:32:30.200] [v_gststream] [debug] Starting pipeline:[v4l2src device=/dev/video11 io-mode=auto do-timestamp=true ! video/x-raw,format=NV12, framerate=30/1 ! mpph264enc rc-mode=cbr bps=6000 width=1920 height=1080 rotation=0 gop=5 ! queue ! h264parse config-interval=-1 ! rtph264pay mtu=1440 !  appsink drop=true name=out_appsink]
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_enc: MPP_ENC_SET_RC_CFG bps 6000 [5625 : 6375] fps [30:30] gop 5
Aug 07 00:32:30 openhd mpp[2090]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1088]
Aug 07 00:32:30 openhd mpp[2090]: mpp_enc: mode cbr bps [5625:6000:6375] fps fix [30/1] -> fix [30/1] gop i [5] v [0]
root@openhd:/opt# journalctl | grep mpp
Aug 07 00:32:30 openhd openhd[2090]: [2023-08-07 00:32:30.200] [v_gststream] [debug] Starting pipeline:[v4l2src device=/dev/video11 io-mode=auto do-timestamp=true ! video/x-raw,format=NV12, framerate=30/1 ! mpph264enc rc-mode=cbr bps=6000 width=1920 height=1080 rotation=0 gop=5 ! queue ! h264parse config-interval=-1 ! rtph264pay mtu=1440 !  appsink drop=true name=out_appsink]
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_info: mpp version: 3418cca5 author: Raphael       2023-07-30 Update build_chroot.sh
Aug 07 00:32:30 openhd mpp[2090]: mpp_enc: MPP_ENC_SET_RC_CFG bps 6000 [5625 : 6375] fps [30:30] gop 5
Aug 07 00:32:30 openhd mpp[2090]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1088]
Aug 07 00:32:30 openhd mpp[2090]: mpp_enc: mode cbr bps [5625:6000:6375] fps fix [30/1] -> fix [30/1] gop i [5] v [0]
```

mpp does register the change, but the real bitrate stays 17Mbit (also tried with pbps instead of kbps, no change)

HermanChen, 2023-08-07:
bps 6000 is 6 kbps. It is too small. Multiply by 1000 will be ok.
Also please check the qp init/min/max config.

HermanChen, 2023-08-07:
@JeffyCN What is the default qp config in gst?

JeffyCN, 2023-08-07:
```
  if (!self->bps)
    self->bps =
        GST_VIDEO_INFO_WIDTH (info) * GST_VIDEO_INFO_HEIGHT (info) / 8 * fps;

  if (!self->bps || self->rc_mode == MPP_ENC_RC_MODE_FIXQP) {
    /* BPS settings are ignored */
  } else if (self->rc_mode == MPP_ENC_RC_MODE_CBR) {
    /* CBR mode has narrow bound */
    mpp_enc_cfg_set_s32 (self->mpp_cfg, "rc:bps_target", self->bps);
    mpp_enc_cfg_set_s32 (self->mpp_cfg, "rc:bps_max",
        self->bps_max ? : self->bps * 17 / 16);
    mpp_enc_cfg_set_s32 (self->mpp_cfg, "rc:bps_min",
        self->bps_min ? : self->bps * 15 / 16);
  } else if (self->rc_mode == MPP_ENC_RC_MODE_VBR) {
    /* VBR mode has wide bound */
    mpp_enc_cfg_set_s32 (self->mpp_cfg, "rc:bps_target", self->bps);
    mpp_enc_cfg_set_s32 (self->mpp_cfg, "rc:bps_max",
        self->bps_max ? : self->bps * 17 / 16);
    mpp_enc_cfg_set_s32 (self->mpp_cfg, "rc:bps_min",
        self->bps_min ? : self->bps * 1 / 16);
  }

```

raphaelscholle, 2023-08-15:
Sorry for the late answer, didn't have time to look at this problem for the last days, today I tested a little and even reduced the bitrate by 50% just to look at the effects of it. I now recorded my desktop and have openhd open for calculating the bitrate and set it (again I just set 50% of what I set up in openhd into my gstreamer pipeline)

I also have the journalctl open just to see what happens, when selecting the bitrate mpp reports the right setting, but the stream doesn't actually have the bitrate that is reported (and also not the bitrate times2)
In the end you'll notice that the bitrate (despite showing cbr) is all over the place.

https://youtu.be/nTPuaZzrq6g

HermanChen, 2023-08-16:
Can you record the output stream from the encoder? Just several seconds is enough.

JeffyCN, 2023-08-16:
default qp config of mpph264enc plugin is:
```
#define DEFAULT_PROP_LEVEL 40   /* 1080p@30fps */
#define DEFAULT_PROP_PROFILE GST_MPP_H264_PROFILE_HIGH
#define DEFAULT_PROP_QP_INIT 26
#define DEFAULT_PROP_QP_MIN 0   /* Auto */
#define DEFAULT_PROP_QP_MAX 0   /* Auto */
#define DEFAULT_PROP_QP_MAX_STEP -1     /* Auto */
```

HermanChen, 2023-08-16:
emm...
The min / max qp should not be zero. Set min to 12 max to 48 is better. Step to 1

JeffyCN, 2023-08-16:
oops, it depends on rc mode, so should be:

```
  } else if (mppenc->rc_mode == MPP_ENC_RC_MODE_CBR) {
    /* NOTE: These settings have been tuned for better quality */
    mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h264:qp_max",
        self->qp_max ? self->qp_max : 28);
    mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h264:qp_min",
        self->qp_min ? self->qp_min : 4);
    mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h264:qp_step",
        self->qp_max_step >= 0 ? self->qp_max_step : 8);
  } else if (mppenc->rc_mode == MPP_ENC_RC_MODE_VBR) {
    mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h264:qp_max",
        self->qp_max ? self->qp_max : 40);
    mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h264:qp_min",
        self->qp_min ? self->qp_min : 12);
    mpp_enc_cfg_set_s32 (mppenc->mpp_cfg, "h264:qp_step",
        self->qp_max_step >= 0 ? self->qp_max_step : 8);
  }
```

HermanChen, 2023-08-16:
qp_max : 28 is too small. Set to 48 is better.

JeffyCN, 2023-08-16:
https://github.com/JeffyCN/rockchip_mirrors/commit/b6f58385079415880f252af81b6ffcba971d1018

Update RC configs based on the newest MPP's mpi_enc_test.c

raphaelscholle, 2023-08-16:
record pipeline : gst-launch-1.0 -e v4l2src device=/dev/video11 io-mode=auto do-timestamp=true ! video/x-raw,format=NV12,framerate=30/1 ! queue ! videorate ! video/x-raw,framerate=30/1 ! queue ! mpph264enc rc-mode=cbr bps-min=4000000 bps=4000000 bps-max=4000000 width=1920 height=1080 rotation=0 gop=8 ! h264parse config-interval=-1 ! queue ! matroskamux name=mux ! filesink location=output_video.mkv

first video is with no movement in the video
second video is with movement

https://fra1.digitaloceanspaces.com/openhd-images/Downloader/Marketing/output_video.mkv

https://fra1.digitaloceanspaces.com/openhd-images/Downloader/Marketing/output_video2.mkv

both using the same pipeline, and having a quite similar playing time

HermanChen, 2023-08-17:
It seems that the max qp is limited to 28.
Also gop 8 is too small. The fps is 30 and gop is better to be 60 or higher (multiple times of fps).
And bps-min=4000000 bps=4000000 bps-max=4000000 is better to set min to 3.6Mbps, target 4Mbps, max 4.4Mbps that is 10% margin.

So the main problem is the limitation of max qp 28. Change it to 48 will be fine.

raphaelscholle, 2023-08-17:
That's quite awful, compared to low-power hardware like the pi, which can even do 0.
I want to use it to send lossy video, just sending an i-frame every 2 seconds makes it completely unusable.

HermanChen, 2023-08-17:
Do you mean gop = 0? Mpp support gop 1 for all P frame enecoding, gop 1 for all I frame encoding and gdr for no I frame encoding.
The gop length is a suggestion for I frame gap. If it is too short then too many I frame will make higher bitrate and lower quality.
And higher bitrate also means higher loss probability in transmission. So it better to balance the gop and bitrate in a certain way.

How to set the qp limit in gst by command line? @JeffyCN

JeffyCN, 2023-08-17:
try this:
https://github.com/JeffyCN/rockchip_mirrors/commit/b6f58385079415880f252af81b6ffcba971d1018

raphaelscholle, 2023-08-31:
I got the bitrate to be fixed when setting a min/max qp value on h264.
Interestingly for h265 this doesn't work, still getting 17-30Mbit there

moo-the-cow, 2023-09-24:
ok I also did some tests. I was using mpph265enc
bps was set to 1000000 (1000kbps)
rc-mode was on "cbr"

it seems that the default values for qp-min and qp-max cause issues. it worked when I set it to
qp-max = 1
qp-min = 26

HermanChen, 2023-09-25:
min = 10
max = 51
It is the correct config.

moo-the-cow, 2023-09-25:
ok but this is misleading.
gst-inspect says:
```qp-init             : Initial QP (lower value means higher quality)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 51 Default: 26```

so if I wanna go for quality isn't qp-max = 1 the best?

HermanChen, 2023-09-26:
Some hw encoder has bug on zero qp.
When qp in range from 1 ~ 12 the bitrate will be very high.
If you want lower bitrate the qp min/max should be larger.

HermanChen, 2023-09-26:
For H.264/H.265 default qp is in range from 0 to 51 and 26 is the normal init qp.

moo-the-cow, 2023-09-26:
ok great! thank you so much, this was very helpful I can live with having to set this. Works for me!

closed 2023-09-28 by HermanChen

eva93x, 2024-02-06:
Apologies for commenting on this closed issue, but I'm new to GitHub. I'm seeing this same issue, and I'd appreciate some help understanding how to fix it.

I've tried various combinations of qp-min, qp-max, bps, bps-min, bps-max and either rc-mode of vbr and cbr, but I can't come up with a configuration that actually respects the bitrate I'm trying to target.

For example, I'm currently trying "mpph265enc bps=4000000 gop=60 qp-max=51 qp-min=10 rc-mode=cbr" and the output is varying between about 8Mb/s to 12Mb/s with the resulting log: "mpp[112706]: mpp_enc: mode cbr bps [3750000:4000000:4250000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]" The calculated bitrate ranges look fine for the application, but the encoder seemingly does not respect them, not matter what I try.

@moo-the-cow Would you be able to share what settings you found worked? Thanks!

If it helps, various library packages (gstreamer, mpp) appear to be from someone called amazingfate, which appears to be the community recommended way to install mpp.

Thanks, I appreciate any help people can help with.