rockchip-linux/mpp #430: mjpeg  Specify output RGB

zzzgno5, 2023-08-08:
I learned from the previous  #187 that mjpeg decoding can adjust the output format. Is it feasible to adjust the output format to RGB? I use demo mpi_dec_test specified that the output is in rgb format and cannot succeed.

I have tried id 65536~65549
The following log is just one of the logs tested

`
hal_jpegd_rkv: setup_output_fmt unsupported output format 65536
mpi_dec_test: 0x5596ea2760 decoded frame 0
hal_jpegd_rkv: setup_output_fmt unsupported output format 65536
:mpi_dec_test: 0x5596ea2760 decoded frame 1
 hal_jpegd_rkv: setup_output_fmt unsupported output format 65536
mpi_dec_test: 0x5596ea2760 decoded frame 2
hal_jpegd_rkv: setup_output_fmt unsupported output format 65536
mpi_dec_test: 0x5596ea2760 decoded frame 3
hal_jpegd_rkv: setup_output_fmt unsupported output format 65536
mpi_dec_test: 0x5596ea2760 decoded frame 4

`
please help me

HermanChen, 2023-08-08:
Which platform are you using?
Some chip's jpeg decoder do not have the post-process capability and they can not convert the YUV420 output to RGB output.

HermanChen, 2023-08-08:
It seems that you are using rkjpegd hardware decoder. It does not have the color conversion capability.
So it is better to output YUV frame and then use rga to convert it to RGB format.

zzzgno5, 2023-08-08:
> It seems that you are using rkjpegd hardware decoder. It does not have the color conversion capability. So it is better to output YUV frame and then use rga to convert it to RGB format.

Okay, I got it. However I'm just a little confused about whether our MPP has the ability to decode into RGB?
There's a very strange situation here . Id 65537, also known as bgr565, actually succeeded.
But the rgb data is not displayed correctly when viewed in FFmpeg or Yuview

The log is as follows

mpp[246791]: mpi_dec_utils: max frames :  100
mpp[246791]: mpi_dec_test: mpi_dec_test start
mpp[246791]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[246791]: mpi_dec_test: 0x55a6f26760 mpi_dec_test decoder test start w 368 h 640 type 8
mpp[246791]: mpi_dec_test: zzzgno output format gogogogo is 65537
mpp[246791]: mpi_dec_test: zzzgno output format  is 65537
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 0
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 1
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 2
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 3
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 4
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 5
.............
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 98
mpp[246791]: mpi_dec_test: 0x55a6f26760 decoded frame 99
mpp[246791]: mpi_dec_test: decode 100 frames time 1167 ms delay   4 ms fps 85.65
mpp[246791]: mpi_dec_test: test success max memory 0.00 MB

zzzgno5, 2023-08-08:
> Which platform are you using? Some chip's jpeg decoder do not have the post-process capability and they can not convert the YUV420 output to RGB output.

RK3588

HermanChen, 2023-08-08:
It seems that 65537 is not blocked and we will fix it.
RK3588 has two jpeg hardware decoder: rkjpegd with 1080p@120fps capability and vdpu with 1080p@30fps capability.

If you fix the jpegd_mode to VPU_CLIENT_VDPU2_PP then you can get vdpu jpeg decoder with post-process.
https://github.com/rockchip-linux/mpp/blob/4fe51a6c2961656624df1f8b560e6a79e15a543f/mpp/hal/vpu/jpegd/hal_jpegd_api.c#L114

You can have a try.

zzzgno5, 2023-08-08:
> It seems that 65537 is not blocked and we will fix it. RK3588 has two jpeg hardware decoder: rkjpegd with 1080p@120fps capability and vdpu with 1080p@30fps capability.
>
> If you fix the jpegd_mode to [...]

Thanks a lot for your help!

Sherman86, 2023-08-08:
> Okay, I got it. However I'm just a little confused about whether our MPP has the ability to decode into RGB? There's a very strange situation here . Id 65537, also known as bgr565, actually [...]

1. Do you show me the command you viewed rgb565 in FFmpeg? Yuview do not support rgb565 display.
2. if possible, upload the rgb data decoded by mpp.

zzzgno5, 2023-08-08:
> 1. Do you show me the command you viewed rgb565 in FFmpeg? Yuview do not support rgb565 display.
> 2. if possible, upload the rgb data decoded by mpp.
Like this ：
ffplay -f rawvideo -pixel_format rgb565 -video_size 368x640 output.rgb

Sherman86, 2023-08-08:
> Like this ： ffplay -f rawvideo -pixel_format rgb565 -video_size 368x640 output.rgb

try:
ffplay -f rawvideo -pixel_format rgb565be -video_size 368x640 output.rgb

zzzgno5, 2023-08-08:
> try: ffplay -f rawvideo -pixel_format rgb565be -video_size 368x640 output.rgb

Success, rgb data display is normal. It's amazing~

Thank you

Sherman86, 2023-08-08:
> I learned from the previous #187 that mjpeg decoding can adjust the output format. Is it feasible to adjust the output format to RGB? I use demo mpi_dec_test specified that the output is in rgb format and cannot succeed.
> 
> I have tried id 65536~65549 The following log is just one of the logs tested

Do you really test all the formats from 65536 to 65549?

1. Hardware supports rgb565 and rgb888 output
2. rgb data display is normal.

zzzgno5, 2023-08-08:
> Do you really test all the formats from 65536 to 65549?
>
> 1. Hardware supports rgb565 and rgb888 output
> 2. rgb data display is normal.

Yes, I'm sure, only 65537 can work ~

mpp[273569]: mpi: color: id 65536 0x10000 RGB565
mpp[273569]: mpi: color: id 65537 0x10001 BGR565
mpp[273569]: mpi: color: id 65538 0x10002 RGB555
mpp[273569]: mpi: color: id 65539 0x10003 BGR555
mpp[273569]: mpi: color: id 65542 0x10006 RGB888
mpp[273569]: mpi: color: id 65543 0x10007 BGR888
mpp[273569]: mpi: color: id 65546 0x1000a ARGB8888
mpp[273569]: mpi: color: id 65547 0x1000b ABGR8888
mpp[273569]: mpi: color: id 65548 0x1000c BGRA8888
mpp[273569]: mpi: color: id 65549 0x1000d RGBA8888

zzzgno5, 2023-08-08:
ffplay -f rawvideo -pixel_format rgb565be -video_size 368x640 output.rgb

I just realized that based on the id 65537 , it should be bgr565be. However, if I use bgr565be, the color is clearly incorrect, and using rgb565be is the right one, which is obviously a bit strange

Sherman86, 2023-08-08:
[7d3b995.tar.gz](https://github.com/rockchip-linux/mpp/files/12289901/7d3b995.tar.gz)

> I just realized that based on the id 65537 , it should be bgr565be. However, if I use bgr565be, the color is clearly incorrect, and using rgb565be is the right one, which is obviously a bit strange

There is incompatibility in latest mpp version. Please try this patch.

zzzgno5, 2023-08-08:
I tried, it's okay for format 65542.

Sherman86, 2023-08-08:
[1ba284c.tar.gz](https://github.com/rockchip-linux/mpp/files/12290703/1ba284c.tar.gz)
latest patch that fixing this issue.

zzzgno5, 2023-08-09:
So when will these two versions [1ba284c.tar.gz](https://github.com/rockchip-linux/mpp/files/12290703/1ba284c.tar.gz) and [7d3b995.tar.gz](https://github.com/rockchip-linux/mpp/files/12289901/7d3b995.tar.gz) be visible in the develop or release branch?

Sherman86, 2023-08-10:
Probably this week. Waiting for review.

closed 2023-09-28 by HermanChen