rockchip-linux/mpp #431: RGA_BLIT fail: Invalid argument

yrik, 2023-09-04:
I've build the linux with https://github.com/radxa/meta-rockchip/ and trying to run a simple gstreamer pipeline with hardware acceleration. It always fails with `RGA_BLIT fail: Invalid argument` - what can be a reason for that? And how to properly enable hardware acceleration for video resizing?

```
root@rockchip-rk3588-rock-5b:~# GST_VIDEO_CONVERT_USE_RGA=1 GST_DEBUG=2 gst-launch-1.0 v4l2src device=/dev/video0 ! videoscale ! video/x-raw,width=640,height=360 ! fpsdisplaysink video-sink="kmssink plane-id=68"
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
0:00:00.446022545 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3133:gst_v4l2_object_get_nearest_size:<v4l2src0:src> Unable to try format: Invalid argument
0:00:00.446071837 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3005:gst_v4l2_object_probe_caps_for_format:<v4l2src0:src> Could not probe minimum capture size for pixelformat NV12
0:00:00.446109462 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3133:gst_v4l2_object_get_nearest_size:<v4l2src0:src> Unable to try format: Invalid argument
0:00:00.446143587 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3011:gst_v4l2_object_probe_caps_for_format:<v4l2src0:src> Could not probe maximum capture size for pixelformat NV12
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
0:00:00.446488337 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3133:gst_v4l2_object_get_nearest_size:<v4l2src0:src> Unable to try format: Invalid argument
0:00:00.446528587 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3005:gst_v4l2_object_probe_caps_for_format:<v4l2src0:src> Could not probe minimum capture size for pixelformat NV16
0:00:00.446625420 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3133:gst_v4l2_object_get_nearest_size:<v4l2src0:src> Unable to try format: Invalid argument
New clock: GstSystemClock
0:00:00.446661295 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3011:gst_v4l2_object_probe_caps_for_format:<v4l2src0:src> Could not probe maximum capture size for pixelformat NV16
0:00:00.446911837 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3133:gst_v4l2_object_get_nearest_size:<v4l2src0:src> Unable to try format: Invalid argument
0:00:00.447032587 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3005:gst_v4l2_object_probe_caps_for_format:<v4l2src0:src> Could not probe minimum capture size for pixelformat NV24
0:00:00.447102295 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3133:gst_v4l2_object_get_nearest_size:<v4l2src0:src> Unable to try format: Invalid argument
0:00:00.447186587 10998   0x55c443a460 WARN                    v4l2 gstv4l2object.c:3011:gst_v4l2_object_probe_caps_for_format:<v4l2src0:src> Could not probe maximum capture size for pixelformat NV24
0:00:00.571896558 10998   0x55c443a460 WARN          v4l2bufferpool gstv4l2bufferpool.c:1397:gst_v4l2_buffer_pool_dqbuf:<v4l2src0:pool0:src> Driver should never set v4l2_buffer.field to ANY
rga_api version 1.3.2_[0] (RGA is compiling with meson base: $PRODUCT_BASE)
 RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7f8e8f1000, (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1080, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[0, 0x7f8e848000, (nil), (nil), 0]
rect[0, 0, 640, 360, 640, 360, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
 RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7f8e8f1000, (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1080, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[0, 0x7f8e848000, (nil), (nil), 0]
rect[0, 0, 640, 360, 640, 360, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
Redistribute latency...
 RgaBlit(1356) RGA_BLIT fail: Invalid argument RgaBlit(1357) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7f8e0e1000, (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1080, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[0, 0x7f8e038000, (nil), (nil), 0]
rect[0, 0, 640, 360, 640, 360, 1792, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
```

HermanChen, 2023-09-28:
@JeffyCN

JeffyCN, 2023-10-12:
looks like something wrong with rga, please contact rga maintainer(or committer). the 3588 might need special librga and rga driver for legacy rga api.

closed 2023-10-13 by HermanChen

AshleyRoth, 2025-01-10:
i have same problem. how solved?

qin-yuhao, 2025-04-04:
同样的问题

ioncubekh, 2025-04-05:
I am facing same problem while using`mpv` with https://chromewebstore.google.com/detail/play-with-mpv/hahklcmnfgffdlchjigehabfbiigleji
```
This output the user parameters when rga call blit fail
[ffmpeg] filter: RGA blit failed: -22
[lavfi] error on filtering (-542398533)
AO: [pipewire] 192000Hz stereo 2ch double
 (+) Video --vid=1 (*) (vp9 1920x1080 25.000fps)
 (+) Audio --aid=1 --alang=eng (*) (opus 2ch 48000Hz)
File tags:
 Uploader: Hindustan Times
 Channel_URL: https://www.youtube.com/channel/UCm7lHFkt2yB_WzL67aruVBQ
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
VO: [gpu-next] 1920x1080 drm_prime[nv12]
AO: [pipewire] 192000Hz stereo 2ch double
AV: 00:00:00 / 00:05:24 (0%) A-V:  0.003 Cache: 2.1s/405KB RgaBlit(1485) RGA_BLIT fail: Invalid argument
 RgaBlit(1486) RGA_BLIT fail: Invalid argument
handl-fd-vir-phy-hnd-format[0, 77, (nil), (nil), 0, 2560]
rect[0, 0, 1920, 1080, 2304, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[2560, 0, 0, 0, 0, 0, 1]
handl-fd-vir-phy-hnd-format[0, 52, (nil), (nil), 0, 2560]
rect[0, 0, 1920, 1080, 1920, 1080, 2560, 0]
f-blend-size-rotation-col-log-mmu[2560, 0, 0, 0, 0, 0, 1]
This output the user parameters when rga call blit fail
[ffmpeg] filter: RGA blit failed: -22
[lavfi] error on filtering (-542398533)
(...) AV: 00:01:56 / 00:05:24 (36%) A-V:  0.000 Cache: 0.0s RgaBlit(1485) RGA_BLIT fail: Invalid argument
 RgaBlit(1486) RGA_BLIT fail: Invalid argument
handl-fd-vir-phy-hnd-format[0, 49, (nil), (nil), 0, 2560]
rect[0, 0, 1920, 1080, 2304, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[2560, 0, 0, 0, 0, 0, 1]
handl-fd-vir-phy-hnd-format[0, 12, (nil), (nil), 0, 2560]
rect[0, 0, 1920, 1080, 1920, 1080, 2560, 0]
f-blend-size-rotation-col-log-mmu[2560, 0, 0, 0, 0, 0, 1]
This output the user parameters when rga call blit fail
[ffmpeg] filter: RGA blit failed: -22

```