rockchip-linux/mpp #432: mpp_frame_get_buffer错误

xduzhangjiayu, 2023-09-05:
我现在只输入了一帧的码流，调用decode_put_packet和decode_get_frame是正常的，之后mpp_frame_get_width(frame)等函数也可以正常输出图像的长宽等信息。

但是我之后再调用buffer = mpp_frame_get_buffer(frame), 返回的buffer却是空的，无法拿到yuv信息，请问是什么原因呀？感谢

xduzhangjiayu, 2023-09-05:
用的平台是RK3568

HermanChen, 2023-09-05:
第一帧是 info change 帧，是反馈信息给外部分 buffer 用的，参考 mpi_dec_test

xduzhangjiayu, 2023-09-05:
`                    if (NULL == data->frm_grp) {
                        /* If buffer group is not set create one and limit it */
                        ret = mpp_buffer_group_get_internal(&data->frm_grp, MPP_BUFFER_TYPE_ION);
                        if (ret) {
                            mpp_err("%p get mpp buffer group failed ret %d\n", ctx, ret);
                            break;
                        }

                        /* Set buffer to mpp decoder */
                        ret = mpi->control(ctx, MPP_DEC_SET_EXT_BUF_GROUP, data->frm_grp);
                        if (ret) {
                            mpp_err("%p set buffer group failed ret %d\n", ctx, ret);
                            break;
                        }
                    } else {
                        /* If old buffer group exist clear it */
                        ret = mpp_buffer_group_clear(data->frm_grp);
                        if (ret) {
                            mpp_err("%p clear buffer group failed ret %d\n", ctx, ret);
                            break;
                        }
                    }

                    /* Use limit config to limit buffer count to 24 with buf_size */
                    ret = mpp_buffer_group_limit_config(data->frm_grp, buf_size, 24);
                    if (ret) {
                        mpp_err("%p limit buffer group failed ret %d\n", ctx, ret);
                        break;
                    }

                    /*
                     * All buffer group config done. Set info change ready to let
                     * decoder continue decoding
                     */
                    ret = mpi->control(ctx, MPP_DEC_SET_INFO_CHANGE_READY, NULL);
                    if (ret) {
                        mpp_err("%p info change ready failed ret %d\n", ctx, ret);
                        break;
                    }`
不好意思，想问下mpi_dec_test这部分代码是设置默认的内存分配模式是纯内部分配吗？另外，是不是第一帧的info change帧不能读出yuv数据，之后再送入的帧就可以读出yuv数据吗？感谢

HermanChen, 2023-09-05:
test 里是是半内部的方式，还是 control 了一个 buffer_group 进去

xduzhangjiayu, 2023-09-05:
OK明白，那就是说第一个info change帧无法通过mpp_frame_get_buffer拿到yuv数据？如果不行的话，该如何通过mpp_frame_get_buffer把info change帧之后的帧的yuv数据取出来呢?

HermanChen, 2023-09-28:
info change 帧的 buffer 是空的，control ext_group 或者 ready 进去的话，再 get_frame 就可以正常获取到有 buffer 的帧

closed 2023-09-28 by HermanChen