rockchip-linux/mpp #435: 咨询MPP解码block模式

xduzhangjiayu, 2023-09-07:
您好，
我现在在使用Mpp进行解码，实时的往解码器里喂码流，想要设置block模式，设置了INPUT BLOCK和OUTPUT BLOCK如下：
`	RK_U32 block = -1;

	mpi_cmd = MPP_SET_OUTPUT_BLOCK_TIMEOUT;
	param = &block;
	ret = pDecCtx_mpp->mpi->control(pDecCtx_mpp->ctx, mpi_cmd, param);
	if(ret != MPP_OK){
	  	WLog_DBG(TAG,"MPP_SET_OUTPUT_BLOCK ret=%d, ctx=%p",ret, pDecCtx_mpp->ctx);
	}

	mpi_cmd = MPP_SET_INTPUT_BLOCK_TIMEOUT;
	param = &block;
	ret = pDecCtx_mpp->mpi->control(pDecCtx_mpp->ctx, mpi_cmd, param);
	if(ret != MPP_OK){
	   	WLog_DBG(TAG,"MPP_SET_INPUT_BLOCK ret=%d, ctx=%p",ret, pDecCtx_mpp->ctx);
	}`
这样想要实现只有前一帧处理完成再会继续处理下一帧的实时模式。但是如果计算decode_put_packet和decode_get_frame两个函数的运行时间，都小于1ms，看上去每次运行只是把上一帧已经解码完的帧拿出来，而不是解码当前帧，有了延迟，达不到实时的效果。请问以上代码的control那样设置可以达到进一帧出一帧的实时效果吗？还是说我必须使用mpp的高级接口enqueue和dequeue那些才可以达到目的？

感谢解答

HermanChen, 2023-09-07:
解码输出延迟和码流输入模式也相关，按一帧送入还是按长度送入，还有码流是否有输出序和解码序不一致的情况

control IMMEDIATE_OUT 这样是解码序输出，会快一些

xduzhangjiayu, 2023-09-07:
我是用的vp9硬解码，实时每次送入一帧码流（按帧送入，没有B帧）
我设过这个IMMEDIATE_OUT为1，但是不起作用，貌似这个IMMEDIATE_OUT是只针对h264解码的？
还有一个原因也许是我用的纯内部分配模式，第一帧会走mpp_frame_get_info_change，导致第一帧无法输出显示，所以当第二帧输入时只能先把第一帧拿出来，所以这样之后一直会有一帧的延迟，导致每次调用decode_put_packet和decode_get_frame都只是拿的上一帧数据，是这样吗？困扰这个问题挺久的了，感谢楼主解答

HermanChen, 2023-09-07:
info change 帧拿到之后再去获取一次

xduzhangjiayu, 2023-09-07:
您的意思是当第一帧进入到if(mpp_frame_get_info_change)后再用decode_get_frame把第一帧拿出来吗？还是说我必须使用纯外部内存分配模式在第一帧进来之前就提前分配好buffer? 感谢~

HermanChen, 2023-09-07:
如果知道分辨率的话，最高效的方法是纯外部模式分配好，一开始就 commit 进来，info change 的帧都不会出，可以直接出第一帧

xduzhangjiayu, 2023-09-07:
明白了，感谢兄弟的解答~

RockLing-18, 2023-09-08:
> 明白了，感谢兄弟的解答~

大神，纯外部分配内存怎么设置，我这边一直报错

HermanChen, 2023-09-08:
参考下 mpp_buffer_test

RockLing-18, 2023-09-09:
> 参考下 mpp_buffer_test

您好，demo中有些疑问，代码如下：  
```
  for (i = 0; i < count; i++) {
        // can be change to different buffer allocator here
        ret = mpp_buffer_get(NULL, &normal_buffer[i], size);
        if (ret || NULL == normal_buffer[i]) {
            mpp_err("mpp_buffer_test get misc buffer failed ret %d\n", ret);
            goto MPP_BUFFER_failed;
        }

        // NOTE: setup fd / index in necessary
        commit.ptr = mpp_buffer_get_ptr(normal_buffer[i]);
        commit.fd = mpp_buffer_get_fd(normal_buffer[i]);
        commit.index = i;

        ret = mpp_buffer_commit(group, &commit);
        if (MPP_OK != ret) {
            mpp_err("mpp_buffer_test mpp_buffer_commit failed\n");
            goto MPP_BUFFER_failed;
        }
    }
```
normal_buffer[i]申请的内存还是由mppSDK控制的，我在开发中按照这样的流程去使用还是不行，解码后的RGB数据使用memcpy拷贝还是使CPU飙升

HermanChen, 2023-09-09:
这里是流程的例子，这外 dmabuf 的 buffer 可以用外部来的 buffer，可以开 cache
最好不要 cpu 访问，一定要 cpu 访问的话，开一个 buffer group，带上 MPP_BUFFER_FLAGS_CACHABLE 的 flag，然后从这个 buffer group 里 get buffer 出来，不用要 test 里从 NULL group 里获取的方式

RockLing-18, 2023-09-09:
> 这里是流程的例子，这外 dmabuf 的 buffer 可以用外部来的 buffer，可以开 cache 最好不要 cpu 访问，一定要 cpu 访问的话，开一个 buffer group，带上 MPP_BUFFER_FLAGS_CACHABLE 的 flag，然后从这个 buffer group 里 get buffer 出来，不用要 test 里从 NULL group 里获取的方式

有对应的例子的吗？不大明白。
目前的场景最后就是要把图像数据拷贝到程序内存中。
_这外 dmabuf 的 buffer 可以用外部来的 buffer，可以开 cache
最好不要 cpu 访问，一定要 cpu 访问的话，开一个 buffer group，带上 MPP_BUFFER_FLAGS_CACHABLE 的 flag，然后从这个 buffer group 里 get buffer 出来_
这里的流程有详细的吗？dmabuf是如何操作的，还有开一个 buffer group，带上 MPP_BUFFER_FLAGS_CACHABLE 的 flag这个也没有找到例子

HermanChen, 2023-11-24:
试试 test_cache 分支下 431058ad9ded69e9ff95b99040131526344dc798 这个提交里的处理
mpi_dec_test 加一下 -bufmode e 的选项来解码看看