rockchip-linux/mpp #437: 硬件编码出来的264 关键帧NAL值为6

zhang-chuanfeng, 2023-09-08:
[image]

HermanChen, 2023-09-08:
6 是 sei 包

zhang-chuanfeng, 2023-09-08:
你好默认编码264为啥没有I帧出来 是我配置的不对？

zhang-chuanfeng, 2023-09-08:
我的配置如下：
void MpiDecEncode::updateEncCfg(void) {
    RK_S32 fps_in_den = 1;
    RK_S32 fps_in_num = 30;
    RK_S32 fps_out_den = 1;
    RK_S32 fps_out_num = 30;

    RK_S32 bps = _weight * _height / 8 * (fps_out_num / fps_out_den);

    mpp_enc_cfg_set_s32(_encoder.cfg, "prep:width", _decoder.width);
    mpp_enc_cfg_set_s32(_encoder.cfg, "prep:height", _decoder.height);
    mpp_enc_cfg_set_s32(_encoder.cfg, "prep:hor_stride", _decoder.hor_stride);
    mpp_enc_cfg_set_s32(_encoder.cfg, "prep:ver_stride", _decoder.ver_stride);
    mpp_enc_cfg_set_s32(_encoder.cfg, "prep:format", _decoder.fmt);
    mpp_enc_cfg_set_s32(_encoder.cfg, "prep:color", _decoder.color);

    mpp_enc_cfg_set_s32(_encoder.cfg, "rc:mode", MPP_ENC_RC_MODE_VBR);

    //     /* fix input / output frame rate */
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:fps_in_flex", 0);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:fps_in_num", fps_in_num);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:fps_in_denorm", fps_in_den);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:fps_out_flex", 0);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:fps_out_num", fps_out_num);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:fps_out_denorm", fps_out_den);

    mpp_enc_cfg_set_s32(_encoder.cfg, "rc:bps_max", bps * 17 / 16);
    mpp_enc_cfg_set_s32(_encoder.cfg, "rc:bps_min", bps * 15 / 16);

    mpp_enc_cfg_set_s32(_encoder.cfg, "rc:gop", fps_out_num * 10); // 30 frames
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:bps_target", bps);

    // /* drop frame or not when bitrate overflow */
    // mpp_enc_cfg_set_u32(_encoder.cfg, "rc:drop_mode", MPP_ENC_RC_DROP_FRM_DISABLED);
    // mpp_enc_cfg_set_u32(_encoder.cfg, "rc:drop_thd", 20);        /* 20% of max bps */
    // mpp_enc_cfg_set_u32(_encoder.cfg, "rc:drop_gap", 1);         /* Do not continuous drop frame */

    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:qp_init", -1);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:qp_max", 51);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:qp_min", 10);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:qp_max_i", 51);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:qp_min_i", 10);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "rc:qp_ip", 2);

    // mpp_enc_cfg_set_s32(_encoder.cfg, "codec:type", MPP_VIDEO_CodingAVC);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "h264:profile", 66);
    // /*
    //     * H.264 level_idc parameter
    //     * 10 / 11 / 12 / 13    - qcif@15fps / cif@7.5fps / cif@15fps / cif@30fps
    //     * 20 / 21 / 22         - cif@30fps / half-D1@@25fps / D1@12.5fps
    //     * 30 / 31 / 32         - D1@25fps / 720p@30fps / 720p@60fps
    //     * 40 / 41 / 42         - 1080p@30fps / 1080p@30fps / 1080p@60fps
    //     * 50 / 51 / 52         - 4K@30fps
    //     */
    // mpp_enc_cfg_set_s32(_encoder.cfg, "h264:level", 40);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "h264:cabac_en", 1);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "h264:cabac_idc", 0);
    // mpp_enc_cfg_set_s32(_encoder.cfg, "h264:trans8x8", 1);

    int ret = _encoder.mpi->control(_encoder.ctx, MPP_ENC_SET_CFG, _encoder.cfg);
    if (ret) {
        ErrorL << "mpi control enc set cfg failed ret " << ret;
        return;
    }
    // MppPollType timeout = MPP_POLL_NON_BLOCK;
    // _encoder.mpi->control(_encoder.ctx, MPP_SET_INPUT_TIMEOUT, &timeout);

    MppPollType timeout = MPP_POLL_BLOCK;
    _encoder.mpi->control(_encoder.ctx, MPP_SET_OUTPUT_TIMEOUT, &timeout);

    MppEncHeaderMode header_mode = MPP_ENC_HEADER_MODE_DEFAULT;
    _encoder.mpi->control(_encoder.ctx, MPP_ENC_SET_HEADER_MODE, &header_mode);
}

zhang-chuanfeng, 2023-09-08:
> 

MPP_ENC_SET_IDR_FRAME   我这边强制I帧后返回的类型是 67 不是I帧

HermanChen, 2023-09-08:
默认配置下，I 帧会带 sps (nalu type 7) pps (8) SEI (6) SEI (6) 然后是 slice (5)，用 winhex 等 16 进制工具看一下

zhang-chuanfeng, 2023-09-08:
> winhex

那要单独拆分帧了

zhang-chuanfeng, 2023-09-13:
> > winhex
> 
> 那要单独拆分帧了
mpp_packet_get_segment_info  可以通过获取seg跳过sei帧

closed 2023-09-13 by zhang-chuanfeng