rockchip-linux/mpp #440: 编码1680x1050分辨率花屏

jukeboge, 2023-09-14:
使用：commit da6ad1ee5e8302f10884d2de626a269fa43c7c9f  代码
执行./data/test/mpi_enc_test  -o /data/out.h264 -w 1680 -h 1050 -f 0x00010007 -t 7 
播放out.h264显示花屏，-w 1920 -h 1080测试是显示正常的

HermanChen, 2023-09-15:
最新异常了么？

jukeboge, 2023-09-15:
最新的也异常（符有截图），硬件使用的是RK3588，日志信息如下:
```
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: cmd parse result:
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: input  file name: (null)
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: output file name: /data/out.h264
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: width      : 1680
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: height     : 1050
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: format     : 65543
09-15 10:06:06.332  1960  1960 I mpi_enc_utils: type       : 7
09-15 10:06:06.333  1960  1961 I mpi_enc_test: ./data/test/mpi_enc_test start
09-15 10:06:06.337  1960  1961 I mpp_info: mpp version: db560c14 author: Yandong Lin   2022-07-06 [jpege]: add new client type for vepu2 jpeg only
09-15 10:06:06.337  1960  1961 I mpi_enc_test: 0xb40000713d033ec0 encoder test start w 1680 h 1050 type 7
09-15 10:06:06.342  1960  1961 E mpp_cfg : mpp_enc_cfg_set_s32: cfg prep:flip is invalid
09-15 10:06:06.342  1960  1962 I mpp_enc : MPP_ENC_SET_RC_CFG bps 6615000 [413437 : 7028437] fps [30:30] gop 60
09-15 10:06:06.342  1960  1962 I h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1680:1050] stride [5040:1050]
09-15 10:06:06.342  1960  1962 I mpp_enc : send header for set cfg change input/format/color
09-15 10:06:06.347  1960  1962 I mpp_enc : mode vbr bps [413437:6615000:7028437] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
09-15 10:06:06.348  1960  1961 I utils   : warning: vepu only support 8 aligned horizontal stride in pixel for 24bit RGB with pixel size 3
09-15 10:06:06.348  1960  1961 I utils   : set byte stride to 5056 to match the requirement
09-15 10:06:06.374  1960  1961 I mpi_enc_test: chn 0 encoded frame 0    size 71464   qp 25
09-15 10:06:06.395  1960  1961 I mpi_enc_test: chn 0 encoded frame 1    size 26206   qp 26
09-15 10:06:06.409  1960  1961 I mpi_enc_test: chn 0 encoded frame 2    size 42302   qp 26
09-15 10:06:06.442  1960  1961 I mpi_enc_test: chn 0 encoded frame 3    size 296376  qp 38
09-15 10:06:06.471  1960  1961 I mpi_enc_test: chn 0 encoded frame 4    size 62216   qp 44
09-15 10:06:06.499  1960  1961 I mpi_enc_test: chn 0 encoded frame 5    size 69646   qp 47
09-15 10:06:06.528  1960  1961 I mpi_enc_test: chn 0 encoded frame 6    size 40052   qp 45
09-15 10:06:06.556  1960  1961 I mpi_enc_test: chn 0 encoded frame 7    size 53030   qp 45
09-15 10:06:06.585  1960  1961 I mpi_enc_test: chn 0 encoded frame 8    size 43472   qp 47
09-15 10:06:06.614  1960  1961 I mpi_enc_test: chn 0 encoded frame 9    size 52146   qp 48
09-15 10:06:06.642  1960  1961 I mpi_enc_test: chn 0 encoded frame 10   size 47932   qp 49
09-15 10:06:06.671  1960  1961 I mpi_enc_test: chn 0 encoded frame 11   size 44274   qp 50
09-15 10:06:06.699  1960  1961 I mpi_enc_test: chn 0 encoded frame 12   size 45922   qp 50
09-15 10:06:06.727  1960  1961 I mpi_enc_test: chn 0 encoded frame 13   size 46042   qp 50
09-15 10:06:06.756  1960  1961 I mpi_enc_test: chn 0 encoded frame 14   size 44522   qp 50
09-15 10:06:06.784  1960  1961 I mpi_enc_test: chn 0 encoded frame 15   size 44938   qp 50
09-15 10:06:06.813  1960  1961 I mpi_enc_test: chn 0 encoded frame 16   size 45492   qp 51
09-15 10:06:06.841  1960  1961 I mpi_enc_test: chn 0 encoded frame 17   size 46388   qp 51
```   
画面截图：
[image]

jukeboge, 2023-09-15:
"I utils   : set byte stride to 5056 to match the requirement"有问题，5040已经是24对齐，计算后的值5056 是不对的，试着改了下不花了：
```c
diff --git a/utils/utils.c b/utils/utils.c
index e00cc4fd..86f7b7d7 100644
--- a/utils/utils.c
+++ b/utils/utils.c
@@ -1301,7 +1301,8 @@ MPP_RET fill_image(RK_U8 *buf, RK_U32 width, RK_U32 height,
 
         if (util_check_8_pixel_aligned(not_8_pixel, hor_stride,
                                        8, pix_w, "24bit RGB")) {
-            hor_stride = MPP_ALIGN(hor_stride, 24);
+            hor_stride = ((hor_stride + 23) / 24 * 24);//MPP_ALIGN(hor_stride, 24);
+            mpp_log("new hor_stride %d \n",hor_stride);
             not_8_pixel = 1;
         }
```

Sherman86, 2023-09-15:
[d7d7c85.tar.gz](https://github.com/rockchip-linux/mpp/files/12616580/d7d7c85.tar.gz)
这个补丁应该可以解决问题。

Sherman86, 2023-09-15:
[bb651d2.tar.gz](https://github.com/rockchip-linux/mpp/files/12616697/bb651d2.tar.gz)

closed 2023-09-28 by HermanChen