rockchip-linux/mpp #445: 使用零拷贝的方式解码得到fd,在用oes渲染比较慢

sunqing128, 2023-09-18:
我现在用mpp的mpp_buffer_group_get_external方式解码出4k视频得到fd,然后将其转化为oesimage，生成oes纹理去渲染，测试发现渲染一帧4k需要大概60ms左右，这个是正常吗？

Ryan3812, 2023-09-18:
你好，你的转换oesimage和生成oes纹理有相关demo吗，我最近也在做这个

sunqing128, 2023-09-18:
> 你好，你的转换oesimage和生成oes纹理有相关demo吗，我最近也在做这个

有，但是4k比较慢

Ryan3812, 2023-09-18:
你好，可以分享下相关demo吗，最近在做这个做不出来

HermanChen, 2023-09-19:
试试从 gpu 分 buffer 出来，然后用 fd 导入到 mpp 里解码，这样的流程更常规一些

HermanChen, 2023-09-20:
从 mpp 生成 buffer 再导入到 gpu 体系去，每次 buffer 导入都需要时间，这样比较浪费
可以从 gpu 那边一次性分配多个 buffer 出来，导致到 mpp 里解码，这样解码完成之后把 buffer 给 gpu 的时候，就不需要导入操作了

Ryan3812, 2023-09-20:
> 试试从 gpu 分 buffer 出来，然后用 fd 导入到 mpp 里解码，这样的流程更常规一些

你好，这部分有相关demo吗

sunqing128, 2023-09-20:
> 从 mpp 生成 buffer 再导入到 gpu 体系去，每次 buffer 导入都需要时间，这样比较浪费 可以从 gpu 那边一次性分配多个 buffer 出来，导致到 mpp 里解码，这样解码完成之后把 buffer 给 gpu 的时候，就不需要导入操作了

这样是不是还需要将fd转化成oes纹理？只不过把buffer提前分配了，还是说不用oes纹理，将buffer生成普通2D纹理？

Sherman86, 2023-09-20:
> 你好，这部分有相关demo吗
mpp/base/test/mpp_buffer_test.c

Sherman86, 2023-09-20:
> 这样是不是还需要将fd转化成oes纹理？只不过把buffer提前分配了，还是说不用oes纹理，将buffer生成普通2D纹理？
还需要将fd转化成oes纹理，只不过把buffer提前分配了，以降低导入操作的时间。原本渲染流程不变。

sunqing128, 2023-09-27:
> > 这样是不是还需要将fd转化成oes纹理？只不过把buffer提前分配了，还是说不用oes纹理，将buffer生成普通2D纹理？
> 
> 还需要将fd转化成oes纹理，只不过把buffer提前分配了，以降低导入操作的时间。原本渲染流程不变。

之前在忙其他的，我看了你说的那个mpp_buffer_test.c，我也是参考的那一部分

Sherman86, 2023-09-27:
> 之前在忙其他的，我看了你说的那个mpp_buffer_test.c，我也是参考的那一部分

已经是 gpu 中 导入 buffer 给 mpp 了？

可以看下 ddr 带宽 够不够：
```sh
### GPU 带宽利用率
cat /sys/class/devfreq/ff300000.gpu/load

### VPU 带宽利用率
cat /sys/class/devfreq/ff360000.rkvdec/load
```

可以提高 ddr 频率试试。
```sh
DDR 频率调节相关
定频的场景：
cat /sys/kernel/debug/clk/clk_summary | grep ddr
sclk_ddrc 即为ddr工作频率

开启变频的场景，查看ddr工作频率：
cat /sys/class/devfreq/dmc/cur_freq

变频场景下使用固定频率的方法：

获取固件支持的 DDR 频率：
cat /sys/class/devfreq/dmc/available_frequencies

设置 DDR 频率为 300000000 Hz：
echo userspace > /sys/class/devfreq/dmc/governor 
echo 300000000 > /sys/class/devfreq/dmc/min_freq 
echo 300000000 > /sys/class/devfreq/dmc/userspace/set_freq
```

closed 2023-09-28 by HermanChen