rockchip-linux/mpp #446: 1.5.0-1+git230913.5f1cd0f1~jammy: gstreamer mpph265enc issues using rtmpsrc compositor stopped working

moo-the-cow, 2023-09-18:
I'm using the launchpad packages/repo for the orangepi from https://launchpad.net/~liujianfeng1994/+archive/ubuntu/rockchip-multimedia
The version before this release worked just fine. 

here relevant syslog messages. those are the only ones produced during the start of that pipeline:
```
Sep 18 08:06:10 mpp: unable to create enc vp8 for soc rk3566 unsupported
Sep 18 08:06:10 irlbox mpp[12631]: mpp_info: mpp version: unknown mpp version for missing VCS info
Sep 18 08:06:10 irlbox mpp[12631]: message repeated 2 times: [ mpp_info: mpp version: unknown mpp version for missing VCS info]
```

moo-the-cow, 2023-09-18:
I want to point out that I'm not using vp8 at all. the source video is h264 via rtmpsrc and flvdemux
`h264parse ! avdec_h264 ! videoconvert ! video/x-raw,width={mainVideoWidth},height={mainVideoHeight}`

HermanChen, 2023-09-20:
I seems that gst plugin has some error on calling mpi interface?
@JeffyCN

moo-the-cow, 2023-09-20:
sorry for not being familar with c code. I wish I could provide more infos, but thats literally all the logs it showed me.
luckily I had "librockchip_mpp.so.0" backed up so overwriting the new
`/lib/aarch64-linux-gnu/librockchip_mpp.so.0`
helped, but I'm aware that I will keep having those issues on updates, so I wanted to show it up.
thank you for your help @HermanChen

JeffyCN, 2023-09-20:
it means mpp doesn't suppot vp8 enc, the gst-mpp would not register vp8enc plugin in the initial registering.

closed 2023-10-13 by HermanChen