rockchip-linux/mpp #448: [av1d] certain AV1 video clip cannot be decoded

nyanmisaka, 2023-09-26:
Platform: rk3588, Kernel: 5.10.160

[av1.ivf.zip](https://github.com/rockchip-linux/mpp/files/12730956/av1.ivf.zip)

```
Input #0, ivf, from 'av1.ivf':
  Duration: 00:00:41.71, start: 0.007000, bitrate: 1375 kb/s
  Stream #0:0: Video: av1 (Main) (AV01 / 0x31305641), yuv420p10le(tv, bt2020nc/bt2020/smpte2084), 3840x1604 [SAR 1:1 DAR 960:401], Film Grain, 23.98 fps, 23.98 tbr, 1k tbn
```

The video can be decoded fine via software (dav1d) or other hardware decoders (dxva).

IMHO it seems to be a parser issue. `Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786`
https://github.com/rockchip-linux/mpp/blob/5f1cd0f12d643f017aea61a196b3b766df511bf6/mpp/codec/dec/av1/av1d_cbs.c#L2786

```
[AVHWDeviceContext @ 0xaaaaea5f9680] Opened Rockchip DRM device /dev/dri/card0: driver rockchip version 3.0.0.
mpp[47667]: mpi: mpp_create enter ctx 0xaaaaea64eb10 mpi 0xaaaaea64eb08
mpp[47667]: mpp_info: mpp version: 5f1cd0f1 author: Yandong Lin   2023-09-13 fix[h264e_amend]: fix slice read overread issue
mpp[47667]: mpi: mpp_create leave ret 0 ctx 0xaaaaea5fd0c0 mpi 0xffff9f8362f8
mpp[47667]: mpi: mpp_init enter ctx 0xaaaaea5fd0c0 type 0 coding 16777224
mpp[47667]: av1d_parser: av1d_split_init enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_split_init leave ctx 0xaaaaea61db40
mpp[47667]: mpi: mpp_init leave ctx 0xaaaaea5fd0c0 ret 0
mpp[47667]: mpi: mpi_control enter ctx 0xaaaaea5fd0c0 cmd 310006 parm 0xaaaaea64ed88
mpp[47667]: mpp_dec: can not enable fast parse while hal not support
mpp[47667]: mpi: mpi_control leave ctx 0xaaaaea5fd0c0 ret 0
mpp[47667]: mpi: mpi_control enter ctx 0xaaaaea5fd0c0 cmd 31000b parm (nil)
mpp[47667]: mpi: mpi_control leave ctx 0xaaaaea5fd0c0 ret 0
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (av1_rkmpp) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
mpp[47667]: mpi: mpi_decode_put_packet enter ctx 0xaaaaea5fd0c0 packet 0xaaaaea612c48
mpp[47667]: mpi: mpi_decode_put_packet leave ctx 0xaaaaea5fd0c0 ret 0
mpp[47667]: mpi: mpi_decode_put_packet enter ctx 0xaaaaea5fd0c0 packet 0xaaaaea612c48
mpp[47667]: mpi: mpi_decode_put_packet leave ctx 0xaaaaea5fd0c0 ret 0
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 3 (type 5).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_read_unsigned Invalid value at lr_type[i]: bitstream ended.
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 1 (type 1).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_read_unsigned Invalid value at show_existing_frame: bitstream ended.
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 0 (type 2).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_read_unsigned Invalid value at show_existing_frame: bitstream ended.
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 0 (type 2).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_read_unsigned Invalid value at show_existing_frame: bitstream ended.
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 0 (type 2).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
[... 38 lines cut ...]
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_uncompressed_header Missing reference frame needed for show_existing_frame (frame_to_show_map_idx = 5).
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 1 (type 1).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_read_unsigned Invalid value at allow_warped_motion: bitstream ended.
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 1 (type 1).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_parser_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_cbs: Assertion obu->header.obu_type == unit->type failed at mpp_av1_read_unit:2786
mpp[47667]: av1d_cbs: mpp_av1_read_unsigned Invalid value at show_existing_frame: bitstream ended.
mpp[47667]: av1d_cbs: mpp_av1_read_fragment_content Failed to read unit 0 (type 2).
mpp[47667]: av1d_parser: av1d_split_frame enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream enter ctx 0xaaaaea61db40
mpp[47667]: av1d_parser: av1d_get_frame_stream leave ctx 0xaaaaea61db40
^C^C^CReceived > 3 system signals, hard exiting
```

FumasterLin, 2023-09-27:
[0001-fix-av1d_cbs-fix-read-32bit-data-err.patch](https://github.com/rockchip-linux/mpp/files/12735884/0001-fix-av1d_cbs-fix-read-32bit-data-err.patch)
Please apply the patch base on lastest mpp and test again

nyanmisaka, 2023-09-27:
```
[AVHWDeviceContext @ 0xaaab02773a30] Opened Rockchip DRM device /dev/dri/card0: driver rockchip version 3.0.0.
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (av1_rkmpp) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf60.3.100
  Stream #0:0: Video: wrapped_avframe, drm_prime(progressive), 3840x1604 [SAR 1:1 DAR 960:401], q=2-31, 200 kb/s, 23.98 fps, 23.98 tbn
    Metadata:
      encoder         : Lavc60.3.100 wrapped_avframe
frame= 1000 fps= 87 q=-0.0 Lsize=N/A time=00:00:41.66 bitrate=N/A speed=3.62x    0x
video:469kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
```

@FumasterLin Thanks for your patch. It works fine now.

Will the fix be included in the develop branch any time soon?

FumasterLin, 2023-09-27:
Yes, it will be merge into develop branch in the future.

closed 2023-09-27 by nyanmisaka