rockchip-linux/mpp #451: mpp errors

Wazirri, 2023-10-01:
```
[Sun Oct  1 19:58:14 2023] mpp_rkvdec2 fdf80200.rkvdec: session 40 task 1314609 timeout 0 abort 0 force_dequeue 1
[Sun Oct  1 19:58:14 2023] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[Sun Oct  1 19:58:14 2023] mpp_rkvdec2 fdf80200.rkvdec: reset done
[Sun Oct  1 19:58:16 2023] mpp_rkvdec2 fdf80200.rkvdec: session 40 task 1314634 timeout 0 abort 0 force_dequeue 1
[Sun Oct  1 19:58:16 2023] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[Sun Oct  1 19:58:16 2023] mpp_rkvdec2 fdf80200.rkvdec: reset done
[Sun Oct  1 19:59:33 2023] mpp_rkvdec2 fdf80200.rkvdec: session 40 task 1316454 timeout 0 abort 0 force_dequeue 1
[Sun Oct  1 19:59:33 2023] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[Sun Oct  1 19:59:33 2023] mpp_rkvdec2 fdf80200.rkvdec: reset done
[Sun Oct  1 20:10:28 2023] mpp_rkvdec2 fdf80200.rkvdec: session 40 task 1332158 timeout 0 abort 0 force_dequeue 1
[Sun Oct  1 20:10:28 2023] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[Sun Oct  1 20:10:28 2023] mpp_rkvdec2 fdf80200.rkvdec: reset done
[Sun Oct  1 20:10:30 2023] mpp_rkvdec2 fdf80200.rkvdec: session 40 task 1332183 timeout 0 abort 0 force_dequeue 1
[Sun Oct  1 20:10:30 2023] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[Sun Oct  1 20:10:30 2023] mpp_rkvdec2 fdf80200.rkvdec: reset done
[Sun Oct  1 20:19:46 2023] mpp_rkvdec2 fdf80200.rkvdec: irq on power off
[Sun Oct  1 20:21:51 2023] mpp_rkvdec2 fdf80200.rkvdec: irq on power off
[Sun Oct  1 20:26:42 2023] mpp_rkvdec2 fdf80200.rkvdec: irq on power off
```

mpp gives those errors, after not certain time it crushes firmware, and device cannot boot again.

Any help?

Wazirri, 2023-10-05:
While device booting dmesg gives those errors;

`[    3.161262] mpp_rkvenc fdf40000.rkvenc: Adding to iommu group 5
[    3.161473] mpp_rkvenc fdf40000.rkvenc: probing start
[    3.161911] mpp_rkvenc fdf40000.rkvenc: Failed to get leakage
[    3.161958] mpp_rkvenc fdf40000.rkvenc: pvtm = 88570, from nvmem
[    3.161968] mpp_rkvenc fdf40000.rkvenc: pvtm-volt-sel=1
[    3.162097] mpp_rkvenc fdf40000.rkvenc: avs=0
[    3.162193] mpp_rkvenc fdf40000.rkvenc: failed to find power_model node
[    3.162201] mpp_rkvenc fdf40000.rkvenc: failed to initialize power model
[    3.162206] mpp_rkvenc fdf40000.rkvenc: failed to get dynamic-coefficient
[    3.162590] mpp_rkvenc fdf40000.rkvenc: probing finish
[    3.162911] mpp_rkvdec2 fdf80200.rkvdec: Adding to iommu group 6
[    3.163082] mpp_rkvdec2 fdf80200.rkvdec: rkvdec, probing start
[    3.163319] mpp_rkvdec2 fdf80200.rkvdec: shared_niu_a is not found!
[    3.163327] rkvdec2_init:1022: No niu aclk reset resource define
[    3.163335] mpp_rkvdec2 fdf80200.rkvdec: shared_niu_h is not found!
[    3.163339] rkvdec2_init:1025: No niu hclk reset resource define
[    3.163552] mpp_rkvdec2 fdf80200.rkvdec: leakage=70
[    3.163563] mpp_rkvdec2 fdf80200.rkvdec: leakage-volt-sel=0
[    3.163593] mpp_rkvdec2 fdf80200.rkvdec: pvtm = 88570, from nvmem
[    3.163602] mpp_rkvdec2 fdf80200.rkvdec: pvtm-volt-sel=1
[    3.163727] mpp_rkvdec2 fdf80200.rkvdec: avs=0
[    3.163809] mpp_rkvdec2 fdf80200.rkvdec: failed to find power_model node
[    3.163817] mpp_rkvdec2 fdf80200.rkvdec: failed to initialize power model
[    3.163822] mpp_rkvdec2 fdf80200.rkvdec: failed to get dynamic-coefficient
[    3.163909] mpp_rkvdec2 fdf80200.rkvdec: sram_start 0x00000000fdcc0000
[    3.163917] mpp_rkvdec2 fdf80200.rkvdec: rcb_iova 0x0000000010000000
[    3.163923] mpp_rkvdec2 fdf80200.rkvdec: sram_size 45056
[    3.163928] mpp_rkvdec2 fdf80200.rkvdec: rcb_size 65536
[    3.163934] mpp_rkvdec2 fdf80200.rkvdec: min_width 512
[    3.163972] mpp_rkvdec2 fdf80200.rkvdec: link mode probe finish
[    3.164029] mpp_rkvdec2 fdf80200.rkvdec: probing finish`

After sometimes our program works, rootfs crashing, and it does not boot again.

`DDR V1.16 6f71c736ce typ 23/03/02-20:01:48
In
LP4/4x derate en, other dram:1x trefi
ddrconfig:0
LP4 MR14:0x4d
LPDDR4, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
tdqss: cs0 dqs0: 48ps, dqs1: -72ps, dqs2: -48ps, dqs3: -120ps, 

change to: 324MHz
PHY drv:clk:38,ca:38,DQ:30,odt:0
vrefinner:41%, vrefout:41%
dram drv:40,odt:0
clk skew:0x61

change to: 528MHz
PHY drv:clk:38,ca:38,DQ:30,odt:0
vrefinner:41%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 780MHz
PHY drv:clk:38,ca:38,DQ:30,odt:0
vrefinner:41%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 1560MHz(final freq)
PHY drv:clk:38,ca:38,DQ:30,odt:60
vrefinner:16%, vrefout:29%
dram drv:40,odt:80
vref_ca:00000068
clk skew:0x1c
cs 0:
[... 1233 lines cut ...]
[    7.151396] 112:          0          0          0          0  rockchip_gpio_irq  13 Edge      gt9xx
[    7.152183] PMU:
[    7.152347] pwr       0x00a0: 00000067 00000000 00000003 00000000
[    7.152891] status    0x0098: 00000067 00000000 00000067 00000000
[    7.153433] req       0x0050: 0000006e 00000000 00000000 00000000
[    7.153974] idle      0x0068: 0000006e 00000000 0000ffff 0000000f
[    7.154516] ack       0x0060: 0000006e 00000000 0000006e 00000000
[    7.155062] rockchip-thermal fe710000.tsadc: channal 0: temperature(53 C)
[    7.155656] THERMAL REGS:
[    7.155893] 00000000: 00000fc0 00030033 00000301 00000002 00000000 00000000 00000000 00000000
[    7.156641] 00000020: 000008a8 000008b0 00000000 00000000 00000940 00000000 00000000 00000000
[    7.157389] 00000040: 00000a70 00000a70 00000000 00000000 00000000 00000000 00000000 00000000
[    7.158135] 00000060: 00000004 00000004 00000656 00000656 00000000 00000000 00000000 00000000
[    7.158876] 00000080: 00000000 00000000
[    7.159221] Kernel Offset: disabled
[    7.159532] CPU features: 0x000,08000026,2a00aa18
[    7.159945] Memory Limit: none
[    7.160232] cpu cpu0: cur_freq: 1800000000 Hz, volt: 1100000 uV
[    7.171818] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]---`

Is that possible mpp can cause device to crash?

We still could not understand how?

Thanks for help...

HermanChen, 2023-10-07:
> [ 5.733421] I : [File] : drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 406; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '11:52:21', on 'Oct 3 2023'.

The kernel gpu ko version does not match with the userspace expectation. It is the offical firmware package?

And about the mpp fault it seems that some error video stream is decoding on old version kernel driver without sip reset in loader. It is fixed in new firmware.

closed 2023-10-07 by HermanChen