rockchip-linux/mpp #454: 解码1920*1088分辨率的H264文件，花屏

geiliwanghaichao, 2023-10-07:
芯片型号为RV1126
使用vlc播放H264文件，vlc可以正常播放，
[image]
看到的视频分辨率为1920*1080，缓冲大小为1920*1090，
使用板子播放看到的分辨率和stride为：
[image]
看到的stride为1920*1080，解码后的yuv数据花屏
请问是什么原因呢

HermanChen, 2023-10-07:
1090? 感觉是外部给编码器的配置有问题吧？试试其他分辨率，比如 1280x720 看看~

HermanChen, 2023-10-07:
还有，这个是在用编码器编码吧？不是解码？是 1126 自编自解出来的 yuv 花屏了么？

geiliwanghaichao, 2023-10-07:
@HermanChen 您好，感谢您的回复。H264视频文件是客户提供给我的，我用vlc播放，是可以正常播放的，上面也有截图。使用1126的板子，是fopen读取视频文件，在喂给解码器，将解码后的数据，保存成yuv文件(格式为nv12)，使用yuvplayer工具，看到yuv数据花屏。从mpp的log中看到decoder的日志 stride为1920*1088
[image]

HermanChen, 2023-10-07:
用 ffmpeg 抽出祼码流之后，用 mpi_dec_test 解码看看
你们的解码代码里要按 mpi_dec_test 里解码之后的写文件方式来保存 yuv，1920 的 stride 可能应该不是 1920

geiliwanghaichao, 2023-10-07:
@HermanChen 您好，我使用mpi_dec_test，命令如下：./mpi_dec_test -i case9_truck.h264 -o abc.h264 -t 7，终端打印为：
[image]
使用yuvplayer播放，仍旧是花屏
[image]

Sherman86, 2023-10-08:
片源是采样格式是 444 的，硬件不支持。

geiliwanghaichao, 2023-10-08:
@Sherman86 您好，感谢您的回复。另外请问针对RV1126这款芯片，profile_idc 都支持哪些呢，或者您方便告诉我去哪里查询这个信息呢，谢谢

Sherman86, 2023-10-08:
datasheet 中可以查到，但不对外开放。支持的 profile_idc 和 level 如下所示：

Baseline, main, high, high10 and high 4:2:2(without MBAFF), up to level 5.1 and 4096x2304@30fps

geiliwanghaichao, 2023-10-08:
@Sherman86 好的，多谢

closed 2023-10-08 by geiliwanghaichao