rockchip-linux/mpp #459: How to build as static libary

MrPlatnum, 2023-10-15:
I modified the build script found in build/linux/aarch64 like this:
```#!/bin/bash
# Run this from within a bash shell

set +e

MPP_PWD=`pwd`
MPP_TOP=${MPP_PWD}/../../..

# toolchain detection
check_cmd(){
    "$@" >> /dev/null 2>&1
}
check_system_arm_linux_gcc(){
    check_cmd aarch64-linux-gnu-gcc -v
}

check_system_arm_linux_gcc
if [ $? -eq 127 ];then
    MPP_TOOLCHAIN=/usr/bin
    export PATH=$PATH:${MPP_TOOLCHAIN}
fi

# generate Makefile
cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
      -DHAVE_DRM=ON \
      -DENABLE_STATIC=ON \
      -DENABLE_SHARED=OFF \
      -G "Unix Makefiles" \
      ${MPP_TOP}
```

But it wont build the librockip_mpp as a static library.
I need a static version of this library to inculde it in a static binary of ffmpeg.
All help would be appreciated.

```[ 71%] Built target hal_h264d
[ 71%] Building CXX object mpp/hal/CMakeFiles/mpp_hal.dir/mpp_enc_hal.cpp.o
[ 71%] Building CXX object mpp/hal/CMakeFiles/mpp_hal.dir/hal_task.cpp.o
[ 71%] Building CXX object mpp/hal/CMakeFiles/mpp_hal.dir/mpp_hal.cpp.o
[ 72%] Linking CXX static library libmpp_hal.a
[ 72%] Built target mpp_hal
[ 72%] Building CXX object mpp/CMakeFiles/rockchip_mpp.dir/mpp_info.cpp.o
[ 72%] Building CXX object mpp/CMakeFiles/rockchip_mpp.dir/mpp.cpp.o
[ 72%] Building CXX object mpp/CMakeFiles/rockchip_mpp.dir/mpp_impl.cpp.o
[ 73%] Building CXX object mpp/CMakeFiles/rockchip_mpp.dir/mpi.cpp.o
[ 73%] Linking CXX shared library librockchip_mpp.so
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libc.a(dl-support.o): in function `_dl_aux_init':
(.text+0x30): undefined reference to `_start'
/usr/bin/ld: (.text+0x38): undefined reference to `_start'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libc.a(libc-start.o): in function `call_fini':
(.text+0x4): undefined reference to `__fini_array_start'
/usr/bin/ld: (.text+0x8): undefined reference to `__fini_array_start'
/usr/bin/ld: (.text+0x14): undefined reference to `__fini_array_end'
/usr/bin/ld: (.text+0x18): undefined reference to `__fini_array_end'
/usr/bin/ld: (.text+0x28): undefined reference to `__fini_array_start'
/usr/bin/ld: (.text+0x2c): undefined reference to `__fini_array_start'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libc.a(libc-start.o): in function `__libc_start_main_impl':
(.text+0x378): undefined reference to `__preinit_array_end'
/usr/bin/ld: (.text+0x37c): undefined reference to `__preinit_array_start'
/usr/bin/ld: (.text+0x380): undefined reference to `__preinit_array_end'
/usr/bin/ld: (.text+0x384): undefined reference to `__preinit_array_start'
/usr/bin/ld: (.text+0x3c4): undefined reference to `__init_array_start'
/usr/bin/ld: (.text+0x3c8): undefined reference to `__init_array_end'
/usr/bin/ld: (.text+0x3cc): undefined reference to `__init_array_start'
/usr/bin/ld: (.text+0x3d0): undefined reference to `__init_array_end'

```

Sherman86, 2023-10-16:
[1262d6e.tar.gz](https://github.com/rockchip-linux/mpp/files/12915172/1262d6e.tar.gz)

Attached patch could be referred. All the library referred should be set to static.

MrPlatnum, 2023-10-19:
This does compile correctly but i unfortunatly always get the error:
"ERROR: rockchip_mpp not found using pkg-config"
When trying to include it in a project.
Is the .pc file faulty maybe?

HermanChen, 2023-10-20:
https://github.com/rockchip-linux/mpp/blob/dbaf9f361bbaf7b22d83bbfa4bb6e49c9cf9aa53/mpp/CMakeLists.txt#L52C1-L52C1
Try set the property to static?

MrPlatnum, 2023-10-20:
Thank you again for your response.
It compiles fine as a static lib when i change the line you mentioned but as soon as try to use the lib i get a whole lot of "undefined references" like these:
```pkg-config --exists --print-errors rockchip_mpp
check_func_headers rockchip/rk_mpi.h mpp_create -I/usr/local/include -L/usr/local/lib -lrockchip_mpp
test_ld cc -I/usr/local/include -L/usr/local/lib -lrockchip_mpp
test_cc -I/usr/local/include -L/usr/local/lib
BEGIN /tmp/ffconf.gF7TISdW/test.c
    1   #include <rockchip/rk_mpi.h>
    2   #include <stdint.h>
    3   long check_mpp_create(void) { return (long) mpp_create; }
    4   int main(void) { int ret = 0;
    5    ret |= ((intptr_t)check_mpp_create) & 0xFFFF;
    6   return ret; }
END /tmp/ffconf.gF7TISdW/test.c
gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -std=c11 -fomit-frame-pointer -fPIC -pthread -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/fribidi -I/usr/include/libdrm -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/opus -I/usr/include/opus -DX264_API_IMPORTS -I/usr/local/include -L/usr/local/lib -c -o /tmp/ffconf.gF7TISdW/test.o /tmp/ffconf.gF7TISdW/test.c
gcc -Wl,--as-needed -Wl,-z,noexecstack -I/usr/local/include -L/usr/local/lib -o /tmp/ffconf.gF7TISdW/test /tmp/ffconf.gF7TISdW/test.o -lrockchip_mpp
/usr/bin/ld: /usr/local/lib/librockchip_mpp.a(mpi.cpp.o): in function `_check_mpp_ctx(MpiImpl_t*, char const*)':
/home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:94: undefined reference to `_mpp_err'
/usr/bin/ld: /usr/local/lib/librockchip_mpp.a(mpi.cpp.o): in function `mpi_decode(void*, void*, void**)':
/home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:105: undefined reference to `_mpp_log_l'
/usr/bin/ld: /home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:123: undefined reference to `_mpp_log_l'
/usr/bin/ld: /usr/local/lib/librockchip_mpp.a(mpi.cpp.o): in function `mpi_decode_put_packet(void*, void*)':
/home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:132: undefined reference to `_mpp_log_l'
/usr/bin/ld: /home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:139: undefined reference to `_mpp_log_l'
/usr/bin/ld: /home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:147: undefined reference to `_mpp_log_l'
/usr/bin/ld: /usr/local/lib/librockchip_mpp.a(mpi.cpp.o):/home/server/ffbuild/ffmpeg/mpp/mpp-1.5.0/mpp/mpi.cpp:156: more undefined references to `_mpp_log_l' follow
/usr/bin/ld: /usr/local/lib/librockchip_mpp.a(mpi.cpp.o): in function `mpp_create':
...
```

And finaly the error:
```
collect2: error: ld returned 1 exit status
ERROR: rockchip_mpp not found using pkg-config
```

Do you have any idea what could be going on here?

The project compiled and installed correctly:
```
Install the project...
-- Install configuration: "Debug"
-- Up-to-date: /usr/local/include/rockchip
-- Up-to-date: /usr/local/include/rockchip/mpp_rc_defs.h
-- Up-to-date: /usr/local/include/rockchip/rk_vdec_cmd.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_cfg.h
-- Up-to-date: /usr/local/include/rockchip/rk_mpi.h
-- Up-to-date: /usr/local/include/rockchip/rk_mpi_cmd.h
-- Up-to-date: /usr/local/include/rockchip/mpp_rc_api.h
-- Up-to-date: /usr/local/include/rockchip/mpp_log.h
-- Up-to-date: /usr/local/include/rockchip/rk_type.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_cmd.h
-- Up-to-date: /usr/local/include/rockchip/mpp_log_def.h
-- Up-to-date: /usr/local/include/rockchip/mpp_meta.h
-- Up-to-date: /usr/local/include/rockchip/mpp_packet.h
-- Up-to-date: /usr/local/include/rockchip/mpp_err.h
-- Up-to-date: /usr/local/include/rockchip/rk_vdec_cfg.h
-- Up-to-date: /usr/local/include/rockchip/vpu.h
-- Up-to-date: /usr/local/include/rockchip/rk_hdr_meta_com.h
-- Up-to-date: /usr/local/include/rockchip/mpp_buffer.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_rc.h
-- Up-to-date: /usr/local/include/rockchip/mpp_compat.h
-- Up-to-date: /usr/local/include/rockchip/mpp_task.h
-- Up-to-date: /usr/local/include/rockchip/vpu_api.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_ref.h
-- Up-to-date: /usr/local/include/rockchip/mpp_frame.h
-- Up-to-date: /usr/local/lib/pkgconfig/rockchip_mpp.pc
-- Up-to-date: /usr/local/lib/pkgconfig/rockchip_vpu.pc
-- Up-to-date: /usr/local/lib/librockchip_mpp.a
-- Up-to-date: /usr/local/lib/librockchip_vpu.so.0
-- Up-to-date: /usr/local/lib/librockchip_vpu.so.1
-- Up-to-date: /usr/local/lib/librockchip_vpu.so
-- Up-to-date: /usr/local/bin/mpp_info_test
-- Up-to-date: /usr/local/bin/mpi_dec_test
-- Up-to-date: /usr/local/bin/mpi_dec_mt_test
-- Up-to-date: /usr/local/bin/mpi_dec_nt_test
-- Up-to-date: /usr/local/bin/mpi_enc_test
-- Up-to-date: /usr/local/bin/mpi_enc_mt_test
-- Up-to-date: /usr/local/bin/mpi_rc2_test
-- Up-to-date: /usr/local/bin/mpi_dec_multi_test
-- Up-to-date: /usr/local/bin/vpu_api_test
```