rockchip-linux/mpp #465: 使用mpp_enc_test遇到的问题

Kevin111369, 2023-10-30:
在使用mpi_enc_test时报错：
```
mpp[226406]: camera_source: camera_source_init get dma buf failed
mpp[226406]: mpi_enc_test: new framecap ok
mpp[226406]: mpi_enc_test: open /dev/video1 fail
mpp[226406]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[226406]: mpi_enc_test: 0x7f88001e40 encoder test start w 1920 h 1080 type 7
```

使用mpp_buffer_test测试dma报错：
```
mpp[226736]: mpp_buffer_test: mpp_buffer_test normal mode start
mpp[226736]: mpp_buffer: mpp_buffer_group_init enter
mpp[226736]: mpp_buffer: group   4 mode 0 type 1 ops grp create 
mpp[226736]: mpp_buffer: mpp_buffer_group_init leave
mpp[226736]: mpp_buffer: mpp_buffer_get_unused enter
mpp[226736]: mpp_buffer: mpp_buffer_get_unused leave
mpp[226736]: mpp_buffer: mpp_buffer_create enter
mpp[226736]: mpp_dma_heap: ioctl alloc failed for Bad file descriptor
mpp[226736]: mpp_dma_heap: os_allocator_dma_heap_alloc dma_heap_alloc failed ret -1
mpp[226736]: mpp_buffer: mpp_buffer_create failed to create buffer with size 1024
mpp[226736]: mpp_buffer: mpp_buffer_create leave
mpp[226736]: mpp_buffer_test: mpp_buffer_test mpp_buffer_get mode normal failed
mpp[226736]: mpp_buffer: mpp_buffer_group_deinit enter
mpp[226736]: mpp_buffer: group   4 mode 0 type 1 ops grp release
mpp[226736]: mpp_buffer: group   4 mode 0 type 1 ops grp destroy
mpp[226736]: mpp_buffer: mpp_buffer_group_deinit leave
mpp[226736]: mpp_buffer_test: mpp_buffer_test failed
mpp[226736]: mpp_buffer: ~MppBufferService cleaning misc group
mpp[226736]: mpp_buffer: group   2 mode 0 type 1 ops grp release
mpp[226736]: mpp_buffer: group   2 mode 0 type 1 ops grp destroy
mpp[226736]: mpp_buffer: group   3 mode 1 type 1 ops grp release
mpp[226736]: mpp_buffer: group   3 mode 1 type 1 ops grp destroy
```

HermanChen, 2023-10-31:
camera_source_init get dma buf failed
确认下环境上 /dev/dma_heap/ 目录是否存在，是否用 root 权限运行测试用例

Kevin111369, 2023-10-31:
> camera_source_init get dma buf failed
> 确认下环境上 /dev/dma_heap/ 目录是否存在，是否用 root 权限运行测试用例

这个方法我已经尝试过了，运行 buffer test 还是报错

Kevin111369, 2023-10-31:
> > camera_source_init get dma buf failed
> > 确认下环境上 /dev/dma_heap/ 目录是否存在，是否用 root 权限运行测试用例
> 
> 这个方法我已经尝试过了，运行 buffer test 还是报错

```
root@linaro-alip:~/std# ls /dev/dma_heap/
cma  cma-uncached  system  system-dma32  system-uncached  system-uncached-dma32
root@linaro-alip:~/std/mpp_test# ./mpp_buffer_test 
mpp[5488]: mpp_buffer_test: mpp_buffer_test start with debug 0xf
mpp[5488]: mpp_buffer: group   1 mode 1 type 1 ops grp create 
mpp[5488]: mpp_buffer: mpp_buffer_group_init leave
mpp[5488]: mpp_buffer_test: mpp_buffer_test commit mode with unused status start
mpp[5488]: mpp_buffer: group   2 mode 0 type 1 ops grp create 
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
mpp[5488]: mpp_buffer: mpp_buffer_get_unused leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   2 buffer    0 fd   5 ops buf create  ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_mmap enter
mpp[5488]: mpp_buffer: group   2 buffer    0 fd   5 ops buf mmap    ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_mmap leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   1 buffer    0 fd   6 ops buf commit  ref_count 0 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
mpp[5488]: mpp_buffer: mpp_buffer_get_unused leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   2 buffer    1 fd   7 ops buf create  ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_mmap enter
mpp[5488]: mpp_buffer: group   2 buffer    1 fd   7 ops buf mmap    ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_mmap leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   1 buffer    1 fd   8 ops buf commit  ref_count 0 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
mpp[5488]: mpp_buffer: mpp_buffer_get_unused leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   2 buffer    2 fd   9 ops buf create  ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_mmap enter
mpp[5488]: mpp_buffer: group   2 buffer    2 fd   9 ops buf mmap    ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_mmap leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   1 buffer    2 fd  10 ops buf commit  ref_count 0 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
mpp[5488]: mpp_buffer: mpp_buffer_get_unused leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   2 buffer    3 fd  11 ops buf create  ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_mmap enter
mpp[5488]: mpp_buffer: group   2 buffer    3 fd  11 ops buf mmap    ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_mmap leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   1 buffer    3 fd  12 ops buf commit  ref_count 0 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
mpp[5488]: mpp_buffer: mpp_buffer_get_unused leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   2 buffer    4 fd  13 ops buf create  ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_mmap enter
mpp[5488]: mpp_buffer: group   2 buffer    4 fd  13 ops buf mmap    ref_count 1 caller main
mpp[5488]: mpp_buffer: mpp_buffer_mmap leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_buffer: group   1 buffer    4 fd  14 ops buf commit  ref_count 0 caller main
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
[... 290 lines cut ...]
mpp[5488]: mpp_buffer: mpp_buffer_ref_dec leave
mpp[5488]: mpp_buffer_test: mpp_buffer_test commit mode with used status success
mpp[5488]: mpp_buffer_test: mpp_buffer_test normal mode start
mpp[5488]: mpp_buffer: mpp_buffer_group_init enter
mpp[5488]: mpp_buffer: group   4 mode 0 type 1 ops grp create 
mpp[5488]: mpp_buffer: mpp_buffer_group_init leave
mpp[5488]: mpp_buffer: mpp_buffer_get_unused enter
mpp[5488]: mpp_buffer: mpp_buffer_get_unused leave
mpp[5488]: mpp_buffer: mpp_buffer_create enter
mpp[5488]: mpp_dma_heap: ioctl alloc failed for Bad file descriptor
mpp[5488]: mpp_dma_heap: os_allocator_dma_heap_alloc dma_heap_alloc failed ret -1
mpp[5488]: mpp_buffer: mpp_buffer_create failed to create buffer with size 1024
mpp[5488]: mpp_buffer: mpp_buffer_create leave
mpp[5488]: mpp_buffer_test: mpp_buffer_test mpp_buffer_get mode normal failed
mpp[5488]: mpp_buffer: mpp_buffer_group_deinit enter
mpp[5488]: mpp_buffer: group   4 mode 0 type 1 ops grp release
mpp[5488]: mpp_buffer: group   4 mode 0 type 1 ops grp destroy
mpp[5488]: mpp_buffer: mpp_buffer_group_deinit leave
mpp[5488]: mpp_buffer_test: mpp_buffer_test failed
mpp[5488]: mpp_buffer: ~MppBufferService cleaning misc group
mpp[5488]: mpp_buffer: group   2 mode 0 type 1 ops grp release
mpp[5488]: mpp_buffer: group   2 mode 0 type 1 ops grp destroy
mpp[5488]: mpp_buffer: group   3 mode 1 type 1 ops grp release
mpp[5488]: mpp_buffer: group   3 mode 1 type 1 ops grp destroy
```

HermanChen, 2023-11-16:
mpp[5488]: mpp_dma_heap: ioctl alloc failed for Bad file descriptor
mpp[5488]: mpp_dma_heap: os_allocator_dma_heap_alloc dma_heap_alloc failed ret -1

dma_heap 分配失败，只分配 1024 字节就失败……是 fd 太多了么？

Kevin111369, 2023-11-16:
> mpp[5488]: mpp_dma_heap: ioctl alloc failed for Bad file descriptor mpp[5488]: mpp_dma_heap: os_allocator_dma_heap_alloc dma_heap_alloc failed ret -1
> 
> dma_heap 分配失败，只分配 1024 字节就失败……是 fd 太多了么？

我使用的是firefly的开发板，换了一个开发板还是同样的错误，目前试了四个设备都是这个问题

HermanChen, 2023-11-29:
再最新版本上再试试

closed 2024-05-16 by Kevin111369