rockchip-linux/mpp #469: [h265d] HW resetting tainted the AFBC data that will be used as RGA3 input

nyanmisaka, 2023-11-08:
Hello here! I use AFBC v2 as the output for the MPP decoder to save DDR bandwidth. Then submit the AFBC data to RGA3 for format conversion. Small flaws in some video files may trigger a reset of the rkvdec driver, which is usually imperceptible to the user.

However, after this commit 60986ca, the rkvdec resetting will output *incorrect* AFBC data, causing RGA3 to reset too. Unfortunately, RGA3 driver will freeze the current process when resetting and cannot restore to the previous state like rkvdec driver. Therefore I don't want to trigger any RGA3 driver reset. After reveting the commit, everything returns to normal.

With the commit
```
[ 4241.277948] mpp_rkvdec2 fdc48100.rkvdec-core: resetting...
[ 4241.278100] mpp_rkvdec2 fdc48100.rkvdec-core: reset done
[ 4241.278105] mpp_rkvdec2 fdc38100.rkvdec-core: resetting...
[ 4241.278246] mpp_rkvdec2 fdc38100.rkvdec-core: reset done
[ 4241.285947] rga3_reg: irq handler err! INTR[0x20], HW_STATUS[0xfaaff], CMD_STATUS[0x1]
[ 4241.286000] rga3_reg: RGA3 core[2] soft reset complete.
[ 4241.286026] rga3_reg: win0 FBC decoder error, please check the fbc image of the source.
[ 4241.290291] rk_iommu fdb70f00.iommu: Enable stall request timed out, retry_count = 0, status: 0x000011
[ 4241.291390] rk_iommu fdb70f00.iommu: Disable paging request timed out, retry_count = 0, status: 0x000011
[ 4241.292464] rk_iommu fdb70f00.iommu: Enable stall request timed out, retry_count = 0, status: 0x000011
[ 4246.044161] rga_job: [tgid:4602 pid:4605] destroy request[193886] when the user exits
[ 4246.044172] rga3_reg: RGA3 core[2] soft reset complete.
[ 4246.044189] rga_job: reset core[2] by request[193886] abort
[ 4246.044216] rga_job: request[193886] abort! finished 0 failed 0 running_abort 1 todo_abort 0
[ 4246.044224] rga_job: [tgid:4602 pid:4605] destroy request[193888] when the user exits
[ 4246.044232] rga_job: request[193888] abort! finished 0 failed 0 running_abort 0 todo_abort 1
```

Without the commit
```
[ 4374.589755] mpp_rkvdec2 fdc48100.rkvdec-core: resetting...
[ 4374.589908] mpp_rkvdec2 fdc48100.rkvdec-core: reset done
[ 4374.589912] mpp_rkvdec2 fdc38100.rkvdec-core: resetting...
[ 4374.590050] mpp_rkvdec2 fdc38100.rkvdec-core: reset done
```

Context:
- RK3588
- kernel: 5.10.160-rockchip (rk-5.10-rkr6)
- sample video: [Sony_4k60_10b_clip_1_frame.mp4.zip](https://github.com/rockchip-linux/mpp/files/13294469/Sony_4k60_10b_clip_1_frame.mp4.zip) (Remove '.zip' in file name)

FumasterLin, 2023-11-08:
You need to use mpp_frame_get_errinfo api to check if it is an error frame, and do not send the error frame data to rga3 hw.

nyanmisaka, 2023-11-08:
Thanks, I learned it.
But what was that commit used for? Maybe Performance?

FumasterLin, 2023-11-08:
The commit is to avoid issue that may cause the system hang up in the case of the error stream as input.

nyanmisaka, 2023-11-08:
> The commit is to avoid issue that may cause the system hang up in the case of the error stream as input.

Thanks again.

closed 2023-11-08 by nyanmisaka