rockchip-linux/mpp #475: RK3588上调用mpi_dec_test失败（基于官网Ubuntu20系统）

shiqifeng2000, 2023-11-21:
使用mpp最新代码，切到release分支

检查环境
[image]

检查视频（h264, annexb）
[image]

调用mpi_dec_test
[image]

结果显示
[image]

FumasterLin, 2023-11-21:
release分支已经很久没有维护了，请使用develop分支

shiqifeng2000, 2023-11-21:
> release分支已经很久没有维护了，请使用develop分支

使用develop分支是无响应才切到的release

FumasterLin, 2023-11-21:
默认log是输出到sys了，export mpp_syslog_perror=1

shiqifeng2000, 2023-11-21:
> 默认log是输出到sys了，export mpp_syslog_perror=1

设置后再次执行
[image]

FumasterLin, 2023-11-21:
用root权限执行看下

shiqifeng2000, 2023-11-21:
> 用root权限执行看下

可以了，不过好像不是yuv420p，用nv12貌似正确，请问解码默认输出nv12么

FumasterLin, 2023-11-21:
是的 默认输出nv12

shiqifeng2000, 2023-11-21:
> 是的 默认输出nv12

-f 参数如何调整输出格式呢，我尝试 yuv420p, 数字4, 都不行

FumasterLin, 2023-11-21:
这个不支持配置输出格式哦

shiqifeng2000, 2023-11-21:
> 这个不支持配置输出格式哦

了解，谢谢

closed 2023-11-21 by shiqifeng2000